oscam-emu/oscam-patched-old #2: Missing header file

WanWizard, 2018-10-22:
https://github.com/oscam-emu/oscam-patched/blob/eb9531790e4c1f7a6bbe6a51e749c6cd7cbb0c9d/module-webif.c#L1225

This no longer exists after the split of the emu module.

oscam-emu, 2018-10-22:
Fixed in https://github.com/oscam-emu/oscam-patched/commit/23ea4a439ad0f44f9f7a5e196da1dd998869e7f2

Thanks for reporting, I wonder how I missed that, since it compiled perfectly fine here.

closed 2018-10-22 by oscam-emu

WanWizard, 2018-10-22:
You;'re welcome. More issues though: 
````
libcsmodules.a(module-emulator-drecrypt.o): In function `Drecrypt2ECM':
/usr/src/debug/enigma2-plugin-softcams-oscam-emu/gitAUTOINC+f5e8e82810-r0/git/module-emulator-drecrypt.c:127: undefined reference to `GetEcmLen'
libcsmodules.a(module-emulator-drecrypt.o): In function `DrecryptProcessEMM':
/usr/src/debug/enigma2-plugin-softcams-oscam-emu/gitAUTOINC+f5e8e82810-r0/git/module-emulator-drecrypt.c:290: undefined reference to `GetEcmLen'
libcsmodules.a(module-emulator-irdeto.o): In function `Irdeto2ECM':
/usr/src/debug/enigma2-plugin-softcams-oscam-emu/gitAUTOINC+f5e8e82810-r0/git/module-emulator-irdeto.c:134: undefined reference to `GetEcmLen'
libcsmodules.a(module-emulator-irdeto.o): In function `Irdeto2EMM':
/usr/src/debug/enigma2-plugin-softcams-oscam-emu/gitAUTOINC+f5e8e82810-r0/git/module-emulator-irdeto.c:354: undefined reference to `GetEcmLen'
libcsmodules.a(module-emulator-powervu.o): In function `PowervuECM':
/usr/src/debug/enigma2-plugin-softcams-oscam-emu/gitAUTOINC+f5e8e82810-r0/git/module-emulator-powervu.c:1260: undefined reference to `GetEcmLen'
libcsmodules.a(module-emulator-powervu.o):/usr/src/debug/enigma2-plugin-softcams-oscam-emu/gitAUTOINC+f5e8e82810-r0/git/module-emulator-powervu.c:1735: more undefined references to `GetEcmLen' follow
collect2: error: ld returned 1 exit status
CMakeFiles/oscam.dir/build.make:104: recipe for target 'oscam' failed
````

oscam-emu, 2018-10-22:
Yes, already aware of this. It should be fixed with https://github.com/oscam-emu/oscam-patched/commit/7a75b5e71a50581be0c4c7ef81f598d3b906c841 Thanks for reporting!

FWIW, error is present only when c99 or later is enabled and caused by the improper "inline" declaration in the function prototyping.

WanWizard, 2018-10-23:
Cool, we build our develop branch using AUTOREV, which builds from HEAD. I've currently pinned it on the commit hash of r774 to make sure tonights build succeeds. I'll continue checking tommorow.

PAKO34, 2018-11-02:
drecrypt2 not public keys