oscam-emu/oscam-patched-old #4: can you add biss2? biss ca?

PAKO34, 2018-11-02:
BISS2 Operational Modes
The Scrambler shall support the following four (4) modes of operation:
 Mode 0: No scrambling is applied.
 Mode 1: Components are scrambled by a Session Word (SW), and the SW is transmitted out
of band in clear to the receivers.
 Mode E: Components are scrambled by a Session Word (SW), the SW is encrypted with a
fixed Session Key (SK) and the resulting Encrypted Session Word (ESW) is transmitted out of
band to the receivers.
 Mode CA: Components are scrambled with a Session Word (SW), the SW is encrypted with a
Session Key (SK), and the resulting Encrypted Session Word (ESW), along with the key
information is transmitted in-stream to receivers. Both SW and SK are dynamically changed
during the live event transmission.

oscam-emu, 2018-11-02:
Do you came into any actual transmission encrypted in biss 2?

PAKO34, 2018-11-02:
be out sport 26e

oscam-emu, 2018-11-12:
It would be nice if we can get a log and even better a full .ts recording of a biss2 transmission.

PAKO34, 2018-11-17:
**hi, i am on travel and i dont have 26e signal sorry i cant help you... but % 100 they r using biss2

because vplug cant open it and oscam emu,

if you make biss.mdl for vplug module , vplug ,mdplug,nwemu can open it?**

oscam-emu, 2018-11-17:
No they are not. I'll wait to prove me wrong.

manal3, 2018-12-31:
Comentando

manal3, 2018-12-31:
https://www.mediafire.com/file/0ji2wge77zu5z1l/26E-11919H-12341H-20181219.zip/file ts 26E

oscam-emu, 2019-01-01:
I didn't ask for a recording of the channels mentioned above. I asked a recording of BISS2 encrypted channels and these are NOT BISS2 encrypted. They don't send PMT and they don't send CAT.

manal3, 2019-01-02:
Hi last news about BeautQ WORKS WITH POWER VU VERSIN Emu 3.8

oscam-emu, 2019-01-02:
Which confirms they are not using biss2

manal3, 2019-01-02:
link for files generated by powervu module
BeOutQ
BeOutQ-11919-H
BeOutQ-11939-H

https://www28.zippyshare.com/v/Kzxnmuy8/file.html

PAKO34, 2019-01-02:
More info about BeOutQ

BeOutQ is another `unconventional` DVB-system. They encrypt audio and video data using the AES algo and also they don't transmit any PMT & PAT data and because of that, DVB-players can't detect them! They are invisible to them.
They transmit their channels info and related AES keys of them, using the TOT PID(PID=20 decimal) of 12341H transponder. This transponder doesn't contain any viewable channels. They just fill the dynamic keys and dynamic channels info here. These dynamic info are also encrypted using 2 AES keys, which should be extracted from its official STB's.
After finding (decrypting) channels data (just by tuning to 12341H), you can see its final output in the new TOT tab of the PowerVU Emu, something like the following data:
Frequency: 11919
Polarization: H
Channel 0: SID=1825 ($0721), Video-PID=4839 ($12E7), Audio-PID1=4841 ($12E9), Audio-PID2=4842 ($12EA)
Channel 1: SID=1826 ($0722), Video-PID=2275 ($08E3), Audio-PID1=2277 ($08E5), Audio-PID2=2278 ($08E6)
Channel 2: SID=1827 ($0723), Video-PID=737 ($02E1), Audio-PID1=739 ($02E3), Audio-PID2=740 ($02E4)
Channel 3: SID=1828 ($0724), Video-PID=6377 ($18E9), Audio-PID1=6379 ($18EB), Audio-PID2=6380 ($18EC)
Channel 4: SID=1829 ($0725), Video-PID=3813 ($0EE5), Audio-PID1=3815 ($0EE7), Audio-PID2=3816 ($0EE8)
Channel 5: SID=1831 ($0727), Video-PID=4326 ($10E6), Audio-PID1=4328 ($10E8), Audio-PID2=4329 ($10E9)
Channel 6: SID=1870 ($074E), Video-PID=5864 ($16E8), Audio-PID1=5866 ($16EA), Audio-PID2=5867 ($16EB)
Channel 7: SID=1871 ($074F), Video-PID=3301 ($0CE5), Audio-PID1=3303 ($0CE7), Audio-PID2=3304 ($0CE8)
Channel 8: SID=1872 ($0750), Video-PID=1763 ($06E3), Audio-PID1=1765 ($06E5), Audio-PID2=1766 ($06E6)
Channel 9: SID=1830 ($0726), Video-PID=5351 ($14E7), Audio-PID1=5353 ($14E9), Audio-PID2=5354 ($14EA)

As you can see, they can change the video and audio PID's, dynamically, but ... usually they don't. So you should create these 10 channels `manually` in your DVB-player (such as `File->Define/Load manual channels` menu option in TSReader). Because they don't transmit any PMT & PAT data and in this case, your DVB-players doesn't know how to create and assign them automatically.

This time these viewable channels are located in 11919H transponder. So first you should tune to 12341H to find the keys and channels info and then you should tune to 11919H to watch them!

Tips: 
- 12341H doesn't contain any PAT or PMT data. Which means your DVB-player can't create or assign any specific channel to it automatically. You should create a dummy channel `manually` here to be able to find its keys! Here PIDs are not important and just enter 39,41,42 for them or anything else for this new channel.
- Every time you tune to 12341H, PowerVU Emu will create a new and fresh `BeoutQ-11919-H.tmc` file for you. You can find it in the `plugins` dir.

System requirements: 
These channels transmit HD(1920 * 1080) HEVC, H.265 videos and AAC audios. So your system should have proper codecs (for x265) and enough power to play them. 
Also if your DVB-player allows to define the `stream_type`s, stream_type of HEVC is 36 (decimal) and stream_type of AAC is 15 (decimal). If you don't define them, VLC will not able to play these channels, but KMPlayer with its internal `LAVFilters` works fine (plays x265 very smoothly). The provided `BeoutQ-11919-H.tmc` file, has these info.

About the new BeOutQ's update:
It seems there's no change in the algo, but they switch every few minutes the TP from where they send the keys. So the emu should work `sometimes`!

oscam-emu, 2019-01-02:
Is this a joke? Are you going to copy the entire forum here??????

These channels don't use the BISS2 encryption.

manal3, 2019-01-03:
hello bro what channels work with (AES) emu powervu 3.8 is there any way to
write this in linux receivers?

PAKO34, 2019-01-03:
If  i know everything, why i do  reporting here?? I make it and coded all sources! @oscam-emu

manal3, 2019-01-03:
I hope that you will help me in this

oscam-emu, 2019-04-08:
Added with https://github.com/oscam-emu/oscam-patched/commit/770a5f7ea7b659333d84cb7122147c474227d59d

closed 2019-04-08 by oscam-emu

PAKO34, 2019-04-08:
**I did tell you nauti.

http://www.streamboard.tv/oscam/changeset/11509**