oscam-emu/oscam-patched-old #10: Slow channelchange on AFN powervu

stemkow, 2018-11-07:
With oscam-emu Emu r764 - Add PowerVu hash mode 4 and oscam  http://www.streamboard.tv/svn/oscam@11392 channelchange is very slow (10 seconds) on afn powervu.

I have tracked the problem down to module-dvbapi.c line 4454. Channel changing speed is greatly improved  when removing " || pvu_skip". I am not sure if this is a good fix and it may cause problems on other powervu channels. However, it works for me and it helps pinpoint the problem

oscam-emu, 2018-12-18:
@stemkow is this issue still present?

Can you provide new logs with latest emu (no need to use cacheex for afn now) and with debug level 2+128.

stemkow, 2018-12-30:
I performed an experiment with oscam 11438 and oscam emu 9bc5fdf Emu r782.
Channel change os about 6-8 seconds. With pvu_skip commented out, it is about 2 seconds.

Logs are attached
[oscam-da8abc255-oscamemu-9bc5fdf-7seconds.log](https://github.com/oscam-emu/oscam-patched/files/2717205/oscam-da8abc255-oscamemu-9bc5fdf-7seconds.log)
[oscam-da8abc255-oscamemu-9bc5fdf-2seconds_nopvu_skip.log](https://github.com/oscam-emu/oscam-patched/files/2717206/oscam-da8abc255-oscamemu-9bc5fdf-2seconds_nopvu_skip.log)

closed 2024-04-14 by nautilus7