oscam-emu/oscam-patched-old #13: audio not working on some channels

PeterMaxu, 2018-11-07:
audio not working on some channels without streamrelay on vti image

OscamSatUser, 2018-12-15:
I am having the same issue.  It does it on all my my enigma STBs.  I can record the channel  directly on a .TS file, and it wont have audio too.. So i believe it to be in OSCAM some where.  This only happens with PowerVu channels.   I can send you any logs you need.  Its been driving me nuts for years.

OscamSatUser, 2018-12-15:
I would like to also add that there is a particular transponder mux where all the video / audio format is the same across the 10+ channels.  Ie all the same video/audio format.  There will be audio on half of them and no audio on the other half.

OscamSatUser, 2018-12-15:
2018/12/15 10:48:16 28C61199 c   (dvbapi) Demuxer 0 stop descrambling program number 00D4 (Program 212)
2018/12/15 10:48:16 28C61199 c   (dvbapi) Demuxer 0 ecmpid 0 CAID: 0E00 ECM_PID: 177D PROVID: 000000 
2018/12/15 10:48:16 28C61199 c   (dvbapi) Demuxer 0 found 1 ECMpids and 3 STREAMpids in caPMT
2018/12/15 10:48:16 28C61199 c   (dvbapi) Demuxer 0 found channel in cache and matching prio -> start descrambling ecmpid 0 
2018/12/15 10:48:16 28C61199 c   (dvbapi) Demuxer 0 trying to descramble PID 0 CAID 0E00 PROVID 000000 ECMPID 177D ANY CHID PMTPID 07E8 VPID 07E7
2018/12/15 10:48:16 28C61199 c      (ecm) linuxsatsupport (0E00@000000/0000/00D5/40:0892774BCAD2B5643146DA930850BC36): found (2 ms) by emulator - Program 213
2018/12/15 10:48:16 28C61199 c   (dvbapi) Demuxer 0 found 1 ECMpids and 3 STREAMpids in PMT
2018/12/15 10:48:17 28C61199 c      (ecm) linuxsatsupport (0E00@000000/0000/00D5/40:ABC313727E3B6F5809916CB274962414): found (1 ms) by emulator - Program 213
2018/12/15 10:48:17 28C61199 c      (ecm) linuxsatsupport (0E00@000000/0000/00D5/40:8F176D4829F04BFB37B435412C49B1AA): found (1 ms) by emulator - Program 213
2018/12/15 10:48:18 28C61199 c      (ecm) linuxsatsupport (0E00@000000/0000/00D5/40:6C4148E91F9EAEC275DF5D2F91B018A4): found (2 ms) by emulator - Program 213

OscamSatUser, 2018-12-15:
I also did another test. I turned on stream relay, and i have audio and no audio on the exact same channels as with stream relay off.

131w 3860H30000

Let me figure out how to change the debug level

OscamSatUser, 2018-12-15:
Okay, here is the log.  SID 221 has audio, SID 213 has no audio.  This log is with stream relay still turned on.  

[oscam.log](https://github.com/oscam-emu/oscam-patched/files/2682861/oscam.log)

oscam-emu, 2018-12-15:
First of all you need to update your oscam-emu to a more recent version. I believe you have a very old one (oscam is 11420, emu I can't tell, but I guess around 8 months old).

Does one of your boxes support powervu without stream relay? I would prefer a log without out stream relay at the moment. An additional log with stream relay won't hurt though.

So, update oscam-emu and create to logs with debug 2, with and without stream-relay from a box that supports powervu directly. Thanks.

OscamSatUser, 2018-12-15:
Okay, ill post back in about 5 minutes.

OscamSatUser, 2018-12-15:
Okay, had to add the skip cw feature to get it working again as per the directions, so it took me a few extra minutes.. This is the latest oscam back, with stream relay disabled.

SID 214 audio works
SID 213 NO audio

[oscam.log](https://github.com/oscam-emu/oscam-patched/files/2682948/oscam.log)

OscamSatUser, 2018-12-15:
Sorry, i read back and missed that you asked for with stream relay on too..  Here is that.
SID 221 audio works
SID 213 NO audio

[streamrelay.log](https://github.com/oscam-emu/oscam-patched/files/2683019/streamrelay.log)

lorba7068, 2018-12-17:
Hi fulllog of channels without audio .
[a3ntcs.txt](https://github.com/oscam-emu/oscam-patched/files/2684146/a3ntcs.txt)
[holatv.txt](https://github.com/oscam-emu/oscam-patched/files/2684147/holatv.txt)
[vmov.txt](https://github.com/oscam-emu/oscam-patched/files/2684148/vmov.txt)

121w alls

jlmariense, 2018-12-17:
> @PeterMaxu <https://github.com/PeterMaxu> @jlmariense
> <https://github.com/jlmariense> @lorba7068 <https://github.com/lorba7068>
> @OscamSatUser <https://github.com/OscamSatUser> please try with today's
> emu r780 and report back.
>
> It may fix these issues.

Please add ts delay in newcam or ccamd for star one satellite ku band

oscam-emu, 2018-12-17:
I don't know what "ts delay" is and we don't support oscam in general.

Also this has nothing to do with the present issue.

OscamSatUser, 2018-12-17:
Okay, you are on the right track.  This particular mux now has half as many no audio channels.  Still sid 213 is one that has no audio.  I am sending you another log of  with SID 212 and 213 both of which still have no audio.  I have no audio on 4 remaining channels within this mux.

OscamSatUser, 2018-12-17:
I am also gonna send you a log of a channel that wasn't working with audio and that now is.  
This log as SID 213 which still isn't working and SID 201 which was one that wasn't working and now is.

[213-201.log](https://github.com/oscam-emu/oscam-patched/files/2686359/213-201.log)

OscamSatUser, 2018-12-17:
Also, since we are making progress, i remember another mux that had some missing audio channels.  It turns out one of them is a MPG audio channel.  This log has both channels that have no audio, one is MPG and one is AC3.

This mux is 125w3960

[AC3-MPG.log](https://github.com/oscam-emu/oscam-patched/files/2686416/AC3-MPG.log)

budiarno, 2018-12-17:
Everything is just working fine here:

[image]

MohitSK, 2018-12-17:
@budiarno 
Thanks Dear, for checking the issue on your side
Will you please share the detail of receiver you are using and OSCAM configuration file with SoftCam.key file. 
It looks I messed something wrong in my OSCAM.

budiarno, 2018-12-17:
I use Dinobot4K. Config file as usual, as the OSCam-Emu wiki explain.

BTW, the latest patch even makes powervu via CCcam protocol possible to decrypt the audio properly on AFN and SPN, and the other channels which were in need of CacheEX 1.

MohitSK, 2018-12-17:
@budiarno
Thanks, Please share the key files if you can.
Regards

oscam-emu, 2018-12-17:
@OscamSatUser your logs are no good. You forgot to enable the dvbapi debugging level (128). Also I want the log to contain the pmt parsing as done by oscam-emu. To make sure you have this included, the easiest way is to set the initial logging and restart oscam-emu.

So, to make a proper report and log, do these:

1. Tell what is you box (model, enigma2 or not, etc).

2. Tune to the problematic channel you want to log.

3. Tell if the non working audio is indeed listed (is detected) in your box's audio selection menu.

4. Make sure you have stream relay disabled. At this point I am interested in problem with dvb-api. Stream relay problems will be taken care of at another time.

5. Enable logging in oscam-emu with debug level 130 (2+128). Because we want to restart oscam-emu, we want the _initial debug level_ to be set to 130. You can do this via the webif, or by adding `initial_debuglevel = 130` in the oscam.conf file under the global section. It should look similar to this:
```
[global]
logfile                       = /usr/keys/oscam.log
initial_debuglevel            = 130
maxlogsize                    = 1000
waitforcards_extra_delay      = 100
preferlocalcards              = 1
readerrestartseconds          = 3
```
6. Now restart oscam-emu. Log for about 10 seconds for channels that send ecm every 1 second, or for about 60 seconds for channels sending ecms every 10 seconds. The log does not need to be very long.

7. Finally, providing a small (200-300 MB) recording (.ts file) might help.

If you do all these steps correctly, your log should look like this: [demo-oscam.log](https://github.com/oscam-emu/oscam-patched/files/2687222/demo-oscam.log)

OscamSatUser, 2018-12-17:
Okay, here is the first channel.
DM920 4.3.1r7
Dream-Elite 6.0 
Enigma2

No Audio, SID 213.
Shows audio PID of 2022

[213.log](https://github.com/oscam-emu/oscam-patched/files/2687442/213.log)
[213.zip](https://github.com/oscam-emu/oscam-patched/files/2687457/213.zip)

OscamSatUser, 2018-12-17:
No Audio SID 212
Shows Audio PID of 2026 detected within ServiceInfo

[212.log](https://github.com/oscam-emu/oscam-patched/files/2687496/212.log)
http://www.filedropper.com/212_1

OscamSatUser, 2018-12-17:
Let me know if those are good... If so, i can send the others that don't work.

oscam-emu, 2018-12-17:
Hi, they look ok.

So, on these 2 channels, the DTS track is not working, right? It should be a trivial fix.

OscamSatUser, 2018-12-17:
Okay, ill work on another recording with OSCAM off... And the two channels i sent you, both of them only report 1 audio track.

OscamSatUser, 2018-12-17:
Jeeze, i had to upload the file 3 times before i found a site that didn't give me an error.. Here you go.

https://ufile.io/yuujd

jlmariense, 2018-12-18:
Congratulations on the excellent work. Perfect Oscar. Fixed audio problem.

OscamSatUser, 2018-12-18:
Just to confirm, i can get these channels just fine with audio on my Technomate TM5402m3  Hopefully the guy your passing this off too will have a thought or two... 

Okay, this is another transponder mux.  Shows as audio PID 501

[antenna.log](https://github.com/oscam-emu/oscam-patched/files/2688247/antenna.log)
https://ufile.io/cspec

oscam-emu, 2018-12-18:
Interesting... You mean the AC3 audio, or both AC3 and DTS?

What are the AC3 parameters the technomate reports? My guess is 2.0ch 48KHz, and what bitrate?

oscam-emu, 2018-12-18:
Same thing for Antenna channel. DTS audio is non existent, and the other one (mpeg audio this time) is descrambled fine from the emu.

I'll need a small decrypted sample of Antenna to forward for investigation, like the previous ones.
Does this also play fine on your technomate?

oscam-emu, 2018-12-18:
Good to hear!. You're welcome!

OscamSatUser, 2018-12-18:
Yes, Antenna plays just fine on the Technomate.  It shows in the stats at DVB-S H.264 MPEG Audio VPID 510 APid 501  - If i click the Audio button, i only see one audio track on the technomate.  I cannot get bitrate information from the Technomate.  But maybe the decrypted .ts from the technomate will answer that question.

The audio on the .ts file is Mpeg2 Stereo 192kbps and plays in AVIdemux
Ill record two Decrypted .TS  One on the technomate and one on the DM920.

https://ufile.io/xjy5e
https://ufile.io/9hiji

OscamSatUser, 2018-12-18:
SID 213 on the original mux, i recorded on the technomate.  AVIdemux plays it and shows one audio channel as AC3 Stereo 192kbps.

oscam-emu, 2018-12-18:
I see... Same here for the antenna samples. The technomate one is playing with audio in vlc and mpc.

So, I need another sample of the 212 channel, but this time recorded with the techomate.

OscamSatUser, 2018-12-18:
Also, none of the audios on the NO audio channels will play audio when i stream it to VLC from the DM920 webif.

Okay, here is 213 technomate and DM920 records

I created 3 .ts files.
Technomate Decrypted
DM920 Decrypted
DM920 NO Oscam -Unencrypted

https://ufile.io/4dvc9

OscamSatUser, 2018-12-18:
and here is 212 

3 .ts files
technomate decrypted
DM920 Decrypted
DM920 NO Oscam -Unencrypted

https://ufile.io/m6tfm

oscam-emu, 2018-12-18:
Nice, thanks.

Can you send me the (via email) the ecm key that is used for the 212 file decryption?
And also send the ecm key that was used yesterday for the antenna file recording.

I want to decrypt the files offline, so I can eliminate 100% the possibility of a descrambling problem with emu.

OscamSatUser, 2018-12-18:
I can also confirm that the previous fix you made that you released has fixed a good number of channels.  I also confirm many of the channels on 121w are working good now.  Fight Network was one that had no audio and now has audio.  I am so glad to see progress on this, as these bugs have been here for years!

Yes.. How do i email you?

MohitSK, 2018-12-18:
@oscam-emu 
Dear as per your given instruction I am sharing the oscam log of 
two PowerVu channel (IntelSat 20, 68.5)
1. SONY = Audio - OK, Video - OK 
2. Discovery Jeet (ID India)= Audio - NO, Video - OK

Channels from Sony package working properly and Channels from Discovery Package Video working only.

Here by I am sharing the log file as well '.ts' recorded file and required details as per your given instruction.

1. I am using the cygwin cross-compiled version of 'oscam11441-emu780' on Windows 7.
2. Yes, non working audio is indeed listed (is detected) in my box's audio selection menu.
3. I connected the client receiver(gx6605s, freesat V7, DM 800) to my OSCAM server using Newcamd to view/decrypt channel.

I tried to play the .ts stream of problematic channel Discovery jeet in VLC, AVIdemux, MPC same result-no audio
only video. But Sony channel plays fine.

So, please help and have a look.

https://ufile.io/vkiwv

Regards

MohitSK

oscam-emu, 2018-12-18:
@MohitSK your issue is very simple....

You are using the newcamd protocol, which does not support the extended CW needed by PowerVu. Only the dvbapi supports the extended CW. When NOT using the extended CW, you are limited to only 1 CW per channel.

For the Sony channel the video and the audio streams use the same CW, so they can be descrambled correctly with only 1 CW.

For the other channel, the video and the audio CWs are different, so the audio is not descrambled.

OscamSatUser, 2018-12-19:
Okay, got another Mux with many no audios.  I'm uploading one of them.

103w 4020 v 30000 
Bandamax SD

https://ufile.io/1p902
[oscam.log](https://github.com/oscam-emu/oscam-patched/files/2695403/oscam.log)

oscam-emu, 2018-12-25:
@OscamSatUser problem identified! It's the oscam and oscam-emu code in the dvbapi, not enigma2, to be blamed. Easy to fix, but thinking what's the best way to do it without the risk of breaking audio in other channels.

All uploaded channels suffered from the same issue, so all should be fixed now.

oscam-emu, 2018-12-27:
The fix has been committed https://github.com/oscam-emu/oscam-patched/commit/443fd2b828db29a9b86a74abdce26dd7c3023707 and included in r782.

Please check again ALL powervu channels you receive in your area and report whether:
1. Previously working channels are still working.
2. Previously non working channels work now.

asifagaria, 2019-01-02:
audio not working on channel im using oscam on tvheadend power vu

oscam-emu, 2019-01-02:
Are we supposed to guess which channel is not working for you?

Tell which channel is that and which satellite/transponder, post a live log like it is described above and maybe attach a full ts recording.

asifagaria, 2019-01-02:
im using oscam emu power vu on tvheadend channel are playing but audio not working

zinonino, 2019-01-10:
Yes, i confirm with last 3-4 changes,  when run one PowerVu channel from 1W (Discovery muxx) audio work, when start smae time second channel from same muxx, first runned channel lost audio, picture work, but lost audio, second work normal, run third channel and second lost audio. I talk for TVheadend and last oscam-11451+latest oscam-emu.

oscam-emu, 2019-01-11:
I can duplicate the problem in my enigma box also. I'll check it later today.

OscamSatUser, 2019-01-18:
> The fix has been committed 443fd2b and included in r782.

Oh, what a wonderful treat to come back from vacation and find this long standing annoyance is NOW fixed..  I tested every single channel, and every single one of them is now working!!!!!

 

Thank you so much, and I'm very thank ful you went that extra step to track it down....  As i said, that has drove me nuts for years!!  I just finally decided to get off my butt and try and make people aware of it. 

 

Thanks again!!!!

 

Sent: Wednesday, December 26, 2018 at 7:05 PM

closed 2019-01-19 by oscam-emu

colesbergkhn, 2019-09-09:
hi,,on 68.5east HBO+CNN dont have audio ,,video is ok,,any solution????bbc news n sky news on 68.5east r also not working on powervu on oscam,,,only sony working fine

oscam-emu, 2019-09-09:
Hi, post a log with debug 128 + 2 enabled, starting from before tuning to the channel. For each problematic channel make a new log. It should not be long. Just 4-5 seconds. Also post your config files (oscam.conf and oscam.server).

colesbergkhn, 2019-09-10:
> Hi, post a log with debug 128 + 2 enabled, starting from before tuning to the channel. For each problematic channel make a new log. Just 4-5 seconds is enough. Also post your config files (oscam.conf and oscam.server).

thanks for reply,,,i m new in this can u plz explain me how can i get the log with debug 128 +2?sorry for that

oscam-emu, 2019-09-10:
Go to webif and select config. Then set logging to something like that:

[image]

Restart oscam, and zap to every problematic channel, one after the other. Then go to the path you saved your log, get the file and post it here. If there is a ".prev" file there, post it as well.

colesbergkhn, 2019-09-13:
> What exactly is wrong? You can't create the log files, or you can't find them in your receiver?

i m struggling to get the log files,,,can i send u the channels recording?

oscam-emu, 2019-09-13:
I still don't understand what "get" refers to...
How did you created the recording? I might or might be able to see what the problem is (if there is any).

colesbergkhn, 2019-09-14:
look my box is android i m using android oscam on it,,,the other guy using octagon box said his side hbo 0n 68.5 east has sound but my side only video no sound,,, other guy told me i must record the channel n send it to me,,,its a 388mb file,, the file looks like this,,if u think u can do somthing on plz send me the email i ll send u the file,tnx
[image]

colesbergkhn, 2019-09-14:
ok i have sent u the mail on this email oscam.emu@gmail.com

oscam-emu, 2019-09-14:
Unfortunately, your sample has some errors and cannot be parsed properly, but audio on hbo and cnn is proper mpeg audio, so oscam should be able to descramble it just fine.

Without a log I can't do much.

Are you able to access oscam webif?
Have you set up oscam logging like I explained earlier?
Are you able to access your receivers files with ftp?

You can also try this: Set oscam to initial debug level 128 (like in the picture), tune to cnn or hbo, restart oscam and go to the webif --> livelog page and press "stop log". Then copy and paste here the log info.

colesbergkhn, 2019-09-16:
ok i ll try again n post tnx

colesbergkhn, 2019-09-17:
> Unfortunately, your sample has some errors and cannot be parsed properly, but audio on hbo and cnn is proper mpeg audio, so oscam should be able to descramble it just fine.
>
> Without a log I can't do [...]

I CANT ACCES TO OSCAM WEINF,,WHEN I USE 10.0.0.12:8888 OPEN ITS SHOWS ME THIS ERROR 
[image]

[6 more replies]