oscam-emu/oscam-patched-old #14: still no more than 100 pv emm are possible

PeterMaxu, 2018-11-08:
with actual version still no more than 100 pv emm are possible.

when sending emm manually to reader these emm are working, but when receiving emm over the air they are not recognized.

oscam-emu, 2018-11-08:
OK, let's take things from the beginning...

You have to provide information about your box and image you use, full oscam configs and full softcam.key file (the latter should be sent to us via email because of the sensitive data in it -- you can send all your files via email also).

I previously said that there is a 32 emm key restriction per group id. After looking the code again, this is not always the case. It's is so only in certain cases. 

The correct statement is that: **The maximum usable emm keys for a channel (service id) are 32.**

There is nothing related to the number "100" in the code, either for emm keys per group id or even for total emm keys.

Reminder: in order to make the auto update function to work you have to use groups in your powervu keys.

oscam-emu, 2018-11-12:
As explained in private, this is how it works. AU works with 32 emm keys per srvid. So, it might work better for some channels and worse for others.

I will add an option in the next release that will enable the user to configure the number of emm keys. It would be useful for users with lots of groups and emm keys in their softcam.key file, but may cause memory issues on some receivers.

For more detailed explanation on how the auto update algorithm works and best practices for powervu key configuration, refer to the wiki: https://github.com/oscam-emu/oscam-patched/wiki/PowerVU-config#note-3

closed 2018-11-12 by oscam-emu