oscam-emu/oscam-patched-old #16: what is this problem and how to solve it

egydoctor2010, 2018-11-20:
this problem occur when trying to open powervu channels
joeuser openpli5 with his drivers for spark 7162 stb

2018/11/20 16:08:17 20219815 c   (dvbapi) ERROR: Can't open device /dev/dvb/adapter0/demux0 (errno=24 Too many open files)
2018/11/20 16:08:18 20219815 c   (dvbapi) Demuxer 0 restarting decodingrequests after 2 ms with 1 enabled and 0 disabled ecmpids!
2018/11/20 16:08:18 20219815 c   (dvbapi) Demuxer 0 trying to descramble PID 0 CAID 0E00 PROVID 000000 ECMPID 1779 ANY CHID PMTPID 1920 VPID 1921

budiarno, 2018-11-21:
Try restart your box.

egydoctor2010, 2018-11-22:
after i restart oscam the problem disappear but i want to know the cause

oscam-emu, 2018-11-22:
Something crashed or was not terminated correctly...

egydoctor2010, 2018-12-29:
i find the solution 
the problem was in load balancer mode 
and the problem solved

egydoctor2010, 2018-12-29:
the working configs in in oscam.conf is

lb_mode                       = 0

thank you admin and you can mark the topic solved and close it again

closed 2019-01-02 by oscam-emu