oscam-emu/oscam-patched-old #18: Can u add compel encrypted systems in oscam emu?

PAKO34, 2018-12-06:
Can u add compel encrypted systems in oscam emu?

Compel Control is a network management system that provides comprehensive control of satellite-based broadcast networks for video, audio and data. Compel provides reliable control of broadcast networks in real-time and of the connected devices like Unity IRD or iPump Server).

`#ifndef _GNU_SOURCE
#define _GNU_SOURCE //needed for PTHREAD_MUTEX_RECURSIVE on some plattforms and maybe other things; do not remove
#endif
#include <stdio.h>
#include <stdlib.h>
#include <stdarg.h>
#include <string.h>
#include <stdint.h>
#include <stdbool.h>

#define SCT_LEN(sct) (3+((sct[1]&0x0f)<<8)+sct[2])

#define EMU_STREAM_SERVER_MAX_CONNECTIONS 8
#define EMU_STREAM_MAX_AUDIO_SUB_TRACKS 16

	typedef struct 
	{
		uint32_t pvu_des_ks[8][2][32];
		int8_t pvu_csa_used;
		void* pvu_csa_ks[8];
	} emu_stream_client_key_data;
	
	typedef struct 
	{
		int8_t have_pat_data;
		int8_t have_pmt_data;
		int8_t have_ecm_data;
		uint8_t data[1024+208];
		uint16_t data_pos;
		uint16_t srvid;
		uint16_t pmt_pid;
		uint16_t ecm_pid;
		uint16_t video_pid;
		uint16_t audio_pids[EMU_STREAM_MAX_AUDIO_SUB_TRACKS];
		uint8_t audio_pid_count;
		int16_t ecm_nb;
		emu_stream_client_key_data key;
	} emu_stream_client_data;

uint32_t b2i(int32_t n, const uint8_t *b)
{
	switch(n) {
	case 2:
		return  (b[0] <<  8) |  b[1];
	case 3:
		return  (b[0] << 16) | (b[1] <<  8) |  b[2];
	case 4:
		return ((b[0] << 24) | (b[1] << 16) | (b[2] <<8 ) | b[3]) & 0xffffffff;
	default:
		return 0;
	}
	return 0;
}

uint8_t *i2b_buf(int32_t n, uint32_t i, uint8_t *b)
{
	switch(n) {
	case 2:
		b[0] = (i>> 8) & 0xff;
		b[1] = (i    ) & 0xff;
		break;
	case 3:
		b[0] = (i>>16) & 0xff;
		b[1] = (i>> 8) & 0xff;
		b[2] = (i    ) & 0xff;
	case 4:
[... 663 lines cut ...]
						fwrite(stream_buf+startOffset+(i*packetSize), 1, packetSize, writefd);
					}
				}
					 
				remainingDataPos = startOffset+(packetCount*packetSize);
				remainingDataLength = bytesRead-remainingDataPos;
				
				if(remainingDataPos < remainingDataLength)
					{ memmove(stream_buf, stream_buf+remainingDataPos, remainingDataLength); }
				else
					{ memcpy(stream_buf, stream_buf+remainingDataPos, remainingDataLength); }
				
				bytesRead = remainingDataLength;
			}
		}
			
	}
	
	fclose(writefd);
	fclose(streamfd);
	free(data);
	free(stream_buf);
	return 0;
}
`

oscam-emu, 2018-12-13:
Both of these CAs don't use CSA or DES for ts scrambling, so they can be implemented only through stream relay. To do that, the stream relay module has to be modified a little bit, something we plan to do in the future. No ETA at the moment. Sorry.

PAKO34, 2018-12-13:
You know best but openbox receiver support latest micro hd receiver 
T2-MI
MIS
4K SUPPORT
SRG SUPPORT

CAN you add srg support master key in patch?

oscam-emu, 2018-12-13:
Sure, if you give me the code, I can add it...

PAKO34, 2018-12-13:
Can you exract sat card files ?

oscam-emu, 2019-01-02:
Compel added to emu r783.

oscam-emu, 2019-01-19:
Rosscrypt1 was added to emu r785.

ajameseduserv, 2019-02-08:
Hi nautilus7, I cannot get this working on Mutant HD/AX51 on 40E on 3675R. Image is OpenATV. I took a known working oscam config, created bouquet for stream relay and I am just getting blocks on screen.

`2019/02/06 18:03:13 00000000 s debug_level=64 
2019/02/06 18:03:13 00000000 s (main) System name = Linux 
2019/02/06 18:03:13 00000000 s (main) Host name = mutant51 
2019/02/06 18:03:13 00000000 s (main) Release = 4.10.12 
2019/02/06 18:03:13 00000000 s (main) Version = #2 SMP Sun Nov 19 03:43:01 CET 2017 
2019/02/06 18:03:13 00000000 s (main) Machine = armv7l 
2019/02/06 18:03:13 00000000 s (main) Stb model = hd51 
2019/02/06 18:03:13 00000000 s (main) Stb boxtype = hd51 
2019/02/06 18:03:13 00000000 s (emu) OSCam-Emu version 788 
2019/02/06 18:04:50 00000000        (emu) Stream client 0 connected
2019/02/06 18:04:50 00000000        (emu) Stream client 0 request / 1:0:1:4:1:1111:1918E5B:0:0:0
2019/02/06 18:04:50 4ABEFDC2 c   (dvbapi) Demuxer 0 ecmpid 0 CAID: A101 ECM_PID: 1555 PROVID: 000000 
2019/02/06 18:04:50 4ABEFDC2 c   (dvbapi) Demuxer 0 found 1 ECMpids and 5 STREAMpids in caPMT
2019/02/06 18:04:50 4ABEFDC2 c   (dvbapi) Demuxer 0 no suitable readers found that can be used for decoding!
2019/02/06 18:04:50 4ABEFDC2 c   (dvbapi) Demuxer 0 found 1 ECMpids and 5 STREAMpids in PMT
2019/02/06 18:04:51 4ABEFDC2 c   (dvbapi) Demuxer 1 ecmpid 0 CAID: A101 ECM_PID: 1555 PROVID: 000000 
2019/02/06 18:04:51 4ABEFDC2 c   (dvbapi) Demuxer 1 found 1 ECMpids and 5 STREAMpids in caPMT
2019/02/06 18:04:51 4ABEFDC2 c   (dvbapi) Demuxer 1 no suitable readers found that can be used for decoding!
2019/02/06 18:04:53 4ABEFDC2 c   (dvbapi) Demuxer 0 no enabled matching ecmpids -> decoding is waiting for matching readers!
2019/02/06 18:04:55 4ABEFDC2 c   (dvbapi) Demuxer 1 no enabled matching ecmpids -> decoding is waiting for matching readers!
2019/02/06 18:04:58 4ABEFDC2 c   (dvbapi) Demuxer 0 no enabled matching ecmpids -> decoding is waiting for matching readers!
2019/02/06 18:05:00 4ABEFDC2 c   (dvbapi) Demuxer 1 no enabled matching ecmpids -> decoding is waiting for matching readers!`

Stream relay is obviously relaying stream, else I'd get no picture at all but no decoding.

`stream_relay_enabled = 1
stream_emm_enabled = 1
stream_relay_ctab = A101
stream_ecm_delay = 700`

oscam-emu, 2019-02-08:
Post a log with debug level 4, which is for stream relay.

captainjack64, 2019-02-08:
[Here's the link to log](https://www.dropbox.com/s/n9lx5sbrpj41mic/oscam-r787.log?dl=0)

[Link to .ts from 40E](https://www.dropbox.com/s/2v6vz1c0hs934fr/20190208%201649%20-%201%20KanalR%2BMayak%20-%20instant%20record.ts?dl=0)

[Link to what it looks like on screen with stream relay](https://www.dropbox.com/s/kzszptnit20uxch/20190206_165139.jpg?dl=0)

oscam-emu, 2019-02-08:
Unfortunately, there is a problem with boxes like yours that use the bcm7251s SoC (others are Octagon sf4008 and GI et11000, and perhaps more).

Audio and video pids in all rosscrypt channels I've seen are partially scrambled, meaning that some packets are marked as "scrambled" and some others marked as "not scrambled". This confuses the box (box's driver) which thinks all packets are "not scrambled".

Marking all packets as  not scrambled, makes it impossible to work in oscam-emu and any other application. All recordings made form these 3 receivers are not working (cannot be descrambled) offline with my "poc" program as well.

At this point there is nothing you can do... Sorry.

captainjack64, 2019-02-09:
Ah that's a shame. And I guess there's no way to manually mark them as scrambled either? All a driver issue....

oscam-emu, 2019-02-09:
No, there is not, because not all packets are originally scrambled. In these channels there's a mix of scrambled and not scrambled packets.

All you can do is to tell the openatv guys about this issue, so they ask the manufacturer to fix it.

loxel, 2019-02-09:
Hi
Is there a chance to improve the compel system ? It works very unstable, the picture is very unstable.
Greetings

oscam-emu, 2019-02-09:
Yes, we're working on it.

oscam-emu, 2019-02-16:
Improved Compel code is included in r789.

closed 2019-02-16 by oscam-emu