oscam-emu/oscam-patched-old #19: namespace hash calculation

momi133, 2019-01-10:
namespace hash is calculated by the enigma2 namespace (orbital position, frequency, polarization) and the service id of the channel.
1. how does oscam calculate it? please explain in detail. please! 
2. I'm writing a plugin for enigma2 to open biss channels. is it impossible to get namespace hash from oscam to assign a variable in python code of plugin?
thanks a lot.

oscam-emu, 2019-01-10:
See https://github.com/oscam-emu/oscam-patched/blob/master/module-emulator-biss.c#L393

and https://github.com/oscam-emu/oscam-patched/blob/master/module-dvbapi.c#L2216

oscam-emu, 2019-01-11:
I assume you know how enigma2 creates the namespace (what info is included and how it is formed).

There are 2 namespace hashes, because there are 2 namespaces in enigma.
When the tsid and the onid are deemed valid by enigma, the frequency info is stripped off and the namespace contains only the orbital position. This is described here: https://github.com/oscam-emu/oscam-patched/wiki/BISS-config#note-6

So in case of "full" namespace (with frequency), we use only the namespace and the service id in the hash. We place them like this: [srvid][namespace], so we get a 2+4=6 byte long sequence. Then we create a crc32 hash of these (caid is 0x2600 for biss and used as initial value in our crc32).

In case of stripped namespace, we use the tsid and onid as well in our hash, so it is still unique. So we place then like this: [srvid][tsid][onid][namespace] and we get a 2+2+2+4=10 byte sequence, from which we calculate the crc32 hash again.

Another thing you should know is that our namespace is not exactly the same as enigma's. In our namespace, we clear the 4 most significant bits (in case of DVB-T or DVB-C they are 0xF or 0xE respectively - for DVB-S they are 0) and we replace them with 0xA. This is our flag, so the emu can tell that this is the namespace and not another pid like audio etc. This is done here: https://github.com/oscam-emu/oscam-patched/blob/master/module-dvbapi.c#L2216

So the ecm you asked is a byte sequence, part of which is the srvid, namespace, tsid, onid. The b2i and i2b functions mean buffer-to-integer and integer-to-buffer and are used to take some bytes from a buffer, e.g. take the namespace from the ecm, or vice versa, e.g. place the srvid to a specific position in the the ecm.

momi133, 2019-01-11:
srvid = 0001
FULL namespace = 00450ACB
stripped namespace = 0ACB
TSID = 04AB
ONID = 0EB5

"full" namespace CASE =  [srvid][namespace] = 000100450ACB >>>is it right?
stripped namespace CASE=  [srvid][tsid][onid][namespace] = 000104AB0EB50ACB >>>is it right?

crc32 function:  is it right?
[image]

please get a valid web address about crc32 algorithm and it's function 

>  our namespace is not exactly the same as enigma's. we clear the 4 most significant bits

Is your namespace, crc32 output or input? it seems crc32 output.

thanks for replying and responsibility

oscam-emu, 2019-01-13:
Stripped namespace is 4 byte long, as full namespace. 00460000 is the correct. Also you forgot the flag we put in the beginning. After flagging full namespace is A0460ACB and stripped is A0460000. It's a simple "or", `ens = ens | 0xA0000000`

crc32 algorith in general, takes a byte string as input and calculates a 32 bit (4 byte) output string, our namespace hash. In python it looks like this:

```
table = array('L')

for byte in range(256):
    crc = 0
    for bit in range(8):
        if (byte ^ crc) & 1:
            crc = (crc >> 1) ^ 0xEDB88320
        else:
            crc >>= 1
        byte >>= 1
    table.append(crc)

def crc32(string):

    value = 0x2600 ^ 0xffffffffL
    for char in string:
        value = table[(ord(char) ^ value) & 0xff] ^ (value >> 8)
    return value ^ 0xffffffffL
```
And you use it like this:

```
    def get_namespace_hash(self):

        ref = self.session.nav.getCurrentlyPlayingServiceReference()
        sid = ref.getUnsignedData(1)
        tsid = ref.getUnsignedData(2)
        onid = ref.getUnsignedData(3)
        ens = ref.getUnsignedData(4)

        # check if we have stripped or full namespace
        if ens & 0xFFFF == 0:
            data = "%04X%04X%04X%08X" % (sid, tsid, onid, ens | 0xA0000000)
        else:
            data = "%04X%08X" % (sid, ens | 0xA0000000)
        return crc32(binascii.unhexlify(data))
```

momi133, 2019-01-13:
thank a lot! my problem is solved!
I am ashamed because I waste your time!
I admire and respect you because you are noble, unselfish and morally good.
I have a Criticism about oscam-emu(how read biss keys from softcam.key) that I will propose you tomorrow!
thanks for replying and responsibility

closed 2019-01-14 by oscam-emu

sentarokun, 2021-04-19:
Hi @oscam-emu , would you help me on this? I would like to understand how the oscam namespace hash is calculated. Can you follow this example?
data: E2 namespace:  dec 4598562 (including freq) 
hex namespace: => 0x00462B22
modified namespace  0xA0462B22 (adding oscam identifier)
[srvid][oscam namespace]: 0001A0462B22

How to go further? How do you calc the CRC32 on the above result.

Thak you so much.

oscam-emu, 2021-04-20:
The crc32 function is defined here: https://github.com/oscam-emu/oscam-patched/blob/master/oscam-string.c#L714

A python version that produces the same hash is:

```
table = array('L')

for byte in range(256):
    crc = 0
    for bit in range(8):
        if (byte ^ crc) & 1:
            crc = (crc >> 1) ^ 0xEDB88320
        else:
            crc >>= 1
        byte >>= 1
    table.append(crc)

def crc32(caid, string):

    value = caid ^ 0xffffffffL
    for char in string:
        value = table[(ord(char) ^ value) & 0xff] ^ (value >> 8)
    return value ^ 0xffffffffL

```

sentarokun, 2021-04-21:
So I need to study some Python before attempting it. But please, in my example above, is it correct to set the string (in the crc32 func) to "0001A0462B22"? Is it the correct approach?

thx!!

oscam-emu, 2021-04-21:
Yes, read the comments in the code:

https://github.com/oscam-emu/oscam-patched/blob/master/module-emulator-biss.c#L421
https://github.com/oscam-emu/oscam-patched/blob/master/module-emulator-biss.c#L426

sentarokun, 2021-04-22:
Hi, thank you.
I tried the whole process till Python, but the result differs from the suggested hash in the livelog.
Could you please have a fast look at my steps?
1)   Getting the E2 namespace from the DreamboxEDIT (freq 12550V, TSID 0001, ONID 0000, SID 0008  ==> E2 namespace 0x0046b106 ) or by the box itself
2)  Changing the namespace by adding the 0xA identifier: ==> 0xA046B106
3)  Adding the SRVID (SID 0008) ==> 0008A046B106
4)  Running the python CRC32 function, by using: 
		-	CAID (as an integer or its hex representation 0x2600)
		-	ens (as a string, uppercase or lowercase? "0008A046B106", "0008a046b106")
5) As a result ==> a39cb7c7 (not matching with the suggested hashes from the livelog: E2F2E27D , E700CC3D)

The python code is attached.

Thanks!

[hash.py.txt](https://github.com/oscam-emu/oscam-patched/files/6360382/hash.py.txt)

oscam-emu, 2021-04-22:
0x0008A046B106 is correct. You need to convert the hex data to acsii string first:

data = 0x0008A046B106
string = binascii.unhexlify(data)
b = crc32(0x2600, string)

sentarokun, 2021-04-22:
Great! Thank you so much for your support!