oscam-emu/oscam-patched-old #20: [makefile] Fixed linking of empty SoftCam.Key file on aarch64 boxes

sattotal, 2019-01-14:
patch doesn't work on Cygwin based compilations
with patch I got below errors:

> ld: unknown option »-z« ld: please useoption --help for information
> objcopy: 'SoftCam.Key.o': No such file

oscam-emu, 2019-01-14:
Please test with latest commit.

sattotal, 2019-01-17:
Thanks for fast answer but the problem still exists because ld doesn't support the option -z  in the Makefile

oscam-emu, 2019-01-19:
Can't you upgrade/update to a version that supports the -z option?

https://stackoverflow.com/questions/29178445/what-is-the-z-option-for-in-this-gcc-compiler-command

sattotal, 2019-01-21:
for Cygwin I have updated to GNU ld (GNU Binutils) 2.31.1 (latest version) .
However there is no -z Option available
without "-z noexecstack" I can build without error

sattotal, 2019-01-29:
a possible workaround is using
#if defined(__CYGWIN__)
..
#else 

#endif

oscam-emu, 2019-02-01:
thanks will look into it, when I finished some other things I do.

oscam-emu, 2019-02-05:
The linking method was reverted to the old one, so everything should be back to normal.

A new config option WITH_SOFTCAM is introduced that allows linking or not of the SoftCam.Key file during compilation.

closed 2019-02-05 by oscam-emu

sattotal, 2019-02-05:
#2078 [osemu] New config option to control SoftCam.Key linking in binary 
Thanks for this well working fix ;)