oscam-emu/oscam-patched-old #26: constant CW file and constcw protocol problems!!!

PAKO34, 2019-02-03:
**hi,
oscam says " we have got reader and it can be read biss keys in constantcw file and protocol name is constcw " but they are protocol i think dont work with home sharing / biss sharing..

i did try constant.cw file sharing with hd mini receiver but channels dont open

normal emulator can open channels hd mini receiver

i dont never try local sat tuner in biss channels . for example, dm800 and oscam with biss channels,

oscam cant do it

can you help ?
regards**

PAKO34, 2019-02-03:
### **hola budiarno? how to understand this?**

oscam-emu, 2019-02-05:
@PAKO34, I don't understand what you're writing and what you're trying to do...

Emu only supports BISS, not constant cw. There is a dedicated module for constant cw in oscam.

oscam-emu, 2019-02-06:
Please don't spam...

closed 2019-02-06 by oscam-emu