oscam-emu/oscam-patched-old #28: make failed

Shporterator, 2019-02-07:
CC      oscam-time.c
CC      oscam-work.c
CC      oscam.c
CONF    build/aarch64-linux-gnu/config.c
LINK    Distribution/oscam-1.20_svn11494-788-aarch64-linux-gnu.debug
collect2: fatal error: ld terminated with signal 11 [Segmentation fault]
compilation terminated.
Makefile:437: recipe for target 'Distribution/oscam-1.20_svn11494-788-aarch64-linux-gnu.debug' failed
make[1]: *** [Distribution/oscam-1.20_svn11494-788-aarch64-linux-gnu.debug] Error 1
Makefile:406: recipe for target 'all' failed
make: *** [all] Error 2

Linux rockpro64 4.4.154-1128-rockchip-ayufan-g61b4b1151f9a #1 SMP Tue Dec 25 14:31:31 UTC 2018 aarch64 aarch64 aarch64 GNU/Linux

oscam-emu, 2019-02-07:
Disable SoftCam.Key linking by running `./config.sh --disable WITH_SOFTCAM` or use a more recent compiler/toolchain for aarch64.

fairbird, 2019-02-09:
Hi friend ...
Please Can you give the commits responsible for auto update softcams.key to PowerVU ?!
Because after update my files I think break the feature of Auto update >>

So please just show me only that commits responsible for auto update if you can ..
Thank you

closed 2019-02-11 by oscam-emu