oscam-emu/oscam-patched-old #32: Cartoon Network Channels of AsiaSat-7 Create Problem

athartanwri, 2019-02-20:
Hello Dear Experts,

I want to discuss a problem, I am using in Origional DreamBox DM520 receiver.

All Oscam, Ncam and other cams for PowerVu and other scrambled channels are working fine with me. 

But when I tune to AsiaSat-7 (105.6 East ) TP-3960 H Turner International Asia, Cartoon Network Channels. Then my other scrambled channels don't open unless I fully restart my receiver.

And when I use this same configuration in Amiko Alien Enigma2 receiver, these Cartoon Network Channels are working fine with other scrambled channels.

Kindly look into the matter and suggest the solution.

Thanking in Advance.

oscam-emu, 2019-02-20:
This is a known problem on some receivers that has to do with switching from the encryption algorithm DES (used by most powervu channels) back to the CSA algorithm (used by all other channels).

It is purely a driver issue of your box and has nothing to do with oscam-emu. You should ask your box manufacturer or image supplier to fix it.

dhwz, 2019-02-20:
It's not an issue as the hardware is behaving DVB compliant, DES is not a valid encryption for DVB.

dhwz, 2019-02-20:
Can you please reopen this issue. Thats in my opinion a bug inside of the powervu emu code.
See https://github.com/oscam-emu/oscam-patched/blob/master/module-emulator-powervu.c#L1841

cw_ex->algo_mode has to be CW_ALGO_MODE_CBC not CW_ALGO_MODE_ECB for CSA thats why the driver is ignoring this command.

this would be correct:
						cw_ex->mode = CW_MODE_MULTIPLE_CW;
+
+						if (csaUsed)
+						{
+							cw_ex->algo = CW_ALGO_CSA;
+							cw_ex->algo_mode = CW_ALGO_MODE_CBC;
+						}
+						else
+						{
+							cw_ex->algo = CW_ALGO_DES;
+							cw_ex->algo_mode = CW_ALGO_MODE_ECB;
+						}

oscam-emu, 2019-02-21:
Interesting... But this alone will not explain why we cannot switch back to CSA after watching a DES channel, I think. Perhaps there is be also another bug in https://github.com/oscam-emu/oscam-patched/blob/master/module-dvbapi.c#L7288 which it should be like in line 7292 instead...

But If I make the change you propose and the change I propose, then I get a error message in my box:

`
2019/02/21 02:46:23 7060F235 c   (dvbapi) ERROR: ioctl(CA_SET_DESCR_MODE): Invalid argument
`

I get this in all CSA channels (not only in those using CSA), but everything is working like it should. I can watch both CSA and DES channels without a problem.

I have an idea what might be wrong... The CSA algorithm does not have different modes, so it should not really matter if we set CBC or ECB mode. But the spec say that the algorithm works similar to CBC mode...

So most manufacturers seem to ignore this wrong setting and have no problem, others just throw an error but keep working, and others like Dreambox don't work at all...

I have compiled a binary [oscam-emu-test.zip](https://github.com/oscam-emu/oscam-patched/files/2887325/oscam-emu-test.zip) with the changes I mention above. @athartanwri, can you please check it it runs (I am not sure I compiled with the correct toolchain) and test if CSA channels (normal ones and Powervu ones) and DES channels work now without the need to restart every time.

dhwz, 2019-02-21:
You are right there might be more places in the source where it has to be changed, I didn't check the whole source right now.

I think that some other manufacturers maybe set it automatically to the right value, ECB and CBC are working different. So they just ignore or throw out an error but switching it to CBC if you set CW_ALGO_CSA and maybe also in the other direction if you set CW_ALGO_DES and set it to ECB.

As said on newer Dreambox HW it has to be set properly otherwise it won't work as the driver will ignore any invalid setting.

athartanwri, 2019-02-21:
Hello Dear thanks 😊 you so much.

Now these channels are not creating any problem and working smoothly and fine with PowerVU and other scrambled Channels.

oscam-emu, 2019-02-21:
@dhwz line 7292 is only used when you have an oscam compiled without extended cw support, which normally this means oscam svn. Line 7288 is used when you have oscam-emu and the channel is not powervu i.e. we do not have multiple cw's per channel.

Both lines do the same, like the lines above them, where the `dvbapi_write_cw()` is called. They prepare the CA descramblers in the box and write the cw's to them.

As you see the `dvbapi_write_cw()` call is also wrong for non powervu channels as well. In line 7288:

`
dvbapi_write_cw(i, er->cw, j, 0, er->cw_ex.algo, er->cw_ex.algo_mode, er->msgid);
`

we use the `er->cw_ex.algo` and `er->cw_ex.algo_mode` which are not set at all! The whole `er->cw_ex` is not used when we have a normal channel and its members `algo` and `algo_mode` are initialized to 0 by the compiler, which equals to CSA and ECB. Completely fucked up code there... Same in the case of powervu channels, we again set CSA with wrong mode (ECB) as you pointed out...

So, if the Dreamboxes ignore wrong commands, they completely ignore the ECB setting for the CSA algorithm, and probably use the default which seems to be CBC. 

But when we use a DES powervu channel, the algo mode is set to ECB (correctly). Problem is that it cannot set back to CBC for CSA channels, because we ask them to do it with wrong commands, which they ignore...!

The algorithm mode is set using the `CA_SET_DESCR_MODE` command in line 6862 https://github.com/oscam-emu/oscam-patched/blob/master/module-dvbapi.c#L6862.

@dhwz Thanks for testing! Nice to hear it works for you! I'll have to do more testing and see how other boxes behave with this change. For example, I already found out the my edision only trows an error, but it still works fine. Other boxes might not work, so we have to see.

dhwz, 2019-02-21:
@oscam-emu 
Thx for clarification, but 7292 is still wrong also in svn right? Yes I know it's working because most (or all) are just ignoring it ;)

oscam-emu, 2019-02-21:
Yes it is the same (wrong)... And 7288 also. And it is like this since ever...

dhwz, 2019-02-21:
I think DM900, DM920, DM520, DM525 and DM7080, DM820 (or short said all OE2.5 boxes).

amiralishahrestani, 2019-02-21:
@oscam-emu
could you please atleast give a clue to compile oscam-emu our selves for oe2.5  ?
which lines should i change ?

oscam-emu, 2019-02-21:
There is no secret. I expained above:

in module-emulator-powervu.c change line 1841 to `cw_ex->algo_mode = CW_ALGO_MODE_CBC;`

in module-dvbapi.c change lines 7288 and 7292 to `dvbapi_write_cw(i, er->cw, j, 0, CA_ALGO_DVBCSA, CA_MODE_CBC, er->msgid);`

I am not releasing the fix yet, because I have to be sure that this doesn't break previously working receivers. It seems that all dreamboxes are ok now though.

budiarno, 2019-02-21:
I'll test on my dinobot4k, hope everything just fine with this patch.

budiarno, 2019-02-21:
Okay, so far so good. Nothing breaks, tested back and forth between DES and CSA. It just works as usual.

oscam-emu, 2019-02-21:
Nice!

Any error message like `(dvbapi) ERROR: ioctl(CA_SET_DESCR_MODE): Invalid argument` now or before?

budiarno, 2019-02-21:
It's clean, there's none of that error appears.

amiralishahrestani, 2019-02-21:
thanks oscam-emu dev and also dreambox developer  mr dhwz
work wonderfully on dm900-dm920-520

closed 2019-02-22 by oscam-emu

7687476874, 2020-11-22:
Hello there

I have a TBS 6903 TV card. I use OScam and NCam in Windows 10 operating system. PowerVU channels on Astra 4.9 satellite do not open with emu, the channel does not show anything but a black screen.Tandberg channels do not have anything, it runs How can I resolve channels opened with emu on Astra 4.9 satellite.
[image]
[image]
[image]
[image]
[image]

commit 0dc76fba (2019-02-22): [dvbapi + powervu] Properly set the descrambler mode for the CSA algorithm
When watching a DES powervu channel, switching back to CSA channels required a reboot on some receivers. Other more resilient receivers just threw an error(ioctl(CA_SET_DESCR_MODE): Invalid argument), but continued to work. This was described as a bug of these receivers, but in fact it was a bug of oscam-emu. The correct mode for the CSA algorithm is CBC and not ECB! Thanks to all who helped. This fixes #32
commit 94c001ee (2019-02-22): [dvbapi + powervu] Properly set the descrambler mode for CSA
When watching a DES powervu channel, switching back to CSA channels required a reboot on some receivers. Other more resilient receivers just threw an error(ioctl(CA_SET_DESCR_MODE): Invalid argument), but continued to work. This was described as a bug of these receivers, but in fact it was a bug of oscam-emu. The correct mode for the CSA algorithm is CBC and not ECB! Thanks to all who helped. This fixes #32