oscam-emu/oscam-patched-old #33: Gcam Opens some channels which Oscam / Ncam Don't Open

athartanwri, 2019-02-21:
Dear Experts;

I have to discuss that there are some channels on  **T.P 11510 H of 93.5 east** and **12688 H of 95 East** which are working fine on Gcam 1.5 and Gcam 1.9. I have tested them on both Spark 7162 and DreamBox DM520 on various Images.

While these same channels are not working with Oscam and Ncam. Although I have replaced all configuration files of Gcam 1.9 with Oscam as well as Ncam.

Kindly look into the matter and suggest any solution.

Thanking you in Advance.

oscam-emu, 2019-02-21:
Please post a log with debug level 2+128 and more info about these channels (encryption, etc.)

athartanwri, 2019-02-21:
**Log File of Oscam**

2019/02/21 18:15:13 00000000 s     (main) System name    = Linux
2019/02/21 18:15:13 00000000 s     (main) Host name      = dm520
2019/02/21 18:15:13 00000000 s     (main) Release        = 3.4-4.0-dm520
2019/02/21 18:15:13 00000000 s     (main) Version        = #1 SMP Fri Feb 9 11:33:35 UTC 2018
2019/02/21 18:15:13 00000000 s     (main) Machine        = mips
2019/02/21 18:15:13 00000000 s     (main) Stb model      = dm520
2019/02/21 18:15:13 00000000 s     (main) Stb vumodel    = vuultimo
2019/02/21 18:15:13 00000000 s     (main) Stb boxtype    = vuultimo
2019/02/21 18:15:13 00000000 s     (main) creating pidfile /tmp/.oscam/oscam.pid with pid 2062
2019/02/21 18:15:13 00000000 s      (emu) OSCam-Emu version 787
2019/02/21 18:15:13 00000000 s   (config) userdb reloaded: 1 accounts loaded, 0 expired, 0 disabled
2019/02/21 18:15:13 00000000 s     (main) signal handling initialized
2019/02/21 18:15:13 00000000 s   (config) 3 provid's loaded
2019/02/21 18:15:13 00000000 s   (config) 20 service-id's loaded in 0 ms
2019/02/21 18:15:13 00000000 s   (reader) cccam_server [cccam] creating thread for device city.4k7.me
2019/02/21 18:15:13 00000000 s   (reader) constant.cw [constcw] creating thread for device /etc/tuxbox/config/constant.cw
2019/02/21 18:15:13 00000000 s   (reader) emulator [emu] creating thread for device emulator
2019/02/21 18:15:13 00000000 s     (stat) loadbalancer: could not open /tmp/.oscam/stat for reading (errno=2 No such file or directory)
2019/02/21 18:15:13 00000000 s     (main) waiting for local card init
2019/02/21 18:15:13 4AB847A8 r      (emu) Stream key delayer initialized
2019/02/21 18:15:13 4AB847A8 r      (emu) Stream relay server initialized
2019/02/21 18:15:13 4AB847A8 r   (reader) emulator [emu] Reader initialized (device=emulator, detect=cd, mhz=357, cardmhz=357)
2019/02/21 18:15:13 14F59644 p  (constcw) Local reader: constant.cw (file: /etc/tuxbox/config/constant.cw)
2019/02/21 18:15:13 14F59644 p  (constcw) ERROR: Can't open /etc/tuxbox/config/constant.cw (errno=2 No such file or directory)
2019/02/21 18:15:13 14F59644 p   (reader) constant.cw [constcw] proxy initialized, server /etc/tuxbox/config/constant.cw:0
2019/02/21 18:15:13 00000000        (emu) INFO: FFDecsa parallel mode = 64
2019/02/21 18:15:13 0E6482F6 h    (webif) webif: decompressed 180767 bytes back into 446360 bytes
2019/02/21 18:15:13 0E6482F6 h    (webif) HTTP Server running. ip=0.0.0.0 port=8888
2019/02/21 18:15:13 53E8DB28 p   (reader) city.4k7.me: resolved ip=212.237.22.87
2019/02/21 18:15:13 53E8DB28 p   (reader) cccam_server [cccam] connecting to city.4k7.me:12000
2019/02/21 18:15:14 53E8DB28 p   (reader) cccam_server [cccam] proxy initialized, server city.4k7.me:12000
2019/02/21 18:15:14 4AB847A8 r   (reader) emulator [emu] card detected
2019/02/21 18:15:15 4AB847A8 r   (reader) emulator [emu] found card system emu
2019/02/21 18:15:15 4AB847A8 r   (reader) emulator [emu] THIS WAS A SUCCESSFUL START ATTEMPT No  1 out of max alloted of 1
2019/02/21 18:15:15 4AB847A8 r      (emu) Reading key file: /etc/tuxbox/config/SoftCam.Key
2019/02/21 18:15:15 4AB847A8 r      (emu) Total keys in memory: W:23 V:37 N:4 I:6 S:2 F:176 P:622 D:0 T:167
2019/02/21 18:15:16 00000000 s     (main) init for all local cards done
2019/02/21 18:15:16 00000000 s (emmcache) loading emmcache disabled since no reader is using it!
2019/02/21 18:15:16 00000000 s (emmcache) loading emmstats disabled since no reader is using it!
2019/02/21 18:15:16 00000000 s (anticasc) anti cascading disabled
2019/02/21 18:15:16 752CC975 c   (client) plain dvbapi-client granted (dvbapiau, au=auto (3 reader))
2019/02/21 18:15:16 752CC975 c   (dvbapi) dvbapi channelcache loaded from /etc/tuxbox/config/oscam.ccache
2019/02/21 18:15:16 752CC975 c   (dvbapi) Detected /dev/dvb/adapter0/demux0 Api: 0, userconfig boxtype: 1 maximum amount of possible filters is 32 (oscam limit is 32)
2019/02/21 18:15:18 752CC975 c   (dvbapi) Demuxer 0 ecmpid 0 CAID: 1712 ECM_PID: 172C PROVID: 000000 
2019/02/21 18:15:18 752CC975 c   (dvbapi) Demuxer 0 ecmpid 1 CAID: 0BC1 ECM_PID: 1501 PROVID: 000000 
2019/02/21 18:15:18 752CC975 c   (dvbapi) Demuxer 0 ecmpid 2 CAID: 0B00 ECM_PID: 0E60 PROVID: 000000 
2019/02/21 18:15:18 752CC975 c   (dvbapi) Demuxer 0 found 3 ECMpids and 8 STREAMpids in caPMT
2019/02/21 18:15:18 752CC975 c   (dvbapi) Demuxer 0 no suitable readers found that can be used for decoding!
2019/02/21 18:15:18 752CC975 c   (dvbapi) Demuxer 0 trying to descramble PID 2 CAID 0B00 PROVID 000000 ECMPID 0E60 ANY CHID PMTPID 0E5C VPID 0E5D
2019/02/21 18:15:18 752CC975 c   (dvbapi) Demuxer 0 found 3 ECMpids and 8 STREAMpids in PMT
2019/02/21 18:15:18 14F59644 p  (constcw) ERROR: Can't open /etc/tuxbox/config/constant.cw (errno=2 No such file or directory)
2019/02/21 18:15:18 752CC975 c   (dvbapi) ERROR: ioctl(CA_SET_DESCR_MODE): Invalid argument
2019/02/21 18:15:18 752CC975 c        (-) -- Skipped 1 duplicated log lines --
2019/02/21 18:15:18 752CC975 c      (ecm) dvbapiau (P: 0B00::81D6:0E60:0000 #ECM_L:CF #CW=905DB3A062819477F5A715B15E7303D4 HOP:01): found (231 ms) by cccam_server (P/2/2/2) - SONY MAX 2 Rs 1
2019/02/21 18:15:19 14F59644 p  (constcw) ERROR: Can't open /etc/tuxbox/config/constant.cw (errno=2 No such file or directory)
2019/02/21 18:15:20 752CC975 c   (dvbapi) ERROR: ioctl(CA_SET_DESCR_MODE): Invalid argument
[... cut ...]
2019/02/21 18:15:31 14F59644 p  (constcw) ERROR: Can't open /etc/tuxbox/config/constant.cw (errno=2 No such file or directory)
2019/02/21 18:15:34 752CC975 c   (dvbapi) ERROR: ioctl(CA_SET_DESCR_MODE): Invalid argument
2019/02/21 18:15:34 752CC975 c      (ecm) dvbapiau (P: 0B00::81D6:0E60:0000 #ECM_L:CF #CW=BA6AA9CDACA0BB079ED46DDF0EB67F43 HOP:01): found (2538 ms) by cccam_server (F/2/2/2) - SONY MAX 2 Rs 1
2019/02/21 18:15:44 14F59644 p  (constcw) ERROR: Can't open /etc/tuxbox/config/constant.cw (errno=2 No such file or directory)

athartanwri, 2019-02-21:
**Log File of Oscam** and the Channel is working Fine

2019/02/21 18:15:50 00000000 s     (main) System name    = Linux
2019/02/21 18:15:50 00000000 s     (main) Host name      = dm520
2019/02/21 18:15:50 00000000 s     (main) Release        = 3.4-4.0-dm520
2019/02/21 18:15:50 00000000 s     (main) Version        = #1 SMP Fri Feb 9 11:33:35 UTC 2018
2019/02/21 18:15:50 00000000 s     (main) Machine        = mips
2019/02/21 18:15:50 00000000 s     (main) Stb model      = dm520
2019/02/21 18:15:50 00000000 s     (main) Stb vumodel    = vuultimo
2019/02/21 18:15:50 00000000 s     (main) Stb boxtype    = vuultimo
2019/02/21 18:15:50 00000000 s     (main) creating pidfile /tmp/.gcam/gcam.pid with pid 2185
2019/02/21 18:15:50 00000000 s      (emu) gcam-Emu version 5
2019/02/21 18:15:50 00000000 s   (config) userdb reloaded: 1 accounts loaded, 0 expired, 0 disabled
2019/02/21 18:15:50 00000000 s     (main) signal handling initialized
2019/02/21 18:15:50 00000000 s   (config) 3 provid's loaded
2019/02/21 18:15:50 00000000 s   (config) 36 service-id's loaded in 1 ms
2019/02/21 18:15:50 00000000 s   (reader) cccam-1 [cccam] creating thread for device city.4k7.me
2019/02/21 18:15:50 00000000 s     (stat) loadbalancer: could not open /tmp/.gcam/stat for reading (errno=2 No such file or directory)
2019/02/21 18:15:50 00000000 s     (main) waiting for local card init
2019/02/21 18:15:50 623FDEE0 h    (webif) webif: decompressed 179064 bytes back into 446400 bytes
2019/02/21 18:15:50 623FDEE0 h    (webif) HTTP Server running. ip=0.0.0.0 port=8888
2019/02/21 18:15:50 0D186CF1 p   (reader) city.4k7.me: resolved ip=212.237.22.87
2019/02/21 18:15:50 0D186CF1 p   (reader) cccam-1 [cccam] connecting to city.4k7.me:12000
2019/02/21 18:15:51 00000000 s     (main) init for all local cards done
2019/02/21 18:15:51 00000000 s (emmcache) loaded 0 emmcache records from /etc/tuxbox/config/gcam.emmcache in 0 ms
2019/02/21 18:15:51 00000000 s (emmcache) loaded 0 emmstat records from /etc/tuxbox/config/gcam.emmstat in 0 ms
2019/02/21 18:15:51 00000000 s (anticasc) anti cascading disabled
2019/02/21 18:15:51 7062E180 c   (client) plain dvbapi-client granted (dvbapi, au=auto (7 reader))
2019/02/21 18:15:51 7062E180 c   (dvbapi) dvbapi channelcache loaded from /etc/tuxbox/config/gcam.ccache
2019/02/21 18:15:51 7062E180 c   (dvbapi) Detected /dev/dvb/adapter0/demux0 Api: 0, userconfig boxtype: 1 maximum amount of possible filters is 32 (gcam limit is 32)
2019/02/21 18:15:51 0D186CF1 p   (reader) cccam-1 [cccam] proxy initialized, server city.4k7.me:12000
2019/02/21 18:15:55 7062E180 c   (dvbapi) Demuxer 0 ecmpid 0 CAID: 1712 ECM_PID: 172C PROVID: 000000 
2019/02/21 18:15:55 7062E180 c   (dvbapi) Demuxer 0 ecmpid 1 CAID: 0BC1 ECM_PID: 1501 PROVID: 000000 
2019/02/21 18:15:55 7062E180 c   (dvbapi) Demuxer 0 ecmpid 2 CAID: 0B00 ECM_PID: 0E60 PROVID: 000000 
2019/02/21 18:15:55 7062E180 c   (dvbapi) Demuxer 0 found 3 ECMpids and 8 STREAMpids in caPMT
2019/02/21 18:15:55 7062E180 c   (dvbapi) Demuxer 0 trying to descramble PID 2 CAID 0B00 PROVID 000000 ECMPID 0E60 ANY CHID PMTPID 0E5C VPID 0E5D
2019/02/21 18:15:55 7062E180 c   (dvbapi) Demuxer 0 found 3 ECMpids and 8 STREAMpids in PMT
2019/02/21 18:15:55 7062E180 c   (dvbapi) ERROR: ioctl(CA_SET_DESCR_MODE): Invalid argument
2019/02/21 18:15:55 7062E180 c        (-) -- Skipped 1 duplicated log lines --
2019/02/21 18:15:55 7062E180 c      (ecm) dvbapi (0B00@000000/0000/81D6/CF:13163213E5EA774173BFE60EBA886F17): cache3 (208 ms) by cccam-1 - SONY MAX 2 Rs 1
2019/02/21 18:15:58 7062E180 c   (dvbapi) ERROR: ioctl(CA_SET_DESCR_MODE): Invalid argument
2019/02/21 18:15:58 7062E180 c      (ecm) dvbapi (0B00@000000/0000/81D6/CF:7B8B58DBD8C6D1A31EC258E4FA8C15F3): cache3 (2220 ms) by cccam-1 - SONY MAX 2 Rs 1
2019/02/21 18:16:10 7062E180 c   (dvbapi) ERROR: ioctl(CA_SET_DESCR_MODE): Invalid argument
2019/02/21 18:16:10 7062E180 c      (ecm) dvbapi (0B00@000000/0000/81D6/CF:D350200E80688A3DF7998002A2ED46FA): cache3 (1940 ms) by cccam-1 - SONY MAX 2 Rs 1

oscam-emu, 2019-02-22:
These channels open with card sharing. I don't see what is has to do with oscam-emu...

Make sure you use the same config for all emu's. It might work.

athartanwri, 2019-02-22:
I have even replaced the gcam configuration files, by changing name gcam I.e. gcam.server to ncam.server and so on for all remaining gcam configuration files.

I have also repeated it for oscam, yet oscam don't open the channels which gcam opens.

And when I replace Oscan files I.e. from oscam.server to gcam.server likewise for all files. Then gcam opens the channels.

athartanwri, 2019-02-22:
### **Dear Expert here are some differences in Log Files created by Gcam and Oscam with same configuration files and Same Channel.**

2019/02/22 14:02:50 00000000 s     (main) System name    = Linux
2019/02/22 14:02:50 00000000 s     (main) Host name      = dm520
2019/02/22 14:02:50 00000000 s     (main) Release        = 3.4-4.0-dm520
2019/02/22 14:02:50 00000000 s     (main) Version        = #1 SMP Fri Feb 9 11:33:35 UTC 2018
2019/02/22 14:02:50 00000000 s     (main) Machine        = mips
2019/02/22 14:02:50 00000000 s     (main) Stb model      = dm520
2019/02/22 14:02:50 00000000 s     (main) Stb vumodel    = vuultimo
2019/02/22 14:02:50 00000000 s     (main) Stb boxtype    = vuultimo

**Gcam 1.5**
2019/02/22 14:02:50 00000000 s   (reader) constant.cw [constcw] creating thread for device /etc/tuxbox/config/constant.cw

**Oscam 789**
2019/02/22 14:04:52 00000000 s   (reader) constant.cw [unknown] Protocol Support missing. (typ=37)

**On the following Channel**
2019/02/22 14:02:55 667803FC c   (dvbapi) Demuxer 0 ecmpid 0 CAID: 1712 ECM_PID: 172A PROVID: 000000 
2019/02/22 14:02:55 667803FC c   (dvbapi) Demuxer 0 ecmpid 1 CAID: 0BC1 ECM_PID: 14FF PROVID: 000000 
2019/02/22 14:02:55 667803FC c   (dvbapi) Demuxer 0 ecmpid 2 CAID: 0B00 ECM_PID: 0E79 PROVID: 000000 
2019/02/22 14:02:55 667803FC c   (dvbapi) Demuxer 0 found 3 ECMpids and 8 STREAMpids in caPMT

**On Gcam BBC Channel is Working Fine**

2019/02/22 14:02:55 667803FC c   (dvbapi) Demuxer 0 trying to descramble PID 2 CAID 0B00 PROVID 000000 ECMPID 0E79 ANY CHID PMTPID 0E75 VPID 0E76
2019/02/22 14:02:55 4BB234E5 p  (constcw) ERROR: Can't open /etc/tuxbox/config/constant.cw (errno=2 No such file or directory)
2019/02/22 14:02:55 667803FC c   (dvbapi) ERROR: ioctl(CA_SET_DESCR_MODE): Invalid argument
2019/02/22 14:02:55 667803FC c        (-) -- Skipped 1 duplicated log lines --
2019/02/22 14:02:55 667803FC c      (ecm) dvbapiau (P: 0B00::8240:0E79:0000 #ECM_L:98 #CW=67966C57197F9E140C2A6344E868DD21 HOP:01): cache3 (201 ms) by cccam_server (C/2/2/2) - BBC WORLD NEWS Rs 1

**On Oscam BBC Channel is not working**
2019/02/22 14:04:57 00000000   >> OSCam << cardserver log switched, version 1.20_svn, build r11500-789 (mipsel-satdreamgr-linux-gnu)
2019/02/22 14:04:57 09ADC9D9 c   (dvbapi) Demuxer 0 no suitable readers found that can be used for decoding!
2019/02/22 14:04:57 09ADC9D9 c   (dvbapi) Demuxer 0 trying to descramble PID 2 CAID 0B00 PROVID 000000 ECMPID 0E79 ANY CHID PMTPID 0E75 VPID 0E76
2019/02/22 14:04:57 09ADC9D9 c   (dvbapi) Demuxer 0 found 3 ECMpids and 8 STREAMpids in PMT
2019/02/22 14:04:57 09ADC9D9 c      (ecm) dvbapiau (P: 0B00::8240:0E79:0000 #ECM_L:98 #CW=16CE391D1DD16250B804631FA792A1DA HOP:01): found (203 ms) by cccam_server - BBC WORLD NEWS Rs 1

@oscam-emu Kindly look in the log files and suggest any solution.

oscam-emu, 2019-03-21:
I told you that these channels open with card sharing, not emu. So this doesn't belong here...

I see various problems and errors in your logs. If you spend some time reading them, you will see the problems as well.

closed 2019-03-21 by oscam-emu