oscam-emu/oscam-patched-old #34: RSA KEYS ABOUT?

PAKO34, 2019-02-22:
**Are you add in emu rsa keys? for example **
**second questions? mgcamd autoroll.key file out of for oscam emu?
thank you**

> *#*#*# RSA KEYS #*#*#*#*#*#*#*#*#*#*#
V 021110 C1 E96B9FD44D3D335E ; ; ChainArray SEVMEK46 2018-04-09 04:14:40
V 021110 D1 89CBB54EEB94E350 ; ; Des1_Key SEVMEK46 2018-04-09 04:15:03
V 021110 P1 0704020305000601 ; ; PermArray SEVMEK46 2018-04-09 04:14:47
V 021110 T1 9453847eee7345cfd1d482d3603036ecd6cd9af5da1fe5243e715cea8641ba1528a747c2172edcd920968e752f4a252c0d38ab4ca56e0e8d31644e5e7761189f781dfa85fd065922f7e92d9533a93ae8f1e788015de3d29262465ff21a543b5a0c3d58c939d8ae7f876cbfd569ce35c49b19c105c82bac3c40edb1fcbc990367a4b80aa102431c6852f8beffb6372aefb9a657bb004b29b4db7d1270e1aab53fd083b7e280349121e44d9d3276f066dfde7accc7979e8ca281901e937cc38a6ae67223bd6ff6cab37463c6feb2116d07a008560b096b10e06527149826ebb0afd79ca355a816c0514f491bdd0f79048fad505bf4f313c54889fb42f97b44cb8b ; ; TransformTable SEVMEK46 2018-04-09 04:14:30
V 021110 X1 12BF4D2F2A10F590 ; ; XORArray SEVMEK46 2018-04-09 04:14:53
N 2111 M1 7508665801DC6815F8DF965641A8C27CAE056ED0249CF60F60DEDD75A3A84D486E85285220F133439F038BFE54975543C8DFC4205067985E5CDAD7ABA4CF48A7 ; ;Digi TV Cablu (Romania Cable   2014-06-29 00:26:18
P 0009 00233C13 245B87F60D2D9D ; AFN EMM SEVMEK46 2018-07-14 20:10:20
P 0009 004E0256 D3A527AB8A6EF3 ; AFN EMM SEVMEK46 2018-07-14 20:10:46
P 0009 003E042A 440610E23210D9 ; AFN EMM SEVMEK46 2018-07-14 20:10:59
P 0009 00541E85 689F03A1EE5DA3 ; AFN EMM SEVMEK46 2018-07-14 20:11:09
P 0E00 00 00540716 ; AFN EMM KEYS 9E  2018-10-21 14:17:54
I 060400 M2 AE652B210BF89FC69507609842FD303E ; Bulsat (39°E) (ECM IV)  2015-02-12 15:28:46
I 060400 M1 98B4DCAD44E8C9504C3F4E51692A7047 ; Bulsat (39°E) (ECM Seed)  2015-02-12 15:29:38
N 7301 M1 7544E982065D53E7CF049AF2E725544A50268DA1462F15D8A197ECE4B80FE221122F92AD56324380A04878EBF563427DD85C38645907583E23AC5153E300CF84 ; UPC Key SEVMEK46 2018-05-03 06:54:14

oscam-emu, 2019-02-22:
WTF? What is this nonsense? Why do you post keys publicly?

What exactly is your question? What should we do with these keys anyway?

PAKO34, 2019-02-22:
**mgcamd using autoroll.key file

is oscam emu use autoroll.key file?**

oscam-emu, 2019-02-22:
oscam-emu operates differently than mgcamd.

closed 2019-02-22 by oscam-emu