oscam-emu/oscam-patched-old #38: Docker Container!

mmBesar, 2019-03-08:
Sorry if it's not the right place! but could you please create a docker container for "oscam-patched" "https://github.com/oscam-emu/oscam-patched.git"
or tell me how to patch this one "https://hub.docker.com/r/linuxserver/oscam" to become like "https://github.com/oscam-emu/oscam-patched.git" as I tried to replace the oscam file with yours but it did not work!
Thank you for the amazing project.

fat-fred, 2019-03-13:
In the Dockerfile from linuxserver you Must change line 31 from:
" bzr branch lp:oscam /tmp/oscam-svn && \"
To this:
" git clone https://github.com/oscam-emu/oscam-patched.git /tmp/oscam-svn && \"
and then Build the docker for your Self. 
Perhaps you have to install "git" also in the docker.

mmBesar, 2019-03-15:
@fat-fred thank you
I have no idea how to do it! I just use the the docker in openmediavault.
If you could give me a tutorial on how to do so, I'll be grateful.
Docker is the best way to add services to my server, but I just use it, no idea how to build one.
Thanks.

closed 2019-04-18 by oscam-emu

mmBesar, 2019-04-18:
So, no hope!

oscam-emu, 2019-04-18:
What else do you need? You got your answer. If you're not able to edit a url inside a file, don't expect anyone to do it for you.

mmBesar, 2019-04-18:
Thanks, but I need to know how, and I'll do it myself.
I'M using docker plugin for openmediavault, it just get and build the docker in one step, I don't know which file to edit.
Put me in the right direction and I will do it, any online guide or toturial will do.
Thanks again, I appreciate your work highly.

mmBesar, 2019-07-19:
@fat-fred Thank you, I've done it.
@oscam-emu Thanks!