oscam-emu/oscam-patched-old #39: feature: #define MAX_DEMUX

Saentist, 2019-03-10:
There is some transponders with lot of services and oscam cannot descramble all with dvbapi

it is needed to edit **module-dvbapi.h** to
#define MAX_DEMUX 32
or even bigger number

If is configurable by web-ui even better.

oscam-emu, 2019-03-21:
I am not so sure the problem is the number demuxers. Probably the box runs out of descramblers before it runs out of demuxers.

Can you tell me what box is that and what satellite/transponder?

Can you change the MAX_DEMUX to what ever it suits you and compile a binary to test if it helps?

Can you tell me how many demuxers there are in your box? Check under `/dev/dvb/adapter0/`, `/dev/dvb/adapter1/` etc...

Saentist, 2019-03-21:
just one 
```
/dev/dvb
├── adapter0
│   ├── demux0
│   ├── dvr0
│   ├── frontend0
│   └── net0
└── adapter1
    ├── demux0
    ├── dvr0
    ├── frontend0
    └── net0

2 directories, 8 files
```

adapter 1 is my DVB-T with free channels 

more info 
https://github.com/braice/MuMuDVB/issues/237

oscam-emu, 2019-03-21:
Oh, this is a pc. I thought it was an enigma2 box or another stb.

Well, if I increase MAX_DEMUX to 32 the memory consumption will increase (and the cpu usage) with no real benefit to the 99,99% of the users.

The majority of the people using oscam-emu has an stb, so I'll have to say **no** to your request.

Since your're using a pc you can easily edit the demux number to what you like and compile your own binary. It's not that difficult.

Thanks for understanding.

Saentist, 2019-03-21:
Lets imagine that I have Dreambox 920 with FCB tuner how will proceed?

oscam-emu, 2019-03-21:
How many demuxers and how many descramblers has a dm920 with fcb tuner? I don't think that 16 are not enough for that...

Saentist, 2019-03-21:
DM920 2x8 tuners
One transponder with 20 services simultaneous streaming is enough no difference what hardware used.

oscam-emu, 2019-03-24:
@Saentist I will do some testing and probably increase the number of demuxes in the next release.

oscam-emu, 2019-04-09:
MAX_DEMUX increased to 32 in oscam 11511. Thanks for being patient.

closed 2019-04-09 by oscam-emu