oscam-emu/oscam-patched-old #40: Found a problem in PVU AU !

mihaip1948, 2019-03-21:


mihaip1948, 2019-03-21:
I've made a test with Zgemma H9 to decode AFN using AU feature ! 
- it started in some 30 seconds ! perfect !
- then I've tried Astra 4.8e 12303 Diva Romania ... does not open ! 
- but I've enter in Readers-Emulator ... no change ... Save !
- decoding started in that very moment
- then the same on Astra 4.8e 12322 Discovery Word ... open !
- next tried the same  on Thor 1w 12303 ... all channels open !!!

For me this is a issue in EMU ... some how must be ... stimulated 
to start in this case !!!

oscam-emu, 2019-03-21:
What does "does not open" mean? It can mean that you don't have the correct ecm key, or a thousand different things...

I need logs (with debug enabled) demonstrating the problem... which I don't think there is anyway.
AU can take few seconds or several minutes to find new keys. Just by saving the reader does not make things work.

oscam-emu, 2019-04-18:
I am closing this, as no logs are provided.

closed 2019-04-18 by oscam-emu