oscam-emu/oscam-patched-old #41: make failed

Shporterator, 2019-04-08:
build/aarch64-linux-gnu/module-emulator-biss.o: In function `biss_read_pem':
/home/git/oscam-patched/module-emulator-biss.c:842: undefined reference to `PEM_read_RSAPrivateKey'
/home/git/oscam-patched/module-emulator-biss.c:867: undefined reference to `i2d_RSA_PUBKEY'
/home/git/oscam-patched/module-emulator-biss.c:876: undefined reference to `EVP_MD_CTX_new'
/home/git/oscam-patched/module-emulator-biss.c:882: undefined reference to `EVP_sha256'
/home/git/oscam-patched/module-emulator-biss.c:882: undefined reference to `EVP_DigestInit_ex'
/home/git/oscam-patched/module-emulator-biss.c:883: undefined reference to `EVP_DigestUpdate'
/home/git/oscam-patched/module-emulator-biss.c:884: undefined reference to `EVP_DigestFinal_ex'
/home/git/oscam-patched/module-emulator-biss.c:885: undefined reference to `EVP_MD_CTX_free'
collect2: error: ld returned 1 exit status
Makefile:437: recipe for target 'Distribution/oscam-1.20_svn11509-795-aarch64-linux-gnu.debug' failed
make[1]: *** [Distribution/oscam-1.20_svn11509-795-aarch64-linux-gnu.debug] Error 1
Makefile:406: recipe for target 'all' failed
make: *** [all] Error 2

_
root@rockpro64:/home# uname -a
Linux rockpro64 4.4.154 #1 SMP Thu Feb 7 17:40:53 UTC 2019 aarch64 aarch64 aarch64 GNU/Linux

oscam-emu, 2019-04-08:
Don't you read the [release notes](https://github.com/oscam-emu/oscam-patched/releases/tag/oscam11509-emu795)?

> From this version and on, building OSCam-Emu will require linking with libcrypto due to the extedned use of cryptographic functions in BISS2 code. This means that you must enable the USE_SSL=1 or USE_LIBCRYPTO=1 switches when compiling.

Shporterator, 2019-04-08:
Sorry, no.

closed 2019-04-08 by Shporterator