oscam-emu/oscam-patched-old #42: DVBAPI Stop Descrambling Some SPN Channels (after r11511)

budiarno, 2019-04-13:
Hi oscam-emu.

I found this error after compiling to the latest source of oscam-emu (r11512), and the error is known to be since r11511 (as also reported on streamboard). Some channels on SPN Intelsat 20 are effected after the changes, for example: PIX India and SAB India.

Log file:
[oscam-emu-r11512-SPN-PixIndia.txt](https://github.com/oscam-emu/oscam-patched/files/3075456/oscam-emu-r11512-SPN-PixIndia.txt)
[oscam-emu-r11512-SPN-SAB.txt](https://github.com/oscam-emu/oscam-patched/files/3075457/oscam-emu-r11512-SPN-SAB.txt)

Recorded TS files on my enigma2
These with oscam-emu on (running), about a minute long:
https://www.mediafire.com/file/fybdz3465sx8k88/PIX-India-Int20.zip/file (PIX India)
https://www.mediafire.com/file/eu8d6t9ls7zaxkc/SAB-India-Int20.zip/file (SAB India)

This one with oscam-emu off (not running) about two minutes long:
https://www.mediafire.com/file/fybdz3465sx8k88/PIX-India-Int20.zip/file (PIX India)

I'm sorry I can't provide a recorded file via PC because I don' have dvbcard.

If more things I should do, please tell me.

Thank you!

oscam-emu, 2019-04-13:
I don't know what is the problem with the pmt file. And I can't understand how the MAX_STREAM_INDICES affect the finding and reading the pmt file.

Can you send me a log with the oscam-emu you compiled with MAX_STRREAM_INDICES 32?
In the meanwhile I'll find out what this pmt file does...

budiarno, 2019-04-13:
Ok, I'll send via email.

oscam-emu, 2019-04-13:
I still don't understand what is the relation between the pmt file and the MAX_STREAM_INDICES.

Nevertheless, there is a bug when MAX_STREAM_INDICES  is not equal to MAX_ECM_PIDS (which is also 32), but is only exposed only with powervu channels with "weird" stream ordering like these two. In this case oscam-emu should crash, like in another report I got.

Anyway, I increased the number back to 32 for now, but I will do some more improvements in the future (which will bring the number back to 8 again - but this time without problems).

closed 2019-04-13 by oscam-emu