oscam-emu/oscam-patched-old #43: Oscam EMU 11512

mihaip1948, 2019-04-15:


mihaip1948, 2019-04-15:
Oscam EMU 11512 cannot be build due to some link
errors ! refferences from BISS module ...

mihaip1948, 2019-04-16:
CONF	build/aarch64-buildroot-linux-gnu/config.c
LINK	Distribution/oscam-1.20_svn11512-795-aarch64-buildroot-linux-gnu.debug
/home/mihaip/aarch64/bin/../lib/gcc/aarch64-buildroot-linux-gnu/8.2.0/../../../../aarch64-buildroot-linux-gnu/bin/ld: build/aarch64-buildroot-linux-gnu/module-emulator-biss.o: in function `biss_read_pem':
/home/mihaip/oscam-e2/oscam/oscam-svn/module-emulator-biss.c:842: undefined reference to `PEM_read_RSAPrivateKey'
/home/mihaip/aarch64/bin/../lib/gcc/aarch64-buildroot-linux-gnu/8.2.0/../../../../aarch64-buildroot-linux-gnu/bin/ld: /home/mihaip/oscam-e2/oscam/oscam-svn/module-emulator-biss.c:867: undefined reference to `i2d_RSA_PUBKEY'
/home/mihaip/aarch64/bin/../lib/gcc/aarch64-buildroot-linux-gnu/8.2.0/../../../../aarch64-buildroot-linux-gnu/bin/ld: /home/mihaip/oscam-e2/oscam/oscam-svn/module-emulator-biss.c:876: undefined reference to `EVP_MD_CTX_new'
/home/mihaip/aarch64/bin/../lib/gcc/aarch64-buildroot-linux-gnu/8.2.0/../../../../aarch64-buildroot-linux-gnu/bin/ld: /home/mihaip/oscam-e2/oscam/oscam-svn/module-emulator-biss.c:882: undefined reference to `EVP_sha256'
/home/mihaip/aarch64/bin/../lib/gcc/aarch64-buildroot-linux-gnu/8.2.0/../../../../aarch64-buildroot-linux-gnu/bin/ld: /home/mihaip/oscam-e2/oscam/oscam-svn/module-emulator-biss.c:882: undefined reference to `EVP_DigestInit_ex'
/home/mihaip/aarch64/bin/../lib/gcc/aarch64-buildroot-linux-gnu/8.2.0/../../../../aarch64-buildroot-linux-gnu/bin/ld: /home/mihaip/oscam-e2/oscam/oscam-svn/module-emulator-biss.c:883: undefined reference to `EVP_DigestUpdate'
/home/mihaip/aarch64/bin/../lib/gcc/aarch64-buildroot-linux-gnu/8.2.0/../../../../aarch64-buildroot-linux-gnu/bin/ld: /home/mihaip/oscam-e2/oscam/oscam-svn/module-emulator-biss.c:884: undefined reference to `EVP_DigestFinal_ex'
/home/mihaip/aarch64/bin/../lib/gcc/aarch64-buildroot-linux-gnu/8.2.0/../../../../aarch64-buildroot-linux-gnu/bin/ld: /home/mihaip/oscam-e2/oscam/oscam-svn/module-emulator-biss.c:885: undefined reference to `EVP_MD_CTX_free'
collect2: error: ld returned 1 exit status
Makefile:437: recipe for target 'Distribution/oscam-1.20_svn11512-795-aarch64-buildroot-linux-gnu.debug' failed
make[1]: *** [Distribution/oscam-1.20_svn11512-795-aarch64-buildroot-linux-gnu.debug] Error 1
Makefile:406: recipe for target 'all' failed
make: *** [all] Error 2
rm: cannot remove '*.debug': No such file or directory
mv: cannot stat './oscam-svn/Distribution/oscam-1*': No such file or directory
root@ubuntu:/home/mihaip/oscam-e2/oscam#

oscam-emu, 2019-04-16:
It needs linking with libcrypto since the last release. Don't read the release log? https://github.com/oscam-emu/oscam-patched/releases/tag/oscam11509-emu795

closed 2019-04-16 by oscam-emu

mihaip1948, 2019-04-17:
Unfortunately ... even after link with libcrypto ... I have build errors !!!
OpenSSL was installed , ./config give "whatever Linux" them "make"

CONF	build/aarch64-buildroot-linux-gnu/config.c
LINK	Distribution/oscam-1.20_svn11512-795-aarch64-buildroot-linux-gnu.debug
/home/mihaip/aarch64/bin/../lib/gcc/aarch64-buildroot-linux-gnu/8.2.0/../../../../aarch64-buildroot-linux-gnu/bin/ld: ../openssl-1.1.1a/libcrypto.a: error adding symbols: file in wrong format

Please a little help !!!

oscam-emu, 2019-04-17:
Post you make command

mihaip1948, 2019-04-17:
OpenSSL is build with this script   : 

wget http://www.openssl.org/source/openssl-1.1.1a.tar.gz
tar -xf openssl-1.1.1a.tar.gz
cd openssl-1.1.1a
./Configure linux-aarch64 no-afalgeng no-aria no-asan no-asm no-async no-autoalginit no-autoerrinit no-autoload-config no-bf no-blake2 no-camellia no-capieng no-cast no-chacha no-cmac no-cms no-comp no-crypto-mdebug no-crypto-mdebug-backtrace no-ct no-deprecated no-devcryptoeng no-dgram no-dh no-dsa no-dso no-dtls no-dynamic-engine no-ec no-ec2m no-ecdh no-ecdsa no-ec_nistp_64_gcc_128 no-egd no-engine no-err no-external-tests no-filenames no-fuzz-libfuzzer no-fuzz-afl no-gost no-heartbeats no-idea no-makedepend no-md2 no-md4 no-msan no-multiblock no-nextprotoneg no-ocb no-ocsp no-pic no-poly1305 no-posix-io no-psk no-rc2 no-rc4 no-rc5 no-rdrand no-rfc3779 no-rmd160 no-scrypt no-sctp no-seed no-shared no-siphash no-sm2 no-sm3 no-sm4 no-sock no-srp no-srtp no-sse2 no-ssl no-ssl-trace no-stdio no-tests no-threads no-tls no-ts no-ubsan no-ui-console no-unit-test no-whirlpool no-weak-ssl-ciphers no-zlib no-zlib-dynamic no-ssl3 no-ssl3-method no-tls1 no-tls1-method no-tls1_1 no-tls1_1-method no-tls1_2 no-tls1_2-method no-tls1_3 no-dtls1 no-dtls1-method no-dtls1_2 no-dtls1_2-method
make
cd ../

... build ends with no error ... only a compile warning ! but
lincrypto.a ... still have wrong format !!!

oscam-emu, 2019-04-17:
I meant to post your oscam-emu make command, not openssl make command... Anyway, I make my openssl for aarch64 like this:

```
make clean
export CC=/home/oscam-emu/toolchains/aarch64/bin/aarch64-buildroot-linux-gnu-gcc
export RANLIB=/home/oscam-emu/toolchains/aarch64/bin/aarch64-buildroot-linux-gnu-ranlib
./Configure linux-aarch64 shared --prefix=/home/oscam-emu/toolchains/aarch64/aarch64-buildroot-linux-gnu/sysroot/usr/
make
make test
sudo make install
```

Depending on the name and the location of your aarch64 toolchain, you'll have to adjust the paths accordingly.

Then you build oscam-emu with:

```
make -j 6 CROSS=/home/oscam-emu/toolchains/aarch64/bin/aarch64-buildroot-linux-gnu- USE_LIBCRYPTO=1 USE_UTF8=1
```

mihaip1948, 2019-04-17:
Well ! you help was valuable ... as always !!! I've build successful   ... thanks again !!!