oscam-emu/oscam-patched-old #44: rejected group (no matching reader)

amiralishahrestani, 2019-04-16:


oscam-emu, 2019-04-17:
I am not sure what are you trying to do... Is seems there is something wrong with your user/reader groups?

amiralishahrestani, 2019-04-17:
if i disable LoadBalancer everythings will be ok and i don't get (no matching reader from AFN)

oscam-emu, 2019-04-17:
Neither I get it... because I am not a magician. I need a log with proper debug level and your config files to get an idea of what might be wrong.

amiralishahrestani, 2019-04-17:
[log.txt](https://github.com/oscam-emu/oscam-patched/files/3090736/log.txt)

`[global]
logfile                       = /tmp/oscam.log
nice                          = -1
preferlocalcards              = 1
block_same_ip                 = 0
block_same_name               = 0
ecmfmt                        = c:p:s:d:i:o #ECM_L:l #CW=w HOP:j
disablecrccws_only_for        = 0E00:000000

[cache]

[lcd]
enablelcd                     = 1
lcd_outputpath                = /tmp
lcd_hideidle                  = 1

[streamrelay]
stream_relay_enabled          = 0
stream_ecm_delay			  = 300
stream_emm_enabled            = 1
stream_relay_ctab  			  = 0E00

[dvbapi]
enabled                       = 1
au                            = 1
pmt_mode                      = 0
delayer                       = 60
ecminfo_type                  = 0
user                          = amirali
read_sdt                      = 1
write_sdt_prov                = 1
extended_cw_api               = 1
boxtype                       = dreambox

[webif]
httpport                      = 8888
httppiconpath                 = /media/usb/picon
httprefresh                   = 10
httpshowpicons                = 1
httpshowmeminfo               = 1
httpshowuserinfo              = 1
httpshowreaderinfo            = 1
httpshowecminfo               = 1
httpshowloadinfo              = 1
httpallowed                   = 127.0.0.1,192.168.0.0-192.168.255.255
httposcamlabel                = Persian-Dreambox OSCam
http_status_log               = 1

[cs357x]
port = 12345

`
`
[reader]
label                         = Biss&PowerVU
protocol                      = emu
device                        = /etc/tuxbox/config/SoftCam.Key
disablecrccws_only_for        = 0E00:000000
caid                          = 0500,0604,090F,0E00,1010,1801,2600,2602,2610,4AE1
detect                        = cd
ident                         = 0500:000000,007400,007800,021110,023800;0604:000000;090F:000000;0E00:000000;1010:000000;1801:000000,001101,002111,007301;2600:000000;2602:000000;2610:000000;4AE1:000011,000014,0000FE
group                         = 1
emmcache                      = 2,1,2,1
emu_auproviders               = 0604:010200;0E00:000000;1010:000000;2610:000000;4AE1:000011,000014,0000FE

[reader]
label                         = constant.cw
protocol                      = constcw
device                        = /etc/tuxbox/config/constant.cw
disablecrccws_only_for        = 0E00:000000
caid                          = 2600,0B00,0B02,0500,0963,06AD,0940
group                         = 1

[reader]
label                         = Windows10
enable                        = 0
protocol                      = cs357x
device                        = 192.168.88.50,12345
user                          = amirali
disablecrccws_only_for        = 0E00:000000
group                         = 1
emmcache                      = 1,3,2,0
`

oscam-emu, 2019-04-18:
I don't understand what you're trying to do and I don't see loadbalancer enabled (http://www.streamboard.tv/wiki/OSCam/en/Config/oscam.conf#lb_mode)

jeanniquini, 2019-05-13:
I'm also having this problem, even the readers and the user being in the same group and not specifying chid, ident, service and caid. When I disable loadbalancer, everything works correctly.

oscam-emu, 2019-08-30:
Probably the load balancer does not support the extended cw needed for powervu.

closed 2019-08-30 by oscam-emu

palaufi, 2022-03-12:
Why are you so arrogant with others? Only because you wrote this piece of code? making feel people stupid doesn't make you smarter.