oscam-emu/oscam-patched-old #45: can't build at deb9

vlelin, 2019-04-28:
CONF	build/x86_64-linux-gnu/config.c
CC	oscam.c
CC	oscam-work.c
LINK	Distribution/oscam-1.20_svn11517-795-x86_64-linux-gnu.debug
build/x86_64-linux-gnu/module-emulator-biss.o: In function `biss_read_pem':
/root/oscam-patched/module-emulator-biss.c:842: undefined reference to `PEM_read_RSAPrivateKey'
/root/oscam-patched/module-emulator-biss.c:867: undefined reference to `i2d_RSA_PUBKEY'
/root/oscam-patched/module-emulator-biss.c:876: undefined reference to `EVP_MD_CTX_new'
/root/oscam-patched/module-emulator-biss.c:882: undefined reference to `EVP_sha256'
/root/oscam-patched/module-emulator-biss.c:882: undefined reference to `EVP_DigestInit_ex'
/root/oscam-patched/module-emulator-biss.c:883: undefined reference to `EVP_DigestUpdate'
/root/oscam-patched/module-emulator-biss.c:884: undefined reference to `EVP_DigestFinal_ex'
/root/oscam-patched/module-emulator-biss.c:885: undefined reference to `EVP_MD_CTX_free'
collect2: error: ld returned 1 exit status
Makefile:435: recipe for target 'Distribution/oscam-1.20_svn11517-795-x86_64-linux-gnu.debug' failed
make[1]: *** [Distribution/oscam-1.20_svn11517-795-x86_64-linux-gnu.debug] Error 1
Makefile:404: recipe for target 'all' failed
make: *** [all] Error 2

caveman99, 2019-04-28:
read the release notes. you need to link a crypto lib...

oscam-emu, 2019-04-29:
Yeap, you need to compile with `USE_LIBCRYPTO=1`

closed 2019-04-29 by oscam-emu