oscam-emu/oscam-patched-old #46: Link error trying to build static-libcrypto

mihaip1948, 2019-05-12:


mihaip1948, 2019-05-12:
Normal link with shared libcrypto is working ! but build with static libraries fail 
due to multiple definitions  in oscam-biss 

INK	Distribution/oscam-1.20_svn11518-796-armv7a-unknown-linux-android24.debug
/home/mihaip/oscam-ndk/oscam-svn/../android-ndk-r19/toolchains/llvm/prebuilt/linux-x86_64/bin/../lib/gcc/arm-linux-androideabi/4.9.x/../../../../arm-linux-androideabi/bin/ld: error: /home/mihaip/oscam-ndk/android-ndk-r19/toolchains/llvm/prebuilt/linux-x86_64/sysroot/usr/lib/libcrypto.a(des_enc.o): multiple definition of 'DES_SPtrans'
/home/mihaip/oscam-ndk/oscam-svn/../android-ndk-r19/toolchains/llvm/prebuilt/linux-x86_64/bin/../lib/gcc/arm-linux-androideabi/4.9.x/../../../../arm-linux-androideabi/bin/ld: build/armv7a-unknown-linux-android24/cscrypt/mdc2.o: previous definition here
/home/mihaip/oscam-ndk/oscam-svn/../android-ndk-r19/toolchains/llvm/prebuilt/linux-x86_64/bin/../lib/gcc/arm-linux-androideabi/4.9.x/../../../../arm-linux-androideabi/bin/ld: error: /home/mihaip/oscam-ndk/android-ndk-r19/toolchains/llvm/prebuilt/linux-x86_64/sysroot/usr/lib/libcrypto.a(des_enc.o): multiple definition of 'DES_encrypt1'
/home/mihaip/oscam-ndk/oscam-svn/../android-ndk-r19/toolchains/llvm/prebuilt/linux-x86_64/bin/../lib/gcc/arm-linux-androideabi/4.9.x/../../../../arm-linux-androideabi/bin/ld: build/armv7a-unknown-linux-android24/cscrypt/mdc2.o: previous definition here
clang: error: linker command failed with exit code 1 (use -v to see invocation)
Makefile:435: recipe for target 'Distribution/oscam-1.20_svn11518-796-armv7a-unknown-linux-android24.debug' failed
make[1]: *** [Distribution/oscam-1.20_svn11518-796-armv7a-unknown-linux-android24.debug] Error 1

oscam-emu, 2019-05-12:
I am aware of this. The problem is that the/cscrypt/mdc2.c file, which contains code taken from libcrypto itself (thus common function names)  is built anyway, despite using libcrypto.

I sent [a fix for this in oscam](http://www.streamboard.tv/oscam/changeset/11512/trunk), but they reverted it because it broke the native x64 build in debian/ubuntu, as the openssl/libcrypto pre-built package that comes with these distros does not contain the mdc2 algorithm. 

And after that, I didn't try anything different to solve this problem.

mihaip1948, 2019-05-13:
Ok tahanks !

closed 2019-05-13 by mihaip1948