oscam-emu/oscam-patched-old #48: 65w-75w mux global

PAKO34, 2019-05-22:
**mode unmask : 0 ( in use )
Hash mode cw : 1 ( not in use )
mode cw : 8/20/35/41 changing every 2 sec.**

`ECM[Video]: 
ECM: 80 30 45 50 3F 20 0E 00 55 08 F2 76 5B 17 B0 6D 46 8F 00 F3 A0 00 FF 38 C2 FF 9A 20 D3 20 26 2D 76 50 48 DD 98 0A 30 09 00 6F B2 6A D9 7E 09 0A 19 83 90 03 D5 D3 27 96 A5 27 FA 86 DE 1A F3 18 79 BF CF 0F 3F CC 56 BC
Unmasked ECM: 80 30 45 30 3F 20 0E 00 00 08 F2 76 5B 17 B0 6D 46 8F 00 F3 A0 00 FF 38 C2 FF 9A 20 D3 20 26 2D 76 50 48 DD 00 0A 30 09 00 6F B2 6A D9 7E 09 0A 19 83 90 03 D5 D3 27 96 A5 27 FA 86 DE 1A F3 18 79 BF CF 0F 8F FB 23 E3
IsDecrypted: True
DVB Scrambling Algo: DES
Needs Unmasking: True
ECM Counter: $F3
Nanos: [5]->($20, $3E) 
Nano $20 Extra Data Length: 8
Mode CW: 8 (in use)
Mode Unmask: 0 (in use)
Hash Mode CW: 1 (not in use)
Hash Mode Unmask: 0 (in use)
Seed CW ECM: FF D3 FF 2D 38 76 C2 20 48 26 50 00 00 00 00 00 
ECM CalculateSeed: 80 30 45 30 3F 20 0E 00 00 08 F2 76 5B 17 B0 6D 46 8F 00 F3 A0 00 66 0B 8E B6 DE A0 F5 83 3C FC FC 72 EE F5 00 0A 30 09 00 F5 00 B2 A3 FE 01 01 64 07 E6 AD C0 02 82 A0 82 E0 00 00 00 99 00 00 00 C6 0A 30 8F FB 23 E3 
Base Cw: F5 83 3C FC FC 72 EE 
Seed VPid: CF D2 34 96 25 81 00 00 
Seed APid-1: 12 6A 22 71 9C 63 00 00 
Seed APid-2: 12 6A 22 71 9C 63 00 00 
Seed APid-3: 07 6C C7 1A C6 95 00 00 
Seed APid-4: 07 6C C7 1A C6 95 00 00 
Seed HSD: 26 91 E7 EE FB A3 00 00 
S-Box Type: Primary Tables
Xor Mode: Normal mode[0]
Ident: $A30
KeyIdx: 0
Key: FA 04 EA 18 EF 11 2B 
OddCwAudio1: F8 04 64 E6 51 49 79 C7 
EvenCwAudio1: 85 01 CD 62 E5 51 3D C8 
OddCwAudio2: F8 04 64 E6 51 49 79 C7 
EvenCwAudio2: 85 01 CD 62 E5 51 3D C8 
OddCwAudio3: 23 01 E9 75 B3 15 9E 79 
EvenCwAudio3: 16 61 EC 52 5E 7F 46 85 
OddCwAudio4: 23 01 E9 75 B3 15 9E 79 
EvenCwAudio4: 16 61 EC 52 5E 7F 46 85 
OddCwVideo: EC B0 08 34 92 73 6B 62 
EvenCwVideo: 4C BC 52 4F 97 61 7F 79 
OddCwHSD: 64 6E EC 89 34 DC 91 92 
EvenCwHSD: 7A F7 83 1C 2C 64 7C 7A 

TS Before CSA: 
47 02 76 D9 E3 5F 47 3B B5 EF C1 E2 B2 D5 5A AC 40 6A 33 B4 13 47 F0 79 45 63 DA DF A9 A6 A9 0C 04 ED 3D 02 7E 44 B8 88 4B 2D 2F 64 E7 24 6D DA CB 0D FD AC 47 95 72 C2 AA 2F F3 65 B7 F2 26 0F 0E 0E 62 12 EF 5B C5 F8 02 B5 FD 01 B6 AE B6 07 5B E6 C7 76 EA 42 B0 96 C3 79 50 F4 5E 16 4A D2 25 F0 C0 6D D5 F7 87 71 50 09 5B C7 6D 51 67 B0 06 D5 87 EC 72 43 72 F4 EE 84 0C D4 04 AE F9 5D 54 56 10 58 B9 1D ED C7 EB 08 B2 55 8C C6 1D F4 E0 E7 CD 2F FC 31 FA 5B 6A DF D5 6C 06 9B 78 09 E2 28 10 EA F7 71 DE 71 75 DF DC 50 E2 2D 4D B5 DB 09 46 E9 76 68 4F FF 2B 02 D2 F3 

TS After CSA: 
47 02 76 19 10 8D A8 B7 D4 A0 55 65 C2 73 16 E8 DE 88 19 42 C2 BE DA 31 CB C8 C5 5F 9C 7F D4 34 DE 00 DE A9 7B 9A 66 06 D0 92 07 BD 31 8A C7 24 EC 04 FC 11 9B 75 94 3E C7 82 6F A8 DB F5 03 D5 2B B8 D4 28 26 93 3C DD 64 04 90 1C AF 3D 68 86 63 D3 96 EF 9E 3E 19 50 50 42 47 F4 0F 07 8F 0E FE 58 1C A7 EB D6 DC DD 61 93 DA 04 09 16 F3 F3 64 E2 EA 25 A8 89 A8 95 87 96 EF D1 55 CC 67 83 48 E3 8A 48 B0 C7 09 61 88 52 1E D7 2F 47 9F 1A 73 5C 33 FB 55 10 56 AF 89 E0 F3 FB A5 50 8E EB 14 C8 69 6E 10 9E 6F 41 F6 41 0B 7A AE 98 DD 9B 47 A0 E0 A3 3A 9F 03 F8 67 DC D9 45`

luizaum2, 2019-07-16:
```(0E00@000000/0000/0098/48:ED986B666D58FC14EB773932A73DB5F7): found (16 ms) by emulator - TC Action HD
2019/07/16 19:32:47 70CD83BB c (ecm) cw:
2019/07/16 19:32:47 70CD83BB c (ecm) 00 00 00 00 00 00 00 00 31 C8 9B D0 75 26 01 EA
2019/07/16 19:32:48 70CD83BB c (ecm) get cw for ecm:
2019/07/16 19:32:48 70CD83BB c (ecm) 80 30 45 50 3F 20 0E 00 C6 08 02 85 DE 4C CF F4
2019/07/16 19:32:48 70CD83BB c (ecm) FD 2C 00 D0 B0 00 71 0A C6 0A B5 F5 C7 DD 52 1D
2019/07/16 19:32:48 70CD83BB c (ecm) E6 CD A3 C5 49 09 80 09 00 CB 35 FA 6F E6 3E 27
2019/07/16 19:32:48 70CD83BB c (ecm) 6F CA 52 FD EB 9A DC D5 C5 7C 73 9E D4 56 69 3E
2019/07/16 19:32:48 70CD83BB c (ecm) 43 7E 2C 0D C3 7D 37 B9
2019/07/16 19:32:48 631F3FDF r (emu) csaUsed: 0, xorMode: 0, ecmSrvid: 0098, hashModeCw: 1, modeCW: 25
2019/07/16 19:32:48 631F3FDF r (emu) channel hash: 9345CBB6, group id: 065A
2019/07/16 19:32:48 631F3FDF r (emu) calculated cw 0: C8CD0234CBE32080
2019/07/16 19:32:48 631F3FDF r (emu) calculated cw 1: E923E0D3911373CD
2019/07/16 19:32:48 631F3FDF r (emu) calculated cw 2: 19195D0DCD25D664
2019/07/16 19:32:48 631F3FDF r (emu) calculated cw 3: 19195D0DCD25D664
2019/07/16 19:32:48 631F3FDF r (emu) calculated cw 4: 19195D0DCD25D664
2019/07/16 19:32:48 631F3FDF r (emu) calculated cw 5: 19195D0DCD25D664
2019/07/16 19:32:48 631F3FDF r (emu) calculated cw 6: 94686B6EE96BA813
2019/07/16 19:32:48 631F3FDF r (emu) calculated cw 7: 2CB3B370F1C14C8F
2019/07/16 19:32:48 70CD83BB c (ecm) dvbapi (0E00@000000/0000/0098/48:2758DCBB11830371CE5BC87BF5DA9C1E): found (15 ms) by emulator - TC Action HD
2019/07/16 19:32:48 70CD83BB c (ecm) cw:
2019/07/16 19:32:48 70CD83BB c (ecm) C8 CD 02 34 CB E3 20 80 00 00 00 00 00 00 00 00
2019/07/16 19:32:49 70CD83BB c (ecm) get cw for ecm:
2019/07/16 19:32:49 70CD83BB c (ecm) 81 30 45 50 3F 20 0E 00 94 08 01 F3 9C 27 24 DC
2019/07/16 19:32:49 70CD83BB c (ecm) FF E7 00 D1 B0 00 C1 E5 BC 86 3D 98 5B 90 17 AC
2019/07/16 19:32:49 70CD83BB c (ecm) 68 15 77 A6 E4 09 80 09 00 EB 06 2E D9 CF 58 F9
2019/07/16 19:32:49 70CD83BB c (ecm) 8D 8B 1C 9B D1 71 AB 95 F7 61 69 3D 8F 69 4E D5
2019/07/16 19:32:49 70CD83BB c (ecm) DD CC 97 41 6D FC 65 92
2019/07/16 19:32:49 631F3FDF r (emu) csaUsed: 0, xorMode: 0, ecmSrvid: 0098, hashModeCw: 1, modeCW: 8
2019/07/16 19:32:49 631F3FDF r (emu) channel hash: 9345CBB6, group id: 065A
2019/07/16 19:32:49 631F3FDF r (emu) calculated cw 0: 57646D3E382C8A64
2019/07/16 19:32:49 631F3FDF r (emu) calculated cw 1: B58C468C73F10B2F
2019/07/16 19:32:49 631F3FDF r (emu) calculated cw 2: 4070B58F37E9C807
2019/07/16 19:32:49 631F3FDF r (emu) calculated cw 3: 4070B58F37E9C807
2019/07/16 19:32:49 631F3FDF r (emu) calculated cw 4: 4070B58F37E9C807
2019/07/16 19:32:49 631F3FDF r (emu) calculated cw 5: 4070B58F37E9C807
2019/07/16 19:32:49 631F3FDF r (emu) calculated cw 6: BA58015B7AC47A2C
2019/07/16 19:32:49 631F3FDF r (emu) calculated cw 7: DAB3C7B66DEF8607
2019/07/16 19:32:49 70CD83BB c (ecm) dvbapi (0E00@000000/0000/0098/48:EC70B4B97B95C6F410632BAAFD39C8E3): found (15 ms) by emulator - TC Action HD
2019/07/16 19:32:49 70CD83BB c (ecm) cw:
2019/07/16 19:32:49 70CD83BB c (ecm) 00 00 00 00 00 00 00 00 57 64 6D 3E 38 2C 8A 64
2019/07/16 19:32:50 70CD83BB c (ecm) get cw for ecm:
2019/07/16 19:32:50 70CD83BB c (ecm) 80 30 45 50 3F 20 0E 00 13 08 0C 0A DD DA A4 E5
2019/07/16 19:32:50 70CD83BB c (ecm) 50 D9 00 D2 B0 00 7F D5 F3 00 20 24 AD E8 69 FA
2019/07/16 19:32:50 70CD83BB c (ecm) 54 C1 EC C0 A1 09 80 09 00 0F 69 80 7E F6 AD FE
2019/07/16 19:32:50 70CD83BB c (ecm) 3C C4 B4 E8 95 07 4F F1 66 BD D4 D7 C8 34 73 E3
2019/07/16 19:32:50 70CD83BB c (ecm) 26 B4 4E 7D 11 6C EB 9C
2019/07/16 19:32:50 631F3FDF r (emu) csaUsed: 0, xorMode: 0, ecmSrvid: 0098, hashModeCw: 1, modeCW: 8
2019/07/16 19:32:50 631F3FDF r (emu) channel hash: 9345CBB6, group id: 065A
2019/07/16 19:32:50 631F3FDF r (emu) calculated cw 0: 0E6E8654EA799BD0
2019/07/16 19:32:50 631F3FDF r (emu) calculated cw 1: ABD646F19D919861
2019/07/16 19:32:50 631F3FDF r (emu) calculated cw 2: E6BC4AE98A0B2C3D
2019/07/16 19:32:50 631F3FDF r (emu) calculated cw 3: E6BC4AE98A0B2C3D
2019/07/16 19:32:50 631F3FDF r (emu) calculated cw 4: E6BC4AE98A0B2C3D
2019/07/16 19:32:50 631F3FDF r (emu) calculated cw 5: E6BC4AE98A0B2C3D
2019/07/16 19:32:50 631F3FDF r (emu) calculated cw 6: 3BA132DC01BC761C
2019/07/16 19:32:50 631F3FDF r (emu) calculated cw 7: 6723B3D543FE5B31
2019/07/16 19:32:50 70CD83BB c (ecm) dvbapi (0E00@000000/0000/0098/48:89795CE8D0F34EE021FB0C139C5F9459): found (16 ms) by emulator - TC Action HD
2019/07/16 19:32:50 70CD83BB c (ecm) cw:
2019/07/16 19:32:50 70CD83BB c (ecm) 0E 6E 86 54 EA 79 9B D0 00 00 00 00 00 00 00 00
2019/07/16 19:32:51 70CD83BB c (ecm) get cw for ecm:
2019/07/16 19:32:51 70CD83BB c (ecm) 81 30 45 50 3F 20 0E 00 D7 08 D6 FA D1 AB 3C 0D
2019/07/16 19:32:51 70CD83BB c (ecm) E7 CE 00 D3 B0 00 CC 47 6D 19 27 95 C7 7A 32 B9
2019/07/16 19:32:51 70CD83BB c (ecm) 2C CD 87 06 B6 09 80 09 00 D6 B0 1D 4D 29 CC 02
2019/07/16 19:32:51 70CD83BB c (ecm) 54 A5 E3 16 B2 FF E4 DA 97 75 C7 7B E4 39 86 CF
2019/07/16 19:32:51 70CD83BB c (ecm) 2B 43 7E B6 87 3A FC F0
2019/07/16 19:32:51 631F3FDF r (emu) csaUsed: 0, xorMode: 0, ecmSrvid: 0098, hashModeCw: 1, modeCW: 41
2019/07/16 19:32:51 631F3FDF r (emu) channel hash: 9345CBB6, group id: 065A
2019/07/16 19:32:51 631F3FDF r (emu) calculated cw 0: 4CB6CDD9EF524A19
2019/07/16 19:32:51 631F3FDF r (emu) calculated cw 1: 5D7A0425FDF44992
[... 231 lines cut ...]
[... cut ...]
2019/07/16 19:33:04 631F3FDF r (emu) calculated cw 3: E58915898967923B
2019/07/16 19:33:04 631F3FDF r (emu) calculated cw 4: E58915898967923B
2019/07/16 19:33:04 631F3FDF r (emu) calculated cw 5: E58915898967923B
2019/07/16 19:33:04 631F3FDF r (emu) calculated cw 6: 2FE55D38CBD0520E
2019/07/16 19:33:04 631F3FDF r (emu) calculated cw 7: 76B04C3DD6CD0B57
2019/07/16 19:33:04 70CD83BB c (ecm) dvbapi (0E00@000000/0000/0098/48:2D64FD394A0CC3081571A9EF19C24B89): found (15 ms) by emulator - TC Action HD```

nautilus7, 2024-03-26:
@WXbet is the commit linked here somehow related to powervu?

ohmza, 2024-03-26:
@nautilus7 That code is not applicable for the current trunk anymore. As there was a major algorithm change on CAK7 since r11735.

nautilus7, 2024-03-26:
I understand this, I do not understand why someone would link that commit with this issue though...

nautilus7, 2024-03-31:
Feel free to send us any powervu code to add...

closed 2024-03-31 by nautilus7