oscam-emu/oscam-patched-old #49: Build error on Windows (Cygwin)

sat46, 2019-05-31:
Hello :)

Seems that because of changes between 1.0 and 1.1 releases of libssl, I'm unable to build oscam-patched on Windows (Cygwin, x64).

```
[ 98%] Building C object CMakeFiles/oscam.dir/oscam.o
[ 99%] Linking C executable oscam.exe
libcsmodules.a(module-emulator-biss.o):module-emulator-biss.c:(.text+0x17da): undefined reference to `EVP_MD_CTX_new'
libcsmodules.a(module-emulator-biss.o):module-emulator-biss.c:(.text+0x17da): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `EVP_MD_CTX_new'
libcsmodules.a(module-emulator-biss.o):module-emulator-biss.c:(.text+0x1822): undefined reference to `EVP_MD_CTX_free'
libcsmodules.a(module-emulator-biss.o):module-emulator-biss.c:(.text+0x1822): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `EVP_MD_CTX_free'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/oscam.dir/build.make:102: oscam.exe] Error 1
make[1]: *** [CMakeFiles/Makefile2:186: CMakeFiles/oscam.dir/all] Error 2
make: *** [Makefile:150: all] Error 2
```

I've tried to manually upgrade OpenSSL library and headers with these ones: https://bintray.com/vszakats/generic/openssl but that didn't help.

Thanks for any suggestions how to fix that.

oscam-emu, 2019-05-31:
I am not sure what is the problem. [In openssl 1.1.1](https://www.openssl.org/docs/man1.1.1/man3/EVP_MD_CTX_new.html), both `EVP_MD_CTX_new` and `EVP_MD_CTX_free` are available. Maybe something related to cygwin?

sat46, 2019-05-31:
Thanks for reply :)

In module-emulator-biss.c there are still EVP_MD_CTX_create() and EVP_MD_CTX_destroy() in use (old ones).
That made me realize that it was indeed a cygwin issue - seems that when doing an ordinary cygwin upgrade (I'm trying to keep it up to date on weekly basis) I've unintentionally upgraded libssl to 1.1. And my problem was not too old libssl, but too new ;)

**Solution is to roll back libssl to 1.0** - all OK now :)

If anybody else will run across this issue - roll back libssl to 1.0 :)

Sorry for inconvenience.

closed 2019-05-31 by sat46

oscam-emu, 2019-06-01:
[Even in current openssl source code](https://github.com/openssl/openssl/blob/6aa2e59e1c52761cc5ad60170106118d7c1aa090/include/openssl/evp.h#L546), there are `#define` aliases for the older `EVP_MD_CTX_create` and `EVP_MD_CTX_destroy` functions, so the oscam-emu code should work whatever version you use.

I believe your problem was caused by having the wrong version of openssl (wrong architecture) in your toolchain.