oscam-emu/oscam-patched-old #50: Everytime a PowerVU key is updated a line is added in SoftCam.Key

littlesat, 2019-06-26:
Can't we better adapt the line we already have? 

now->
P CCCCCCCC GROUP 0900 ; 11804 V, 9.0°EP 009E0078 00 ED9E4C6374A9AF
P 0900FFFF 00 AAAAAAAAAAAAAA
P 0900FFFF 01 BBBBBBBBBBBBBB
P 0900 XXXXXXXX QQQQQQQQQQQQQQ ;EMM Key

P 0900FFFF 00 DDDDDDDDDDDDDD ; added by Emu Sun Jun 23 23:53:08 2019 UA: XXXXXXXX
P 0900FFFF 01 EEEEEEEEEEEEEE ; added by Emu Sun Jun 23 23:53:08 2019 UA: XXXXXXXX

suggested->
P CCCCCCCC GROUP 0900 ; 11804 V, 9.0°EP 009E0078 00 ED9E4C6374A9AF
P 0900FFFF 00 DDDDDDDDDDDDDD ; updated by Emu Sun Jun 31 23:53:08 2019 UA: XXXXXXXX
P 0900FFFF 01 EEEEEEEEEEEEEE ; updated by Emu Sun Jun 31 23:53:08 2019 UA: XXXXXXXX
P 0900 XXXXXXXX QQQQQQQQQQQQQQ ;EMM Key

The file now grows for ever and the old keys are not needed anymore...

oscam-emu, 2019-06-26:
The new method (1 ecm key per transponder) is based on enigma namespace. Unfortunately, this info is not available to other clients, like non enigma2 stbs, pc clients, etc.

So, this means that oscam-emu will have to do different things based on the ecm key it is about to write in the file. This doesn't seem nice.

Also, before this "new method", way more keys were written and the softcam.key file was growing much faster. Now, only 1-2 ecm keys are written which keeps the file size manageable.

OscamSatUser, 2019-11-19:
I wrote code that updates the softcam.key.  It grabs the latest key, and updates the original ECM lines and on the EMM key line, it modifies the comment to show when that particular emm was last updated, that way i know which EMM keys are being utilized, and which have not.

P 6810426B GROUP 6051 ; 58:3881 
P 6051FFFF 00 NEWKEYNEWKEY ; 58:3881 
P 6051FFFF 01 NEWKEYNEWKEY ; 58:3881 
P 6051 00EMM01 XXXXXXXXXE850E ; EMM 58:3881 
P 6051 00EMM02 XXXXXXXXX46EFB ; EMM 58:3881 #Oscam 11-19-2019 by UA: 00EMM02
P 6051 00EMM03 XXXXXXXXXA3775 ; EMM 58:3881
P 6051 00EMM04 XXXXXXXXX7B0EC ; EMM 58:3881 #Oscam 10-27-2019 by UA: 00EMM04

closed 2024-04-14 by nautilus7