oscam-emu/oscam-patched-old #51: AFN channels only sound no picture

jack3009, 2019-07-09:
Since today no more picture on AFN channels only sound sunday it was all working fine, using latest oscam-emu version 11529-emu 796 ?
Must there be something changed in the emu version or the emu reader in oscam ?

oscam-emu, 2019-07-09:
There is no problem with these channels. Nothing has changed and in the last weeks regarding the setup. Check your config and post a log with debug level 2 + 128.

jack3009, 2019-07-09:
Mmmm very strange using the same config when it works fine. 
debug level 2: 
2019/07/09 20:01:38 76E9FFF3 c   (dvbapi) Demuxer 0 filter2[0] = 80, filter mask[0] = FF, flt&mask = 80 , buffer[0] = 80, buffer[0] & mask = 80
2019/07/09 20:01:38 76E9FFF3 c   (dvbapi) Demuxer 0 Filter 2 fetched ECM data (ecmlength = 0x099):
2019/07/09 20:01:38 76E9FFF3 c   (dvbapi)   80 30 96 50 3F 20 0E 00 EC 08 BF BC 6E 23 04 73 
2019/07/09 20:01:38 76E9FFF3 c   (dvbapi)   7B AC 00 48 B0 80 8C A8 8B 94 08 2C 7B DB E3 89 
2019/07/09 20:01:38 76E9FFF3 c   (dvbapi)   7E A1 71 D5 6B 06 C0 00 00 30 A5 C2 B0 E0 75 30 
2019/07/09 20:01:38 76E9FFF3 c   (dvbapi)   06 41 D4 F8 8C 24 CC D2 85 34 2A C4 6E 37 A4 D1 
2019/07/09 20:01:38 76E9FFF3 c   (dvbapi)   60 B8 37 A3 30 10 27 0E 00 00 00 80 00 8B BF 8D 
2019/07/09 20:01:38 76E9FFF3 c   (dvbapi)   E7 7C 0E 03 21 14 30 10 27 0E 00 00 00 20 00 7D 
2019/07/09 20:01:38 76E9FFF3 c   (dvbapi)   D9 5C BA E0 EB 83 58 CE 30 0D 28 0E 00 00 00 89 
2019/07/09 20:01:38 76E9FFF3 c   (dvbapi)   70 C9 5E 5A C9 CF 88 30 1C 29 0E 00 00 00 2E BD 
2019/07/09 20:01:38 76E9FFF3 c   (dvbapi)   82 35 EE 16 93 48 30 F4 1F 91 42 31 B5 82 AD 4E 
2019/07/09 20:01:38 76E9FFF3 c   (dvbapi)   96 C8 05 66 F3 4F F8 C2 EF 
2019/07/09 20:01:38 76E9FFF3 c   (dvbapi) Demuxer 0 filter2[0] = 80, filter mask[0] = FF, flt&mask = 80 , buffer[0] = 80, buffer[0] & mask = 80
2019/07/09 20:01:38 76E9FFF3 c   (dvbapi) Demuxer 0 Filter 2 fetched ECM data (ecmlength = 0x099):
2019/07/09 20:01:38 76E9FFF3 c   (dvbapi)   80 30 96 50 3F 20 0E 00 EC 08 BF BC 6E 23 04 73 
2019/07/09 20:01:38 76E9FFF3 c   (dvbapi)   7B AC 00 48 B0 80 8C A8 8B 94 08 2C 7B DB E3 89 
2019/07/09 20:01:38 76E9FFF3 c   (dvbapi)   7E A1 71 D5 6B 06 C0 00 00 30 A5 C2 B0 E0 75 30 
2019/07/09 20:01:38 76E9FFF3 c   (dvbapi)   06 41 D4 F8 8C 24 CC D2 85 34 2A C4 6E 37 A4 D1 
2019/07/09 20:01:38 76E9FFF3 c   (dvbapi)   60 B8 37 A3 30 10 27 0E 00 00 00 80 00 8B BF 8D 
2019/07/09 20:01:38 76E9FFF3 c   (dvbapi)   E7 7C 0E 03 21 14 30 10 27 0E 00 00 00 20 00 7D 
2019/07/09 20:01:38 76E9FFF3 c   (dvbapi)   D9 5C BA E0 EB 83 58 CE 30 0D 28 0E 00 00 00 89 
2019/07/09 20:01:38 76E9FFF3 c   (dvbapi)   70 C9 5E 5A C9 CF 88 30 1C 29 0E 00 00 00 2E BD 
2019/07/09 20:01:38 76E9FFF3 c   (dvbapi)   82 35 EE 16 93 48 30 F4 1F 91 42 31 B5 82 AD 4E 
2019/07/09 20:01:38 76E9FFF3 c   (dvbapi)   96 C8 05 66 F3 4F F8 C2 EF 
2019/07/09 20:01:38 76E9FFF3 c   (dvbapi) Demuxer 0 filter2[0] = 80, filter mask[0] = FF, flt&mask = 80 , buffer[0] = 80, buffer[0] & mask = 80
2019/07/09 20:01:38 76E9FFF3 c   (dvbapi) Demuxer 0 Filter 2 fetched ECM data (ecmlength = 0x099):
2019/07/09 20:01:38 76E9FFF3 c   (dvbapi)   80 30 96 50 3F 20 0E 00 EC 08 BF BC 6E 23 04 73 
2019/07/09 20:01:38 76E9FFF3 c   (dvbapi)   7B AC 00 48 B0 80 8C A8 8B 94 08 2C 7B DB E3 89 
2019/07/09 20:01:38 76E9FFF3 c   (dvbapi)   7E A1 71 D5 6B 06 C0 00 00 30 A5 C2 B0 E0 75 30 
2019/07/09 20:01:38 76E9FFF3 c   (dvbapi)   06 41 D4 F8 8C 24 CC D2 85 34 2A C4 6E 37 A4 D1 
2019/07/09 20:01:38 76E9FFF3 c   (dvbapi)   60 B8 37 A3 30 10 27 0E 00 00 00 80 00 8B BF 8D 
2019/07/09 20:01:38 76E9FFF3 c   (dvbapi)   E7 7C 0E 03 21 14 30 10 27 0E 00 00 00 20 00 7D 
2019/07/09 20:01:38 76E9FFF3 c   (dvbapi)   D9 5C BA E0 EB 83 58 CE 30 0D 28 0E 00 00 00 89 
2019/07/09 20:01:38 76E9FFF3 c   (dvbapi)   70 C9 5E 5A C9 CF 88 30 1C 29 0E 00 00 00 2E BD 
2019/07/09 20:01:38 76E9FFF3 c   (dvbapi)   82 35 EE 16 93 48 30 F4 1F 91 42 31 B5 82 AD 4E 
2019/07/09 20:01:38 76E9FFF3 c   (dvbapi)   96 C8 05 66 F3 4F F8 C2 EF 
2019/07/09 20:01:39 76E9FFF3 c   (dvbapi) Demuxer 0 filter2[0] = 80, filter mask[0] = FF, flt&mask = 80 , buffer[0] = 80, buffer[0] & mask = 80
2019/07/09 20:01:39 76E9FFF3 c   (dvbapi) Demuxer 0 Filter 2 fetched ECM data (ecmlength = 0x099):
2019/07/09 20:01:39 76E9FFF3 c   (dvbapi)   80 30 96 50 3F 20 0E 00 EC 08 BF BC 6E 23 04 73 
2019/07/09 20:01:39 76E9FFF3 c   (dvbapi)   7B AC 00 48 B0 80 8C A8 8B 94 08 2C 7B DB E3 89 
2019/07/09 20:01:39 76E9FFF3 c   (dvbapi)   7E A1 71 D5 6B 06 C0 00 00 30 A5 C2 B0 E0 75 30 
2019/07/09 20:01:39 76E9FFF3 c   (dvbapi)   06 41 D4 F8 8C 24 CC D2 85 34 2A C4 6E 37 A4 D1 
2019/07/09 20:01:39 76E9FFF3 c   (dvbapi)   60 B8 37 A3 30 10 27 0E 00 00 00 80 00 8B BF 8D 
2019/07/09 20:01:39 76E9FFF3 c   (dvbapi)   E7 7C 0E 03 21 14 30 10 27 0E 00 00 00 20 00 7D 
2019/07/09 20:01:39 76E9FFF3 c   (dvbapi)   D9 5C BA E0 EB 83 58 CE 30 0D 28 0E 00 00 00 89 
2019/07/09 20:01:39 76E9FFF3 c   (dvbapi)   70 C9 5E 5A C9 CF 88 30 1C 29 0E 00 00 00 2E BD 
2019/07/09 20:01:39 76E9FFF3 c   (dvbapi)   82 35 EE 16 93 48 30 F4 1F 91 42 31 B5 82 AD 4E 
2019/07/09 20:01:39 76E9FFF3 c   (dvbapi)   96 C8 05 66 F3 4F F8 C2 EF 
2019/07/09 20:01:39 76E9FFF3 c   (dvbapi) Demuxer 0 filter2[0] = 80, filter mask[0] = FF, flt&mask = 80 , buffer[0] = 80, buffer[0] & mask = 80
2019/07/09 20:01:39 76E9FFF3 c   (dvbapi) Demuxer 0 Filter 2 fetched ECM data (ecmlength = 0x099):
2019/07/09 20:01:39 76E9FFF3 c   (dvbapi)   80 30 96 50 3F 20 0E 00 EC 08 BF BC 6E 23 04 73 
2019/07/09 20:01:39 76E9FFF3 c   (dvbapi)   7B AC 00 48 B0 80 8C A8 8B 94 08 2C 7B DB E3 89 
2019/07/09 20:01:39 76E9FFF3 c   (dvbapi)   7E A1 71 D5 6B 06 C0 00 00 30 A5 C2 B0 E0 75 30 
2019/07/09 20:01:39 76E9FFF3 c   (dvbapi)   06 41 D4 F8 8C 24 CC D2 85 34 2A C4 6E 37 A4 D1 
2019/07/09 20:01:39 76E9FFF3 c   (dvbapi)   60 B8 37 A3 30 10 27 0E 00 00 00 80 00 8B BF 8D 
2019/07/09 20:01:39 76E9FFF3 c   (dvbapi)   E7 7C 0E 03 21 14 30 10 27 0E 00 00 00 20 00 7D 
[... cut ...]
2019/07/09 20:03:22 76E9FFF3 c   (dvbapi) Demuxer 0 handles 25 emm filters
2019/07/09 20:03:22 7EBF6402 h    (webif) all debug_level=0
2019/07/09 20:03:27 76E9FFF3 c      (ecm) Peke (0E00@000000/0000/006C/99:3A6CFE8E1470AC4CCE1A97CC37ED309B): found (28 ms) by AFN-EMU - AFN
2019/07/09 20:03:37 76E9FFF3 c      (ecm) Peke (0E00@000000/0000/006C/99:48B08EBC51C1A8CDFABF981FD2C4580C): found (26 ms) by AFN-EMU - AFN
2019/07/09 20:03:47 76E9FFF3 c      (ecm) Peke (0E00@000000/0000/006C/99:07A9052F576527A3E15EC24E9ED60025): found (17 ms) by AFN-EMU - AFN

oscam-emu, 2019-07-09:
I meant 1 log with both 2+128 debug level. Anyway, I can see that everything is ok from the logs.

Maybe you're using a separate emu reader for these channels? Just leave 1 emu reader for all channels.

jack3009, 2019-07-09:
No i work with 1 emu reader but I switched the image from OBH to BH and now they are working with BH.
So its good enough for me now. Anyway thx for the reply's.

jack3009, 2019-07-09:
>> OSCam <<  cardserver log switched at Tue Jul  9 21:04:47 2019
-------------------------------------------------------------------------------
2019/07/09 21:04:47 2E162DD9 h >> OSCam << cardserver log switched, version 1.20_svn_cw_64_hardcoded, build r11518-796 (armv7-dm9xx-linux-gnueabihf-ssl-libusb)
2019/07/09 21:04:47 2FBF4F1A c   (dvbapi) Demuxer 0 filter2[0] = 80, filter mask[0] = FF, flt&mask = 80 , buffer[0] = 80, buffer[0] & mask = 80
2019/07/09 21:04:47 2FBF4F1A c   (dvbapi) Demuxer 0 Filter 2 fetched ECM data (ecmlength = 0x099):
2019/07/09 21:04:47 2FBF4F1A c   (dvbapi)   80 30 96 50 3F 20 0E 00 57 08 19 3A 8A E8 DB B8 
2019/07/09 21:04:47 2FBF4F1A c   (dvbapi)   9F B6 00 28 A0 80 DC 6B 03 70 83 93 56 E3 C7 21 
2019/07/09 21:04:47 2FBF4F1A c   (dvbapi)   2B 12 14 AE D2 06 D0 00 00 CE 1F 5D 1F 03 AC 73 
2019/07/09 21:04:47 2FBF4F1A c   (dvbapi)   CB E1 BF 7B 02 5A E3 15 01 CB D2 4C 49 19 74 B3 
2019/07/09 21:04:47 2FBF4F1A c   (dvbapi)   41 E4 EF 6D 30 10 27 0E 00 00 00 80 00 14 C5 4A 
2019/07/09 21:04:47 2FBF4F1A c   (dvbapi)   21 13 61 00 F2 D6 30 10 27 0E 00 00 00 20 00 EC 
2019/07/09 21:04:47 2FBF4F1A c   (dvbapi)   11 E0 63 28 67 31 F8 E8 30 0D 28 0E 00 00 00 C6 
2019/07/09 21:04:47 2FBF4F1A c   (dvbapi)   F5 60 64 1E A8 A6 93 30 1C 29 0E 00 00 00 11 06 
2019/07/09 21:04:47 2FBF4F1A c   (dvbapi)   27 99 AE 10 69 6B F6 53 5B 01 3E 7A DD 52 7E 1C 
2019/07/09 21:04:47 2FBF4F1A c   (dvbapi)   CE 36 AF C0 EA 3C 00 0E 23 
2019/07/09 21:04:47 2FBF4F1A c   (dvbapi) Demuxer 0 filter2[0] = 80, filter mask[0] = FF, flt&mask = 80 , buffer[0] = 80, buffer[0] & mask = 80
2019/07/09 21:04:47 2FBF4F1A c   (dvbapi) Demuxer 0 Filter 2 fetched ECM data (ecmlength = 0x099):
2019/07/09 21:04:47 2FBF4F1A c   (dvbapi)   80 30 96 50 3F 20 0E 00 57 08 19 3A 8A E8 DB B8 
2019/07/09 21:04:47 2FBF4F1A c   (dvbapi)   9F B6 00 28 A0 80 DC 6B 03 70 83 93 56 E3 C7 21 
2019/07/09 21:04:47 2FBF4F1A c   (dvbapi)   2B 12 14 AE D2 06 D0 00 00 CE 1F 5D 1F 03 AC 73 
2019/07/09 21:04:47 2FBF4F1A c   (dvbapi)   CB E1 BF 7B 02 5A E3 15 01 CB D2 4C 49 19 74 B3 
2019/07/09 21:04:47 2FBF4F1A c   (dvbapi)   41 E4 EF 6D 30 10 27 0E 00 00 00 80 00 14 C5 4A 
2019/07/09 21:04:47 2FBF4F1A c   (dvbapi)   21 13 61 00 F2 D6 30 10 27 0E 00 00 00 20 00 EC 
2019/07/09 21:04:47 2FBF4F1A c   (dvbapi)   11 E0 63 28 67 31 F8 E8 30 0D 28 0E 00 00 00 C6 
2019/07/09 21:04:47 2FBF4F1A c   (dvbapi)   F5 60 64 1E A8 A6 93 30 1C 29 0E 00 00 00 11 06 
2019/07/09 21:04:47 2FBF4F1A c   (dvbapi)   27 99 AE 10 69 6B F6 53 5B 01 3E 7A DD 52 7E 1C 
2019/07/09 21:04:47 2FBF4F1A c   (dvbapi)   CE 36 AF C0 EA 3C 00 0E 23 
2019/07/09 21:04:47 2FBF4F1A c   (dvbapi) Demuxer 0 filter2[0] = 80, filter mask[0] = FF, flt&mask = 80 , buffer[0] = 80, buffer[0] & mask = 80
2019/07/09 21:04:47 2FBF4F1A c   (dvbapi) Demuxer 0 Filter 2 fetched ECM data (ecmlength = 0x099):
2019/07/09 21:04:47 2FBF4F1A c   (dvbapi)   80 30 96 50 3F 20 0E 00 57 08 19 3A 8A E8 DB B8 
2019/07/09 21:04:47 2FBF4F1A c   (dvbapi)   9F B6 00 28 A0 80 DC 6B 03 70 83 93 56 E3 C7 21 
2019/07/09 21:04:47 2FBF4F1A c   (dvbapi)   2B 12 14 AE D2 06 D0 00 00 CE 1F 5D 1F 03 AC 73 
2019/07/09 21:04:47 2FBF4F1A c   (dvbapi)   CB E1 BF 7B 02 5A E3 15 01 CB D2 4C 49 19 74 B3 
2019/07/09 21:04:47 2FBF4F1A c   (dvbapi)   41 E4 EF 6D 30 10 27 0E 00 00 00 80 00 14 C5 4A 
2019/07/09 21:04:47 2FBF4F1A c   (dvbapi)   21 13 61 00 F2 D6 30 10 27 0E 00 00 00 20 00 EC 
2019/07/09 21:04:47 2FBF4F1A c   (dvbapi)   11 E0 63 28 67 31 F8 E8 30 0D 28 0E 00 00 00 C6 
2019/07/09 21:04:47 2FBF4F1A c   (dvbapi)   F5 60 64 1E A8 A6 93 30 1C 29 0E 00 00 00 11 06 
2019/07/09 21:04:47 2FBF4F1A c   (dvbapi)   27 99 AE 10 69 6B F6 53 5B 01 3E 7A DD 52 7E 1C 
2019/07/09 21:04:47 2FBF4F1A c   (dvbapi)   CE 36 AF C0 EA 3C 00 0E 23 
2019/07/09 21:04:48 2FBF4F1A c   (dvbapi) Demuxer 0 filter2[0] = 80, filter mask[0] = FF, flt&mask = 80 , buffer[0] = 80, buffer[0] & mask = 80
2019/07/09 21:04:48 2FBF4F1A c   (dvbapi) Demuxer 0 Filter 2 fetched ECM data (ecmlength = 0x099):
2019/07/09 21:04:48 2FBF4F1A c   (dvbapi)   80 30 96 50 3F 20 0E 00 57 08 19 3A 8A E8 DB B8 
2019/07/09 21:04:48 2FBF4F1A c   (dvbapi)   9F B6 00 28 A0 80 DC 6B 03 70 83 93 56 E3 C7 21 
2019/07/09 21:04:48 2FBF4F1A c   (dvbapi)   2B 12 14 AE D2 06 D0 00 00 CE 1F 5D 1F 03 AC 73 
2019/07/09 21:04:48 2FBF4F1A c   (dvbapi)   CB E1 BF 7B 02 5A E3 15 01 CB D2 4C 49 19 74 B3 
2019/07/09 21:04:48 2FBF4F1A c   (dvbapi)   41 E4 EF 6D 30 10 27 0E 00 00 00 80 00 14 C5 4A 
2019/07/09 21:04:48 2FBF4F1A c   (dvbapi)   21 13 61 00 F2 D6 30 10 27 0E 00 00 00 20 00 EC 
2019/07/09 21:04:48 2FBF4F1A c   (dvbapi)   11 E0 63 28 67 31 F8 E8 30 0D 28 0E 00 00 00 C6 
2019/07/09 21:04:48 2FBF4F1A c   (dvbapi)   F5 60 64 1E A8 A6 93 30 1C 29 0E 00 00 00 11 06 
2019/07/09 21:04:48 2FBF4F1A c   (dvbapi)   27 99 AE 10 69 6B F6 53 5B 01 3E 7A DD 52 7E 1C 
2019/07/09 21:04:48 2FBF4F1A c   (dvbapi)   CE 36 AF C0 EA 3C 00 0E 23 
2019/07/09 21:04:48 2FBF4F1A c   (dvbapi) Demuxer 0 filter2[0] = 80, filter mask[0] = FF, flt&mask = 80 , buffer[0] = 80, buffer[0] & mask = 80
2019/07/09 21:04:48 2FBF4F1A c   (dvbapi) Demuxer 0 Filter 2 fetched ECM data (ecmlength = 0x099):
2019/07/09 21:04:48 2FBF4F1A c   (dvbapi)   80 30 96 50 3F 20 0E 00 57 08 19 3A 8A E8 DB B8 
2019/07/09 21:04:48 2FBF4F1A c   (dvbapi)   9F B6 00 28 A0 80 DC 6B 03 70 83 93 56 E3 C7 21 
2019/07/09 21:04:48 2FBF4F1A c   (dvbapi)   2B 12 14 AE D2 06 D0 00 00 CE 1F 5D 1F 03 AC 73 
[... cut ...]
2019/07/09 21:04:54 2FBF4F1A c   (dvbapi)   AF 58 B6 6F 63 D8 30 10 27 0E 00 00 00 20 00 3A 
2019/07/09 21:04:54 2FBF4F1A c   (dvbapi)   54 6B 21 01 04 04 01 BC 30 0D 28 0E 00 00 00 90 
2019/07/09 21:04:54 2FBF4F1A c   (dvbapi)   01 75 22 5E 19 94 A1 30 1C 29 0E 00 00 00 C2 37 
2019/07/09 21:04:54 2FBF4F1A c   (dvbapi)   DB 16 66 27 31 8A 5F EC 3D 4A 7A C2 AC BD 19 E6 
2019/07/09 21:04:54 2FBF4F1A c   (dvbapi)   83 7F 28 40 86 29 6E 5C 27 
2019/07/09 21:04:54 2E162DD9 h    (webif) all debug_level=128

oscam-emu, 2019-07-09:
Then maybe there is a problem with the drivers in your old image. Maybe they do not support the extended cw.

closed 2019-07-09 by oscam-emu

jack3009, 2019-07-09:
Old image ? I'm using always the latest versions. Latest OBH is 4.2.047 and BH 3.0.9B

oscam-emu, 2019-07-09:
I meant "old" like "previous".