oscam-emu/oscam-patched-old #56: Question about Hash

nachopr43, 2019-08-02:
Thanks for all the help.  Why the channel hash seems to change sometimes after a re-scan ?

oscam-emu, 2019-08-05:
The only reason I can think of is because the frequency is changed slightly (by 1 or 2 MHz).

nachopr43, 2019-08-07:
"Since the frequency factors into the hash would it be possible to add code to this effect ?

Calculate hash using current frequency.
Check softcam for match.
If no match, Calculate hash using current frequency -1.
Check softcam for match.
If no match, Calculate hash using current frequency +1.
Check softcam for match.
If no match, Calculate hash using current frequency -2
Check softcam for match.
If no match, Calculate hash using current frequency +2.
Check softcam for match.

This should make hash values more universal and "durable" such that only extreme LNB drift would cause the user to have to redo softcam hash entries."

oscam-emu, 2019-08-07:
The frequency change is caused by the user, e.g if you set a slightly different frequency when scanning or use a different satellite.xml file. It's not the actual frequency that matters, so the LNB drifting does not matter.

Your suggestion (+/- 2 MHz) is used in BISS namespace hash. In feeds, it does make sense to look for adjacent frequencies because almost every time the scan parameters are entered by the user (they are not present in the satellite.xml).

nachopr43, 2019-08-07:
Thanks for quick answer. Unfortunately, there is no effort to maintain sat info here in North America. Satellite.xml info is often wrong, outdated or just missing. That's why Blind scanning is the preferred method and that's how we get hash mismatches with other users and ourselves after a rescan.

closed 2024-04-14 by nautilus7