oscam-emu/oscam-patched-old #57: BBC 27.5W with EMU doesn't work

Monello2, 2019-08-05:
have done the following: I have entered the keys as follows in Softcam. key. 
 

F 10440065 00 D581F94F24112D62 ;BBC One London (27.5°W)
F 10440065 01 D581F94F24112D62 ;BBC One London (27.5°W)
F 107C1005 00 D581F94F24112D62 ;BBC One Scotland (27.5°W)
F 107C1005 01 D581F94F24112D62 ;BBC One Scotland (27.5°W)
F 107D13ED 00 D581F94F24112D62 ;BBC 1 Northern Ireland (27.5°W)
F 107D13ED 01 D581F94F24112D62 ;BBC 1 Northern Ireland (27.5°W)
F 107E0C1D 00 D581F94F24112D62 ;BBC One Wales (27.5°W)
F 107E0C1D 01 D581F94F24112D62 ;BBC One Wales (27.5°W)
F 10BF00C9 00 D581F94F24112D62 ;BBC TWO England (27.5°W)
F 10BF00C9 01 D581F94F24112D62 ;BBC TWO England (27.5°W)
F 110001F5 00 D581F94F24112D62 ;BBC NEWS (27.5°W)
F 110001F5 01 D581F94F24112D62 ;BBC NEWS (27.5°W)
F 11C00191 00 D581F94F24112D62 ;BBC FOUR (27.5°W)
F 11C00191 01 D581F94F24112D62 ;BBC FOUR (27.5°W)
F 1200012D 00 D581F94F24112D62 ;CBBC (27.5°W)
F 1200012D 01 D581F94F24112D62 ;CBBC (27.5°W)
F 12400191 00 D581F94F24112D62 ;CBeebies (27.5°W)
F 12400191 01 D581F94F24112D62 ;CBeebies (27.5°W)
F 12800259 00 D581F94F24112D62 ;BBC Parliament (27.5°W)
F 12800259 01 D581F94F24112D62 ;BBC Parliament (27.5°W)
F 1C000385 00 D581F94F24112D62 ;BBC Red Button (27.5°W)
F 1C000385 01 D581F94F24112D62 ;BBC Red Button (27.5°W)
F 444017D5 00 D581F94F24112D62 ;BBC Two HD (27.5°W)
F 444017D5 01 D581F94F24112D62 ;BBC Two HD (27.5°W)
F 448419C9 00 D581F94F24112D62 ;BBC One HD (27.5°W)
F 448419C9 01 D581F94F24112D62 ;BBC One HD (27.5°W)
F 44C81839 00 D581F94F24112D62 ;ITV1 HD (27.5°W)
F 44C81839 01 D581F94F24112D62 ;ITV1 HD (27.5°W)
F 4500189D 00 D581F94F24112D62 ;Channel 4 London HD (27.5°W)
F 4500189D 01 D581F94F24112D62 ;Channel 4 London HD (27.5°W)
F 45401901 00 D581F94F24112D62 ;Channel 5 HD (27.5°W)
F 45401901 01 D581F94F24112D62 ;Channel 5 HD (27.5°W)
F 46C01B59 00 D581F94F24112D62 ;CBBC HD (27.5°W)
F 44C01B59 01 D581F94F24112D62 ;CBBC HD (27.5°W)
F 4FC01D4D 00 D581F94F24112D62 ;More 4 UK +1 (27.5°W)
F 4FC01D4D 01 D581F94F24112D62 ;More 4 UK +1 (27.5°W)
---------------------------------------------
F 16001FFF 00 D581F94F24112D62 ;
F 16001FFF 01 D581F94F24112D62 ;BBC Radio Five Live (27.5°W)
F 16401FFF 00 D581F94F24112D62 ;
F 16401FFF 01 D581F94F24112D62 ;BBC Radio Five Live Sports Extra (27.5°W)
F 16801FFF 00 D581F94F24112D62 ;
F 16801FFF 01 D581F94F24112D62 ;BBC 6 Music (27.5°W)
F 16C01FFF 00 D581F94F24112D62 ;
F 16C01FFF 01 D581F94F24112D62 ;BBC 4 Xtra (27.5°W)
F 17001FFF 00 D581F94F24112D62 ;
F 17001FFF 01 D581F94F24112D62 ;BBC 1 Xtra (27.5°W)
F 17411FFF 00 D581F94F24112D62 ;
F 17411FFF 01 D581F94F24112D62 ;BBC Asian Network (27.5°W)
F 17801FFF 00 D581F94F24112D62 ;
F 17801FFF 01 D581F94F24112D62 ;BBC World Service Europe (27.5°W)
F 1A401FFF 00 D581F94F24112D62 ;
F 1A401FFF 01 D581F94F24112D62 ;BBC Radio 1 (27.5°W)
F 1A801FFF 00 D581F94F24112D62 ;
F 1A801FFF 01 D581F94F24112D62 ;BBC Radio 2 (27.5°W)
F 1AC01FFF 00 D581F94F24112D62 ;
F 1AC01FFF 01 D581F94F24112D62 ;BBC Radio 3 (27.5°W)
F 1B001FFF 00 D581F94F24112D62 ;
F 1B001FFF 01 D581F94F24112D62 ;BBC Radio 4 (27.5°W)
 
Then I entered the EMU reader in oscam. server as follows:
[reader]
label = emulator
protocol = emu
device = emulator
disablecrccws_only_for = 0E00:000000
caid = 0500,0604,090F,0E00,1010,1801,2600,2602,2610,4AE1
detect = cd
ident = 0500:000000,023800,021110,007400,007800;0604:000000;090F:000000;0E00:000000;1010:000000;1801:000000,007301,001101,002111;2600:000000;2602:000000;2610:000000;4AE1:000011,000014,0000FE
group = 1
emmcache = 2,1,2,1
emu_auproviders = 0604:010200;0E00:000000;1010:000000;2610:000000;4AE1:000011,000014,0000FE
emu_datecodedenabled = 0
 
 
and then copied the file softcam.key to etc/tuxbox/config
 
 
I also edited the lamedb file, entered all PIDs, stopped enigma with "init4", copied the lamedb to etc/engima and restarted enigma with "init3". 
 
Unfortunately, everything remains dark

oscam-emu, 2019-08-05:
I can't tell if it's oscam-emu or enigma2 in fault. I need a log from oscam to see if it's actually asking for a key or not.

Monello2, 2019-08-05:
---------------------------------------------------------
>> OSCam <<  cardserver log switched at Tue Aug  6 00:05:12 2019
-------------------------------------------------------------------------------
2019/08/06 00:05:12 5C2F7A01 r   (reader) SRG-V6 [smartreader] Maturity level [F]=no age limit
2019/08/06 00:05:12 5C2F7A01 r   (reader) SRG-V6 [smartreader] ready for requests
2019/08/06 00:05:12 5C2F7A01 r   (reader) SRG-V6 [smartreader] found card system viaccess
2019/08/06 00:05:12 5C2F7A01 r   (reader) SRG-V6 [viaccess] THIS WAS A SUCCESSFUL START ATTEMPT No  1 out of max allotted of 1
2019/08/06 00:05:12 5C2F7A01 r   (reader) SRG-V6 [viaccess] card detected
2019/08/06 00:05:12 00000000   >> OSCam << cardserver log switched, version build-DreamOSat-1.20_svn, build r11536-797 (arm-linux-gnueabihf-ssl-libusb)
2019/08/06 00:05:12 5C2F7A01 r   (reader) SRG-V6 [viaccess] serial: ##########
2019/08/06 00:05:12 5C2F7A01 r   (reader) SRG-V6 [viaccess] provider: 1, id: ####################, sa: ########, geo: empty
2019/08/06 00:05:13 5C2F7A01 r   (reader) SRG-V6 [viaccess] provider: 2, id: ####################, sa: ########, geo: FF FF FF FF 
2019/08/06 00:05:13 5C2F7A01 r   (reader) SRG-V6 [viaccess] class: 01, expiry date: 2019/08/01 - 2019/09/15
2019/08/06 00:05:13 5C2F7A01 r   (reader) SRG-V6 [viaccess] class: 01, expiry date: 2019/06/27 - 2019/08/11
2019/08/06 00:05:13 5C2F7A01 r   (reader) SRG-V6 [viaccess] class: 01, expiry date: 2019/05/23 - 2019/07/07
2019/08/06 00:05:13 5C2F7A01 r   (reader) SRG-V6 [viaccess] class: 01, expiry date: 2019/04/18 - 2019/06/02
2019/08/06 00:05:13 5C2F7A01 r   (reader) SRG-V6 [viaccess] class: 01, expiry date: 2019/03/14 - 2019/04/28
2019/08/06 00:05:13 5C2F7A01 r   (reader) SRG-V6 [viaccess] class: 01, expiry date: 2019/02/07 - 2019/03/24
2019/08/06 00:05:14 5C2F7A01 r   (reader) SRG-V6 [viaccess] class: 01, expiry date: 2019/01/03 - 2019/02/17
2019/08/06 00:05:14 5C2F7A01 r   (reader) SRG-V6 [viaccess] class: 01, expiry date: 2018/11/29 - 2019/01/13
2019/08/06 00:05:14 5C2F7A01 r   (reader) SRG-V6 [viaccess] class: 01, expiry date: 2018/10/25 - 2018/12/09
2019/08/06 00:05:14 5C2F7A01 r   (reader) SRG-V6 [viaccess] class: 01, expiry date: 2018/09/20 - 2018/11/04
2019/08/06 00:05:14 5C2F7A01 r   (reader) SRG-V6 [viaccess] class: 01, expiry date: 2018/08/26 - 2018/09/26
2019/08/06 00:05:14 5C2F7A01 r   (reader) SRG-V6 [viaccess] class: 02, expiry date: 2018/08/26 - 2018/09/26
2019/08/06 00:05:14 5C2F7A01 r   (reader) SRG-V6 [viaccess] class: 03, expiry date: 2018/08/26 - 2018/09/26
2019/08/06 00:05:14 5C2F7A01 r   (reader) SRG-V6 [viaccess] class: 04, expiry date: 2018/08/26 - 2018/09/26
2019/08/06 00:05:14 5C2F7A01 r   (reader) SRG-V6 [viaccess] class: 05, expiry date: 2018/08/26 - 2018/09/26
2019/08/06 00:05:14 0DC0D959 r   (reader) HDplusHD02 [smartreader] ready for requests
2019/08/06 00:05:14 0DC0D959 r   (reader) HDplusHD02 [smartreader] found card system nagra
2019/08/06 00:05:14 0DC0D959 r   (reader) HDplusHD02 [nagra] THIS WAS A SUCCESSFUL START ATTEMPT No  1 out of max allotted of 1
2019/08/06 00:05:14 0DC0D959 r   (reader) HDplusHD02 [nagra] ROM:    D N A S P 1 8 0
2019/08/06 00:05:14 0DC0D959 r   (reader) HDplusHD02 [nagra] REV:    M e r 0 0 0
2019/08/06 00:05:14 0DC0D959 r   (reader) HDplusHD02 [nagra] SER:    ############
2019/08/06 00:05:14 0DC0D959 r   (reader) HDplusHD02 [nagra] CAID:   1843
2019/08/06 00:05:14 0DC0D959 r   (reader) HDplusHD02 [nagra] Prv.ID: 00 00 34 11 (sysid)
2019/08/06 00:05:14 0DC0D959 r   (reader) HDplusHD02 [nagra] Prv.ID: 00 00 00 00 
2019/08/06 00:05:14 5C2F7A01 r   (reader) SRG-V6 [viaccess] class: 01, expiry date: 2018/08/16 - 2018/09/30
2019/08/06 00:05:14 0DC0D959 r   (reader) HDplusHD02 [nagra] IRD ID: ############
2019/08/06 00:05:14 0DC0D959 r   (reader) HDplusHD02 [nagra] active to: 2019/09/20 02:05
2019/08/06 00:05:14 5C2F7A01 r   (reader) SRG-V6 [viaccess] provider: 3, id: ######################, sa: ########, geo: FF FF FF FF 
2019/08/06 00:05:14 0DC0D959 r   (reader) HDplusHD02 [nagra] -----------------------------------------
2019/08/06 00:05:14 0DC0D959 r   (reader) HDplusHD02 [nagra] |id  |tier    |valid from  |valid to    |
2019/08/06 00:05:14 0DC0D959 r   (reader) HDplusHD02 [nagra] +----+--------+------------+------------+
2019/08/06 00:05:15 0DC0D959 r   (reader) HDplusHD02 [nagra] |8011|0066    |2012/04/26  |2013/04/27  |
2019/08/06 00:05:15 0DC0D959 r   (reader) HDplusHD02 [nagra] |8011|0067    |2010/01/19  |2010/01/20  |
2019/08/06 00:05:15 5C2F7A01 r   (reader) SRG-V6 [viaccess] provider: 4, id: ######################, sa: ########, geo: FF FF FF FF 
2019/08/06 00:05:15 0DC0D959 r   (reader) HDplusHD02 [nagra] |8011|0BEA    |2019/03/10  |2020/03/11  |
2019/08/06 00:05:15 0DC0D959 r   (reader) HDplusHD02 [nagra] -----------------------------------------
2019/08/06 00:05:15 5C2F7A01 r   (reader) SRG-V6 [viaccess] provider: 5, id: ######################, sa: ########, geo: FF FF FF FF 
2019/08/06 00:05:16 5C2F7A01 r   (reader) SRG-V6 [viaccess] provider: 6, id: ######################, sa: ########, geo: FF FF FF FF 
2019/08/06 00:05:17 00000000 s     (main) init for all local cards done
2019/08/06 00:05:17 00000000 s (emmcache) loaded 1335 emmcache records from /tmp/.oscam/oscam.emmcache in 14 ms
2019/08/06 00:05:17 00000000 s (emmcache) loaded 1335 emmstat records from /tmp/.oscam/oscam.emmstat in 11 ms
2019/08/06 00:05:17 00000000 s (anticasc) anti cascading disabled
[... cut ...]
2019/08/06 00:05:30 295B0451 c      (ecm) dvbapi (0500@060200/08DE/4331/30:DB5201BE8B94B0A408063477939E5E64): found (319 ms) by SRG-V6 - SRF 1 HD
2019/08/06 00:05:40 295B0451 c      (ecm) dvbapi (0500@060200/08DE/4331/30:FB3E80652D4D491DA9B8D96DCBD2ED58): found (315 ms) by SRG-V6 - SRF 1 HD
2019/08/06 00:05:41 295B0451 c      (ecm) dvbapi (1843@000000/0000/5274/89:66A98C257184BDA2C97B3D11298C9DB5): found (538 ms) by HDplusHD02 - WELT HD
2019/08/06 00:05:50 295B0451 c      (ecm) dvbapi (0500@060200/08DE/4331/30:1A306791527435BE5A497FD392C617D3): found (316 ms) by SRG-V6 - SRF 1 HD

oscam-emu, 2019-08-06:
What's this? This is not a log from a BBC channel...

closed 2019-08-30 by oscam-emu