oscam-emu/oscam-patched-old #60: Channel CDF in 058w tp 4120h blackscreen

lorba7068, 2019-08-30:
Hello good morning, I have problems for some channels of a tp (4120h / 058w) all other channels open video without problem but just these channels that are the same tp do not open attached log. I am using Vuduo2 without stream relay
[oscam.txt](https://github.com/oscam-emu/oscam-patched/files/3560990/oscam.txt)

oscam-emu, 2019-08-30:
Did they used to work before? They might have switched to newer powevu algorithm that is currently not support by emu.

lorba7068, 2019-08-30:
There are approximately 6 channels added to the tp4120h, the tp 4120h contains more powervu channels and all work without problem only the newly added CDF and CNN channels, I am currently using new system based on the hash of the tp to generate the ecm of each channel via emmkey .

oscam-emu, 2019-08-30:
@lorba7068  why did you closed the issue?

lorba7068, 2019-08-30:
Sorry I'm wrong, my native language is Spanish, I'm sorry I would like to continue with the subject.

oscam-emu, 2019-08-30:
No problem.

I see these channels use the HEVC video codec, not the regular AVC.
Your receiver is not capable of decoding this video stream.

Normally you should only hear audio playing. Is this correct?

lorba7068, 2019-08-30:
It is correct I only listen to audio but I don't get an image.  So I really appreciate your attention.

closed 2019-09-02 by oscam-emu