oscam-emu/oscam-patched-old #61: Power vu not work " STAPI based cpu "

popking159, 2019-08-30:
I have Icone iron plus box , it android 7 stapi based , afn was working fine , now i want to watch power vu channels on astra 4.8 like Betfred channel , i am using latest oscam and emu versions, the emu found the key after about 30 sec but no audio no video , I need help please to open these channels .

oscam-emu, 2019-08-30:
I am not sure if these channels work on emu today. What other users say?
Do the Documentaries on 4.8E work?

popking159, 2019-08-31:
Another friend build his own build and confirm that the channels work , i tried his build , only audio work but black screen , my build no audio and black screen.

oscam-emu, 2019-09-02:
Does your friend use the same box?

Anyway, can you create a log with debug 2 + 128 enabled and post you oscam.conf file?

popking159, 2019-09-02:
[log (2).txt](https://github.com/oscam-emu/oscam-patched/files/3565914/log.2.txt)
config file

> # oscam.conf generated automatically by Streamboard OSCAM 1.20_svn SVN r11546-797
# Read more: http://www.streamboard.tv/svn/oscam/trunk/Distribution/doc/txt/oscam.conf.txt

[global]
logfile                       = /storage/emulated/0/Download
clienttimeout                 = 8000
nice                          = -1
waitforcards                  = 0
lb_save                       = 100
lb_retrylimit                 = 800
lb_savepath                   = /mnt/plugin/oscam

[cache]

[streamrelay]

[dvbapi]
enabled                       = 1
au                            = 1
pmt_mode                      = 5
request_mode                  = 1
delayer                       = 60
user                          = user1

[webif]
httpport                      = 8888
httpallowed                   = 127.0.0.1,192.168.0.0-192.168.255.255,255.255.255.255

> 
my friend uses the same box

oscam-emu, 2019-09-02:
Your log does NOT have debug 128 enabled, but only debug level 2...

Also, it seems that you do not have extended CWs enabled. So, in `webif`, select `config -> dvb-api` and set `api for extended cws` to either **1** or **2**.

popking159, 2019-09-02:
[log (6).txt](https://github.com/oscam-emu/oscam-patched/files/3565999/log.6.txt)
[log (7).txt](https://github.com/oscam-emu/oscam-patched/files/3566000/log.7.txt)
i know that api for extended cws belongs to enigma2 as i have another box spark st7111 and i choose 2 for api for extended cws , but in my new android box , the default value is 0 , and AFN was working on this settings , anyway , i tried your steps and choose 2 but now only biss key work , cccam not work , see the log please .

oscam-emu, 2019-09-02:
Powervu via dvbapi is not going to work without api for extended cw set to 1 or 2.
Unless you use the stream relay option, which is another thing. What is your friend's setup/configuration?

I can't understand much from your logs. They are messy and not complete.

popking159, 2019-09-02:
i used the oscam of my  friend and it works
i noticed that the compiler of his oscam is "arm-linux-androideabi'
while the compiler of mine is "armv7a-unknown-linux-android24-stapi"
is that differance may be the cause ?
his confiig file

> # oscam.conf generated automatically by Streamboard OSCAM 1.20_svn SVN r11546-797
# Read more: http://www.streamboard.tv/svn/oscam/trunk/Distribution/doc/txt/oscam.conf.txt

[global]
logfile                       = /data/plugin/tuxbox/oscam.log
clienttimeout                 = 8000
nice                          = -1
waitforcards                  = 0
preferlocalcards              = 1
ecmfmt                        = P: c:p:s:d:i:o #ECM_L:l #CW=w HOP:j
#ecmfmt = P: c:0p:s:d:i #ECM_L:l #CW=w HOP:j

[cache]

[dvbapi]
enabled                       = 1
au                            = 1
pmt_mode                      = 5
request_mode                  = 1
delayer                       = 60
user                          = dvbapiau
write_sdt_prov                = 1

[webif]
httpport                      = 8888
httpallowed                   = 127.0.0.1,0.0.0.0-255.255.255.255
http_status_log               = 1

oscam-emu, 2019-09-03:
Well, if you did not have the correct binary in the first place, how it it supposed to work?

closed 2019-09-03 by oscam-emu