oscam-emu/oscam-patched-old #64: sony,discovery on 68.5east not working on powervu any solution plz

colesbergkhn, 2019-10-01:
hi,,, on 68.5east sony+discovery channels are not working any solution plz

credoman, 2019-11-22:
Posted By Ghouri Today, 20:46 On Asiasat 7, Sony is autorolling again. Checked with Oscam Emu 11570.

budiarno, 2019-11-22:
it uses ModeCW = 3 by now.

budiarno, 2019-11-22:
> Posted By Ghouri Today, 20:46 On Asiasat 7, Sony is autorolling again. Checked with Oscam Emu 11570.

it uses ModeCW = 3

credoman, 2019-11-27:
> > Posted By Ghouri Today, 20:46 On Asiasat 7, Sony is autorolling again. Checked with Oscam Emu 11570.
> 
> it uses ModeCW = 3 by now.

Sony network off
Asiasat.7 They changed to hashModeCw: 1,
modeCW: 8, modeCW: 35:modeCW: 25: & modeCW: 41 cyclic.

nautilus7, 2024-03-31:
The solution is to send us the code to include it in oscam....

closed 2024-03-31 by nautilus7