oscam-emu/oscam-patched-old #66: I have a problem in emu channels four power vu " thor & Astra 4A 4.8E"  Black screen

GHOSt19dZ, 2019-10-21:
welcome 
I have a problem in emu channels four power vu " thor & Astra 4A 4.8E" 
Brings the code but the channel does not work
Black screen
But when I use it  " Mcas " Works well
[Mcas.tar.gz](https://github.com/oscam-emu/oscam-patched/files/3752222/Mcas.tar.gz)

My device is OPENBOX SX6 HD 
My config
[keys.tar.gz](https://github.com/oscam-emu/oscam-patched/files/3752238/keys.tar.gz)

credoman, 2019-10-27:
all works fine. use instruction oscam emu softcam key

credoman, 2019-10-31:
you may try seen stream iptv power vu 
or enigma 2 for stapi5.

GHOSt19dZ, 2019-10-31:
Look for my config
[keys.tar.gz](https://github.com/oscam-emu/oscam-patched/files/3795922/keys.tar.gz)

and look for oscam log

`

    2019/10/31 21:41:53 00000000       (stat) loadbalancer: statistic saved 23 records to /tmp/.oscam/stat in 1 ms
    2019/10/31 21:43:35 00000000   (anticasc) acasc: restored access for inactive user 'dvbapiau'
    2019/10/31 21:43:35 59B398EE c      (ecm) dvbapiau (P: 0E00:000000:09C4:177D:0000: #ECM_L:40 #CW=207AB6B9C4583BE50000000000000000 HOP:00): found (5 ms) by emulator - Discovery Showcase HD Europ
    2019/10/31 21:43:36 59B398EE c      (ecm) dvbapiau (P: 0E00:000000:09C4:177D:0000: #ECM_L:40 #CW=00000000000000000D853E9146A13E89 HOP:00): found (7 ms) by emulator - Discovery Showcase HD Europ
    2019/10/31 21:43:36 59B398EE c      (ecm) dvbapiau (P: 0E00:000000:09C4:177D:0000: #ECM_L:40 #CW=CB0BD5803462B0E30000000000000000 HOP:00): found (61 ms) by emulator - Discovery Showcase HD Europ
    2019/10/31 21:43:36 59B398EE c      (ecm) dvbapiau (P: 0E00:000000:09C4:177D:0000: #ECM_L:40 #CW=0000000000000000571FD97A52C2E3AD HOP:00): found (3 ms) by emulator - Discovery Showcase HD Europ
    2019/10/31 21:43:37 59B398EE c      (ecm) dvbapiau (P: 0E00:000000:09C4:177D:0000: #ECM_L:40 #CW=DA52047A190DE3250000000000000000 HOP:00): found (3 ms) by emulator - Discovery Showcase HD Europ
    2019/10/31 21:43:38 59B398EE c      (ecm) dvbapiau (P: 0E00:000000:09C4:177D:0000: #ECM_L:40 #CW=0000000000000000F19BBACB3EFE4067 HOP:00): found (3 ms) by emulator - Discovery Showcase HD Europ
    2019/10/31 21:43:39 59B398EE c      (ecm) dvbapiau (P: 0E00:000000:09C4:177D:0000: #ECM_L:40 #CW=B689491615D98A620000000000000000 HOP:00): found (2 ms) by emulator - Discovery Showcase HD Europ
    2019/10/31 21:43:40 59B398EE c      (ecm) dvbapiau (P: 0E00:000000:09C4:177D:0000: #ECM_L:40 #CW=000000000000000004B91325A8CB76E3 HOP:00): found (4 ms) by emulator - Discovery Showcase HD Europ
    2019/10/31 21:43:41 59B398EE c      (ecm) dvbapiau (P: 0E00:000000:09C4:177D:0000: #ECM_L:40 #CW=8CA4CDE380E97FBC0000000000000000 HOP:00): found (4 ms) by emulator - Discovery Showcase HD Europ
    2019/10/31 21:43:42 59B398EE c      (ecm) dvbapiau (P: 0E00:000000:09C4:177D:0000: #ECM_L:40 #CW=0000000000000000D5F4203EFB9E3EC2 HOP:00): found (3 ms) by emulator - Discovery Showcase HD Europ
    2019/10/31 21:43:43 59B398EE c      (ecm) dvbapiau (P: 0E00:000000:09C4:177D:0000: #ECM_L:40 #CW=1F23CD57853745EF0000000000000000 HOP:00): found (4 ms) by emulator - Discovery Showcase HD Europ
    2019/10/31 21:43:44 59B398EE c      (ecm) dvbapiau (P: 0E00:000000:09C4:177D:0000: #ECM_L:40 #CW=00000000000000008F2592AEFE9D3280 HOP:00): found (4 ms) by emulator - Discovery Showcase HD Europ
    2019/10/31 21:43:45 59B398EE c      (ecm) dvbapiau (P: 0E00:000000:09C4:177D:0000: #ECM_L:40 #CW=25D67A19DC4C1F860000000000000000 HOP:00): found (3 ms) by emulator - Discovery Showcase HD Europ
    2019/10/31 21:43:46 59B398EE c      (ecm) dvbapiau (P: 0E00:000000:09C4:177D:0000: #ECM_L:40 #CW=00000000000000004FC70BA1295B4C98 HOP:00): found (3 ms) by emulator - Discovery Showcase HD Europ
    2019/10/31 21:43:47 59B398EE c      (ecm) dvbapiau (P: 0E00:000000:09C4:177D:0000: #ECM_L:40 #CW=3ED920044A7A68AB0000000000000000 HOP:00): found (3 ms) by emulator - Discovery Showcase HD Europ
    2019/10/31 21:43:48 59B398EE c      (ecm) dvbapiau (P: 0E00:000000:09C4:177D:0000: #ECM_L:40 #CW=0000000000000000B5E9311692831CB6 HOP:00): found (6 ms) by emulator - Discovery Showcase HD Europ
    2019/10/31 21:43:49 59B398EE c      (ecm) dvbapiau (P: 0E00:000000:09C4:177D:0000: #ECM_L:40 #CW=9170C2CE4A9485890000000000000000 HOP:00): found (4 ms) by emulator - Discovery Showcase HD Europ
    2019/10/31 21:43:50 59B398EE c      (ecm) dvbapiau (P: 0E00:000000:09C4:177D:0000: #ECM_L:40 #CW=00000000000000007AA4F88A6DF70DDC HOP:00): found (5 ms) by emulator - Discovery Showcase HD Europ
    2019/10/31 21:43:51 59B398EE c      (ecm) dvbapiau (P: 0E00:000000:09C4:177D:0000: #ECM_L:40 #CW=BAE9295D548313190000000000000000 HOP:00): found (4 ms) by emulator - Discovery Showcase HD Europ
    2019/10/31 21:43:52 59B398EE c      (ecm) dvbapiau (P: 0E00:000000:09C4:177D:0000: #ECM_L:40 #CW=0000000000000000EA89802546E0979D HOP:00): found (3 ms) by emulator - Discovery Showcase HD Europ
    2019/10/31 21:43:53 59B398EE c      (ecm) dvbapiau (P: 0E00:000000:09C4:177D:0000: #ECM_L:40 #CW=3D75D65DABC268DF0000000000000000 HOP:00): found (6 ms) by emulator - Discovery Showcase HD Europ
    2019/10/31 21:43:54 59B398EE c      (ecm) dvbapiau (P: 0E00:000000:09C4:177D:0000: #ECM_L:40 #CW=00000000000000008029E6E37A9D98CB HOP:00): found (5 ms) by emulator - Discovery Showcase HD Europ
    2019/10/31 21:43:55 59B398EE c      (ecm) dvbapiau (P: 0E00:000000:09C4:177D:0000: #ECM_L:40 #CW=CDD0B602BA3240380000000000000000 HOP:00): found (5 ms) by emulator - Discovery Showcase HD Europ
    2019/10/31 21:43:56 59B398EE c      (ecm) dvbapiau (P: 0E00:000000:09C4:177D:0000: #ECM_L:40 #CW=00000000000000000D236D109E9708F4 HOP:00): found (5 ms) by emulator - Discovery Showcase HD Europ
    2019/10/31 21:43:57 59B398EE c      (ecm) dvbapiau (P: 0E00:000000:09C4:177D:0000: #ECM_L:40 #CW=FD9DAEE5BA4F02150000000000000000 HOP:00): found (5 ms) by emulator - Discovery Showcase HD Europ
    2019/10/31 21:43:58 59B398EE c      (ecm) dvbapiau (P: 0E00:000000:09C4:177D:0000: #ECM_L:40 #CW=0000000000000000BCEF04433BAD7061 HOP:00): found (5 ms) by emulator - Discovery Showcase HD Europ
[... cut ...]
    2019/10/31 21:49:32 59B398EE c      (ecm) dvbapiau (P: 0E00:000000:09C4:177D:0000: #ECM_L:40 #CW=0000000000000000C7797657D37CAB5E HOP:00): found (5 ms) by emulator - Discovery Showcase HD Europ
    2019/10/31 21:49:33 59B398EE c      (ecm) dvbapiau (P: 0E00:000000:09C4:177D:0000: #ECM_L:40 #CW=9B865BCECDC810DF0000000000000000 HOP:00): found (5 ms) by emulator - Discovery Showcase HD Europ
    2019/10/31 21:49:34 59B398EE c      (ecm) dvbapiau (P: 0E00:000000:09C4:177D:0000: #ECM_L:40 #CW=00000000000000000438EC9EBAB06820 HOP:00): found (4 ms) by emulator - Discovery Showcase HD Europ`

and the channle black screen

credoman, 2019-10-31:
Yes is don't work. Only mcas for stapi5. Read first answer.

closed 2021-05-27 by MOHAMED19OS