oscam-emu/oscam-patched-old #68: do you know 66e sony packets powervu enc.?

PAKO34, 2019-11-05:
**do you know 66e sony packets powervu enc.?**

credoman, 2019-11-22:
Posted By Ghouri Today, 20:46 On Asiasat 7, Sony is autorolling again. Checked with Oscam Emu 11570.

credoman, 2019-11-27:
Sony network off
Asiasat.7 They changed to hashModeCw: 1,
modeCW: 8, modeCW: 35:modeCW: 25: & modeCW: 41 cyclic.

NCODER85, 2019-11-30:
> Sony network off
> Asiasat.7 They changed to hashModeCw: 1,
> modeCW: 8, modeCW: 35:modeCW: 25: & modeCW: 41 cyclic.

Can it will be fix in coming oscam patch?

budiarno, 2019-12-01:
> Can it will be fix in coming oscam patch? Also I am little confused as in oscam logs emu found valid CW. Screenshot attached.
> [image]

That is because the ECM Key is valid, but the cw is actually not valid.

NCODER85, 2019-12-01:
> > Can it will be fix in coming oscam patch? Also I am little confused as in oscam logs emu found valid CW. Screenshot attached.
> > [image]
> 
> That is because the ECM Key is valid, but the cw is actually not valid.

Thanks for clarifying.
When can we expect this fix in oscam patch.

Nabeelbutt786, 2020-08-07:
hello dear  help  me

DRIVANnega, 2022-04-24:
i have theory, but its crazy to say this, i have been noticing a lot with how they did 2019 update on group category by channel hash code and it work great for not manual input our keys for every channel when it done automatic by group channel...to my point i making is, also we having problem with discovery channels entire globe with cycling same hash control word by this group modeCW: 8, modeCW: 35:modeCW: 25: & modeCW: 41 and cyclying every sec yes...here my theory, what if the EMM been working normally and ECM keys are correct but it doesn't know what to do with other modeCW in one group that requires to be in a group of CW into same ECM keys? not sure this work but this longshot in my theory if this helps

nautilus7, 2024-03-31:
A fix can be expected when someone of you send us the code...

closed 2024-03-31 by nautilus7