oscam-emu/oscam-patched-old #71: DreamONE Powervu: Channel black screen.

audi06, 2020-01-22:
Hi.
Powervu problems previously experienced for dm9x0 are now experienced on DreamONE device. Do you have any suggestions on Powervu? No error appears in the Oscam log. Channel black screen.
[oscam.log](https://github.com/oscam-emu/oscam-patched/files/4098405/oscam.log)

https://github.com/oscam-emu/oscam-patched/issues/32#issuecomment-465784047​

oscam-emu, 2020-01-25:
DES descrambling does not work afaik. They probably changed their api, so DM One is not compatible with oscam.

EnoSat, 2024-01-21:
Any progress with OE2.6 ?

audi06, 2024-01-21:
@EnoSat 
no problem

[image]

```
2024/01/22 00:59:41 6FC71678 c      (ecm) dvbapiau (P: 0E00:000000:006D:17B1:0000 #ECM_L:99 #CW=00000000000000001F0B8FB9A78B2D5F HOP:00): found (1 ms) by emulator - AFN (lg)
2024/01/22 00:59:41 6FC71678 c      (ecm) dvbapiau (P: 0E00:000000:006D:17B1:0000 #ECM_L:99 #CW=B4A376CD873203BC0000000000000000 HOP:00): found (1 ms) by emulator - AFN (lg)
2024/01/22 00:59:51 6FC71678 c      (ecm) dvbapiau (P: 0E00:000000:006D:17B1:0000 #ECM_L:99 #CW=0000000000000000011CEE0BE85AB8FA HOP:00): found (1 ms) by emulator - AFN (lg)
2024/01/22 00:59:51 6FC71678 c      (ecm) dvbapiau (P: 0E00:000000:006D:17B1:0000 #ECM_L:99 #CW=B4A376CD873203BC0000000000000000 HOP:00): found (1 ms) by emulator - AFN (lg)
2024/01/22 01:00:01 6FC71678 c      (ecm) dvbapiau (P: 0E00:000000:006D:17B1:0000 #ECM_L:99 #CW=328E17D7E4F79A750000000000000000 HOP:00): found (1 ms) by emulator - AFN (lg)
2024/01/22 01:00:11 6FC71678 c      (ecm) dvbapiau (P: 0E00:000000:006D:17B1:0000 #ECM_L:99 #CW=0000000000000000E439E2FFB8098849 HOP:00): found (1 ms) by emulator - AFN (lg)

EnoSat, 2024-01-22:
no talk about CSA, talk about **DES**