oscam-emu/oscam-patched-old #72: DVBApi v5 support ?

scramatte, 2020-01-23:
Hello,

I need to use OSCAM with MumuDVB.  Looks that each one une distinct DVBApi version 
How can I compile or setup OSCAM to use DVBapi v5 ?

`(dvbapi) Using /tmp/camd.socket listen socket, API forced to DVBAPIv3 (0), userconfig boxtype: 11`

and  MumuDVB

```
MuMuDVB Version 2.1.0_20181020_mumudvb2
 --- Build information ---
Built with CAM support.
Built with SCAM support.
Built with ATSC support.
Built with ATSC long channel names support.
Built with support for DVB API Version 5.11.
Built with support for DVB-T2.
```

Regards

oscam-emu, 2020-01-25:
What oscam calls dvbapi is way different than the respective api in linux, so the the versions do not correlate. So, don't worry about the version. Just use it and tell me if something does not work as you expect.

scramatte, 2020-01-26:
I'm  new with Oscam , and basically I'm unable to unscramble channels with MumuDVB.
When I try to unscramble channel I see ECM requests into oscam log  but channels remains scrambled in MumuDVB. Looks that starts well but for some reason stop to unscramble.

```
3:52:32 78552784 c   (dvbapi) Demuxer 0 ecmpid 0 CAID: 1810 ECM_PID: 0886 PROVID: 000000
23:52:32 78552784 c   (dvbapi) Demuxer 0 found 1 ECMpids and 13 STREAMpids in caPMT
23:52:32 78552784 c   (dvbapi) Demuxer 0 found channel in cache and matching prio -> start descrambling ecmpid 0
23:52:32 78552784 c   (dvbapi) Demuxer 0 trying to descramble PID 0 CAID 1810 PROVID 000000 ECMPID 0886 ANY CHID PMTPID 0000 VPID 00A1
23:52:32 78552784 c   (dvbapi) Demuxer 0 found 1 ECMpids and 13 STREAMpids in PMT
23:52:33 78552784 c      (ecm) anonymous (1810@000000/0000/74CD/8E:B0BF721C3EB32589B4852F32224C707F): found (77 ms) by Channel 1 -
23:52:39 78552784 c      (ecm) anonymous (1810@000000/0000/74CD/8E:B8BC674EC51421872D76E4C7416F6904): found (415 ms) by srv1 - Channel 1
23:52:54 78552784 c      (ecm) anonymous (1810@000000/0000/74CD/8E:C6803BD91E0E56CFF7F104DAF11D7BC8): found (397 ms) by srv1 - Channel 1
23:53:09 78552784 c      (ecm) anonymous (1810@000000/0000/74CD/8E:EC97532B483AD222688DABC25CEB95B3): found (434 ms) by srv1 - Channel 1
23:53:24 78552784 c      (ecm) anonymous (1810@000000/0000/74CD/8E:3D02387EB092BDC0FBB0A59FA0FAD6ED): found (398 ms) by srv1 - Channel 1
23:53:39 78552784 c      (ecm) anonymous (1810@000000/0000/74CD/8E:0BF143CDADA32CEA9CD35890763D21B6): found (429 ms) by srv1 - Channel 1
23:53:54 78552784 c      (ecm) anonymous (1810@000000/0000/74CD/8E:13EB22548A254AB7CC212FAAF881D33C): found (388 ms) by srv1 - Channel 1
23:54:10 78552784 c      (ecm) anonymous (1810@000000/0000/74CD/8E:6DCFEA396959E719A7844D520BA5F505): found (423 ms) by srv1 - Channel 1
23:54:12 78552784 c   (dvbapi) Demuxer 0 stop descrambling program number 74CD (Channel 1

```
but channels still scrambled :
Channel "Chan1" is now highly scrambled (97% of scrambled packets). Card 2

```
--oscam.conf--
[global]
logfile = stdout
fallbacktimeout = 3
bindwait = 5
nice = -1
maxlogsize = 200
waitforcards = 0
preferlocalcards = 1
lb_mode = 1
lb_save = 100
lb_savepath = /usr/local/stat
lb_auto_timeout_t = 26000

[anticasc]
enabled = 1

[cache]

[newcamd]

[dvbapi]
enabled = 1
au = 1
request_mode = 1
pmt_mode = 4
user = tv
boxtype = pc

[webif]
httpport = 8888
serverip = 192.168.100.250
httpuser = root
httppwd = root
httpallowed = 127.0.0.1,192.168.100.0-192.168.100.254

--mumudvb config file --

card=2
lnb_type=universal
freq=11435.50
pol=v
srate=22000
coderate=5/6
autoconfiguration=full
delivery_system=DVBS2
modulation=8PSK
autoconf_sid_list=29901
unicat=1
ip_http=172.16.1.89
scam_support=1
multicast_ttl=128
autoconf_scrambled=1
ring_buffer_default_size=131072
decsa_default_delay=500000
send_default_delay=1500000
sap=0
--

```

devacom, 2020-07-09:
Hi there,
I Have amlogic S905 android box (AVL6211 that support dvbapi-v5).
The built-in oscam (cam_mcas.apk) use dvbapi called dvbapi. This old oscam work fine with the fallowing log:
 2015/01/01 01:40:10 00000000 s     (main) System name    = Linux
2015/01/01 01:40:10 00000000 s     (main) Host name      = localhost
2015/01/01 01:40:10 00000000 s     (main) Release        = 3.14.29
2015/01/01 01:40:10 00000000 s     (main) Version        = #3 SMP PREEMPT Tue Jun 5 20:14:15 CST 2018
2015/01/01 01:40:10 00000000 s     (main) Machine        = aarch64
2015/01/01 01:40:10 00000000 s     (main) The current system time is smaller than the build date (May 19 2017). Waiting up to 1 seconds for time to correct
2015/01/01 01:40:12 00000000 s     (main) Waiting was not successful. OSCam will be started but is UNSUPPORTED this way. Do not report any errors with this version.
2015/01/01 01:40:12 00000000 s      (emu) OSCam-Emu version 738
2015/01/01 01:40:12 00000000 s   (config) userdb reloaded: 2 accounts loaded, 0 expired, 0 disabled
2015/01/01 01:40:12 00000000 s     (main) signal handling initialized
2015/01/01 01:40:12 00000000 s   (config) 10 provid's loaded
2015/01/01 01:40:12 00000000 s   (config) 74 service-id's loaded in 3 ms
2015/01/01 01:40:12 00000000 s     (main) Cannot open /tmp/.oscam/oscam.version (errno=2 No such file or directory)
2015/01/01 01:40:12 00000000 s      (net) cccam: initialized (fd=4, port=12000)
2015/01/01 01:40:12 00000000 s   (reader) emulator [emu] creating thread for device emulator
2015/01/01 01:40:12 00000000 s   (reader) constant.cw [constcw] creating thread for device /data/data/com.streamboard.android.oscam/constant.cw
2015/01/01 01:40:12 00000000 s     (stat) loadbalancer: could not open /tmp/.oscam/stat for reading (errno=2 No such file or directory)
2015/01/01 01:40:12 00000000 s     (main) waiting for local card init
2015/01/01 01:40:12 747B7FB3 r      (emu) Reading key file: /data/data/com.streamboard.android.oscam/SoftCam.Key
2015/01/01 01:40:12 50A01AE4 p  (constcw) Local reader: constant.cw (file: /data/data/com.streamboard.android.oscam/constant.cw)
2015/01/01 01:40:12 50A01AE4 p   (reader) constant.cw [constcw] proxy initialized, server /data/data/com.streamboard.android.oscam/constant.cw:0
2015/01/01 01:40:12 00000000        (emu) INFO: FFDecsa parallel mode = 32
2015/01/01 01:40:12 747B7FB3 r   (reader) emulator [emu] Reader initialized (device=emulator, detect=cd, mhz=357, cardmhz=357)
2015/01/01 01:40:12 3603CFD2 h    (webif) webif: decompressed 178398 bytes back into 440304 bytes
2015/01/01 01:40:12 3603CFD2 h    (webif) HTTP Server running. ip=0.0.0.0 port=8888
2015/01/01 01:40:13 747B7FB3 r   (reader) emulator [emu] card detected
2015/01/01 01:40:14 747B7FB3 r   (reader) emulator [emu] found card system emu
2015/01/01 01:40:14 747B7FB3 r   (reader) emulator [emu] THIS WAS A SUCCESSFUL START ATTEMPT No  1 out of max alloted of 1
2015/01/01 01:40:14 747B7FB3 r      (emu) Reading key file: /data/data/com.streamboard.android.oscam/SoftCam.Key
2015/01/01 01:40:15 00000000 s     (main) init for all local cards done
2015/01/01 01:40:15 00000000 s (anticasc) anti cascading disabled
2015/01/01 01:40:15 00000000 s    (alpum) CPU is S905
2015/01/01 01:40:15 00000000 s    (alpum) i2c bus dev:/dev/i2c-1
2015/01/01 01:40:15 00000000 s    (alpum) alpum_init: open /dev/i2c-1
2015/01/01 01:40:15 7F35E636 c   (client) plain dvbapi-client granted (dvbapiau, au=auto (6 reader))
2015/01/01 01:40:15 7F35E636 c   (dvbapi) Detected /dev/dvb0.demux0 Api: 0, userconfig boxtype: 14 maximum amount of possible filters is 24 (oscam limit is 32)
2015/01/01 01:40:15 00000000 s CHIP = 15
2015/01/01 01:40:15 00000000 s C00F chip
2015/01/01 01:40:15 00000000 s    (alpum) chip id = 00 03 7e ec
2015/01/01 01:40:15 00000000 s (main-entry) cls_ecm
2015/01/01 01:40:16 00000000      (camui) Rquest UI configure
2015/01/01 01:40:16 00000000      (camui) Send UI config
2015/01/01 01:40:16 00000000      (camui) request version
2015/01/01 01:40:17 00000000      (camui) send version: .6399e28
2015/01/01 01:41:16 7F35E636 c (main-entry) cls_ecm

when I use the oscam built from this source, I got:
that /dev/dvb0.ca not found
when I found 
/dev/dvb0.dsc0
/dev/dvb0.dsc1
/dev/dvb0.dvr0
/dev/dvb0.dvr1
/dev/dvb0.dvr2

I tried to open the worked oscam with hex editor I found:
/dev/dvb0.dsc%d        dvb.demux%d         /data/local/tmp/camd.socket
not like the code in module-dvbapi.c
	/* wetek (dvb-api-3)*/      { "/dev/dvb0."   "ca%d",         "demux%d",      "/tmp/camd.socket", DVBAPI_3 },

Ro-Den, 2021-05-24:
It's an issue of a particular MuMuDVB package: https://github.com/braice/MuMuDVB/issues/258

AbuBaniaz, 2024-03-16:
I think this issue can be closed now as problem was fixed elsewhere.

closed 2024-03-16 by oscam-emu