oscam-emu/oscam-patched-old #75: Error compile OSCAM

alextdr2003, 2020-04-18:
Hello, I am trying to compile oscam emu on a Raspberry Pi 4 using Ubuntu 19.10 version.

I receive this error after I disables compiling with softcam:

`LINK    Distribution/oscam-1.20_svn11576-798-aarch64-linux-gnu.debug
/usr/bin/ld: build/aarch64-linux-gnu/module-emulator-biss.o: in function `parse_session_data':
/home/ubuntu/oscam-patched/module-emulator-biss.c:677: undefined reference to `RSA_private_decrypt'
/usr/bin/ld: build/aarch64-linux-gnu/module-emulator-biss.o: in function `biss_read_pem':
/home/ubuntu/oscam-patched/module-emulator-biss.c:835: undefined reference to `PEM_read_RSAPrivateKey'
/usr/bin/ld: /home/ubuntu/oscam-patched/module-emulator-biss.c:860: undefined reference to `i2d_RSA_PUBKEY'
/usr/bin/ld: /home/ubuntu/oscam-patched/module-emulator-biss.c:869: undefined reference to `EVP_MD_CTX_new'
/usr/bin/ld: /home/ubuntu/oscam-patched/module-emulator-biss.c:875: undefined reference to `EVP_sha256'
/usr/bin/ld: /home/ubuntu/oscam-patched/module-emulator-biss.c:875: undefined reference to `EVP_DigestInit_ex'
/usr/bin/ld: /home/ubuntu/oscam-patched/module-emulator-biss.c:876: undefined reference to `EVP_DigestUpdate'
/usr/bin/ld: /home/ubuntu/oscam-patched/module-emulator-biss.c:877: undefined reference to `EVP_DigestFinal_ex'
/usr/bin/ld: /home/ubuntu/oscam-patched/module-emulator-biss.c:878: undefined reference to `EVP_MD_CTX_free'
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:436: Distribution/oscam-1.20_svn11576-798-aarch64-linux-gnu.debug] Error 1
make: *** [Makefile:408: all] Error 2
ubuntu@ubuntu:~/oscam-patched$`

Could you help me please?

espaun206, 2020-04-22:
Hi
did you manage to solve? i have same problem with oscam emu in ubuntu sever 18.4 RPI4

oscam server if I can install it but the emu doesn't

[url=https://ibb.co/HXDhXJM][img]https://i.ibb.co/x3SF34c/patch3.png[/img][/url]

Alfredo-, 2020-05-29:
Hi

Use: 
git clone https://github.com/oscam-emu/oscam-patched.git

works for me

budiarno, 2020-10-03:
Should this command solve the issue?
`apt-get install libssl1.0-dev`

petrkr, 2020-12-25:
use `make menuconfig` and select USE_SSL in ADDONS

https://github.com/oscam-emu/oscam-patched/releases/tag/oscam11509-emu795

check #43

closed 2024-03-31 by nautilus7