oscam-emu/oscam-patched-old #77: Feature CAID mapping for non dvbapi client

Saentist, 2020-07-26:
Currently if DVBAPI used user can type in `oscam.dvbapi`
```M: AAAA BBBB```
and can use card with CAID **AAAA** for services with CAID **BBBB**
https://wiki.streamboard.tv/wiki/OSCam/en/Config/oscam.dvbapi

But in this case clients of NewCam or CCCam etc, cannot use this feature.
It can be used only by local dvbapi descrambling, not even network dvbapi
Is here a way this option to be implemented?

caveman99, 2020-07-26:
This is in OSCAM trunk scope, and not in the EMU domain. This is for EMU development only.

Off topic: The betatunnel feature should provide a similar mechanism to you for remote clients.

Saentist, 2020-07-28:
Whatever I try with betatunnel not work for CAID 4Axx and 55xx
in same time with dvbapi work

nautilus7, 2024-04-13:
Your client, which I suppose use dvbapi, should set the caid mapping. Not the server.

closed 2024-04-13 by nautilus7