oscam-emu/oscam-patched-old #80: different powervu behaviour

jpuigs, 2020-10-11:
I have all powervu and biss keys in a softcam.key (emulator reader) on enigma box A.
I try to get them on enigma box B using cccam protocol, 
All biss keys work on box B, but a very few powervu ones work on box B.
Is it normal ?

If I copy softcam.key file to local emulator reader on box B, then it works fine.

oscam-emu, 2020-10-11:
Yes, it is. cccam and other network protocols in oscam do not support the extended cw structure that is necessary for powervu channels

jpuigs, 2020-10-11:
Thanks, but then, why do a few channels work ? (f.i. EFE TV Hispasat)
Is there any other protocol instead of cccam which works perfectly between 2 oscam machines ?

oscam-emu, 2020-10-12:
Maybe it uses CSA instead of DES for scrambling and the same CW for both video and audio streams.

There is no network protocol the works perfectly with powervu, sorry!

budiarno, 2020-10-12:
> Thanks, but then, why do a few channels work ? (f.i. EFE TV Hispasat)
> Is there any other protocol instead of cccam which works perfectly between 2 oscam machines ?

It might be about the audio stream that is not scrambled.

oscam-emu, 2020-10-12:
That makes more sense.

jpuigs, 2020-10-12:
> > Thanks, but then, why do a few channels work ? (f.i. EFE TV Hispasat)
> > Is there any other protocol instead of cccam which works perfectly between 2 oscam machines ?
> 
> It might be about the audio stream that is not scrambled.

The audio is scrambled.

easyyu, 2021-01-08:
> 
> 
> Yes, it is. cccam and other network protocols in oscam do not support the extended cw structure that is necessary for powervu channels

I'm curious, there is no protocol to connect two oscam-emu (oscam-emu server with oscam-emu on receiver) and send the powervu keys?
I tried every of them, I get the keys but picture stay black.
Thx for answer

7687476874, 2021-01-08:
Why powerVU channels on Astra 4.9 satellite do not work. No sound display.

2021/01/08 21:54:00 5B005F29 c      (ecm) dvbapi (0E00@000000/0000/0708/40:45FC5609F7DB9A48BE9FE960B73ABD2B): found (0 ms) by SoftCam.Key - Science Poland HD
2021/01/08 21:54:01 5B005F29 c      (ecm) dvbapi (0E00@000000/0000/0708/40:26E39A6E98B08574F69F22782E8E5A78): found (1 ms) by SoftCam.Key - Science Poland HD
2021/01/08 21:54:02 5B005F29 c      (ecm) dvbapi (0E00@000000/0000/0708/40:829344DE74E1D8DF8A84DEAF9497D0A0): found (1 ms) by SoftCam.Key - Science Poland HD
2021/01/08 21:54:03 5B005F29 c      (ecm) dvbapi (0E00@000000/0000/0708/40:2C53AAAE217C1EFDDA73922AFF8DFF82): found (0 ms) by SoftCam.Key - Science Poland HD
2021/01/08 21:54:04 5B005F29 c      (ecm) dvbapi (0E00@000000/0000/0708/40:43C21115EFC36E617010F0757210F4BB): found (1 ms) by SoftCam.Key - Science Poland HD
2021/01/08 21:54:05 5B005F29 c      (ecm) dvbapi (0E00@000000/0000/0708/40:A03225C0987AADC9FC918CEF1D140150): found (1 ms) by SoftCam.Key - Science Poland HD
2021/01/08 21:54:06 5B005F29 c      (ecm) dvbapi (0E00@000000/0000/0708/40:2FE0564188CB4CE73C0B3EEEBFE43B09): found (2 ms) by SoftCam.Key - Science Poland HD
2021/01/08 21:54:07 5B005F29 c      (ecm) dvbapi (0E00@000000/0000/0708/40:85EDA772FBAD5C3232FF49E0E30752FB): found (1 ms) by SoftCam.Key - Science Poland HD
2021/01/08 21:54:08 5B005F29 c      (ecm) dvbapi (0E00@000000/0000/0708/40:EE5303A57B54BB68A3768F66BB94D431): found (1 ms) by SoftCam.Key - Science Poland HD
2021/01/08 21:54:09 5B005F29 c      (ecm) dvbapi (0E00@000000/0000/0708/40:A5F513E57B6B10BE67B93F8365DF7EFC): found (2 ms) by SoftCam.Key - Science Poland HD
2021/01/08 21:54:10 5B005F29 c      (ecm) dvbapi (0E00@000000/0000/0708/40:AA9B934A26CD4F6AEC9446907210E61A): found (1 ms) by SoftCam.Key - Science Poland HD
2021/01/08 21:54:11 5B005F29 c      (ecm) dvbapi (0E00@000000/0000/0708/40:264CF570FACC7DBDA23EF2928BBC5D70): found (0 ms) by SoftCam.Key - Science Poland HD

oscam-emu, 2021-01-08:
I need a log with debug level 128 +1 enabled, starting right before you tune to this channel. Also I need some info regarding your setup. Does oscam-emu run from your box (what box?) or do you get the keys from network or something?

easyyu, 2021-01-09:
> I'm curious, there is no protocol to connect two oscam-emu (oscam-emu server with oscam-emu on receiver) and send the powervu keys?
> I tried every of them, I get the keys but picture stay black.
> Thx [...]

Anyone?

7687476874, 2021-01-09:
I am using a Windows 10 OSCam I have a PC TBS6903 tv card It decodes Tandberg channels Astra 4.9 does not decode PowerVU channels.

No picture or sound on Powervu channels
[image]

[image]

[image]

easyyu, 2021-01-09:
You need to use dvbapi in oscam-emu to be able to decode.

Ro-Den, 2021-06-01:
> I need a log with debug level 128 +1 enabled, starting right before you tune to this channel.

@oscam-emu 
[log.txt](https://github.com/oscam-emu/oscam-patched/files/6577170/log.txt)

closed 2024-04-14 by nautilus7