oscam-emu/oscam-patched-old #86: Displaying current key used in live log

ArloG3300, 2021-08-18:
Although I've asked in several forums of compilers who submit versions for Enigma2 receivers, it was suggested I post here.
Is there a way to display or log the current key in the Softcam.Key file that is being used to clear a particular channel?
Using hash, EMM, and ECM keys with pv channels and "added by" UA keys is making for a very large keys file.
Some method on an E2 receiver to display the current key would be very useful.

oscam-emu, 2021-08-19:
Are you talking about powervu? In ecm keys the actual key is not displayed iirc. Only the 8 digit identifier of the key is use is shown I believe. On the the other side, the actual cw used in each crypto cycle can be shown with debug level 128.

How many powervu transponders do you have? With groups (hash) you only need one ecm key per transponder. How often do the ecm keys change these days?

ArloG3300, 2021-08-20:
> Are you talking about powervu? In ecm keys the actual key is not displayed iirc. Only the 8 digit identifier of the key is use is shown I believe. On the the other side, the actual cw used in each [...]

Yes powervu. And with an E2 receiver. Some tell me a DVB card (TBS) may provide more information.
As you see in the example it becomes easy to find the SID and key plus my preassigned group number when a UA is written:
P 01010203 01 XXXXXXXXXXXXXX ; added by Emu Thu Aug  5 11:08:56 2021 UA: 00516291
Using the wiki info it becomes easy to create a key for a transponder and group of channels on it.

Using the hash method new UA's are written:
P 7058FFFF 01 XXXXXXXXXXXXXX ; added by Emu Thu Aug 12 20:01:50 2021 UA: 005DE3B2
This makes it impossible to determine the key because first, Where did the 705 come from? And 2, FFFF tells nothing.

I see CW's in the log (live and saved). Useful?
2021/08/20 02:19:42 **66F0EFC3** r      (emu) calculated cw 0: 2ADF52C4046E2CE3
The numbers in bold are not covered (unless I missed something) in the wiki. Useful?

I know this is a request site. A log level or method in my topic to reveal the actual calculation would be nice to have.
Or an expanded tutorial/wiki to assist this task. ECM's change sometimes daily, sometimes months.

oscam-emu, 2021-08-20:
The hash is automatically calculated from your setup (transponder info of the powervu channel) using the the enigma namespace, tsid and onid. It's just a crc32 string produced by these values as soon as you tune into that transponder. Not something that you calculate by hand, or know a priori. Nevertheless, you can display the hash by activating debug level 2.

The same way (debug level 2), you can display the group id. The group id (7058 in your example) is a random number you specify in your softcam.key. It means nothing, but it has to be unique for each group of channels (transponder). You can choose whatever you like (hex digits only).

Then the ecm key is formed by the group id (7058) and the dummy service id (FFFF), since we are using the same ecm key for all services in that transponder.

I hope this covers your questions. What is that you want to do exactly?

7687476874, 2021-08-20:
How should Oscam Config settings be for PC Windows.

7687476874, 2021-08-20:
How should Oscam Config settings be for PC Windows.

[global]
logfile                       = /dev/tty
clienttimeout                 = 8000
maxlogsize                    = 1024
waitforcards                  = 0
waitforcards_extra_delay      = 100
preferlocalcards              = 1
disablecrccws_only_for        = 0E00:000000

[streamrelay]

[dvbapi]
enabled                       = 1
au                            = 1
pmt_mode                      = 0
request_mode                  = 0
listen_port                   = 11500
delayer                       = 60
user                          = dvbapi
read_sdt                      = 2
write_sdt_prov                = 1
extended_cw_api               = 1
boxtype                       = pc

[webif]
httpport                      = 8888
httpallowed                   = 127.0.0.1,192.168.0.0-192.168.255.255,10.0.0.0-10.255.255.255,172.16.0.0-172.31.255.255,255.255.255.255

7687476874, 2021-08-20:
OSCam TVheadend

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

closed 2024-04-13 by nautilus7