oscam-emu/oscam-patched-old #87: PowerVu and PMT

EnoSat, 2021-11-19:
VideoPID has now been removed from PMT for some programs, but VideoPID is actually in the stream. However, after a manual update of lamedb, this VideoPID is ignored.
[image]
I saw a patch where it was solved by fixed pid settings, but it would be better if oscam added this pid from lamedb
`if(*video_pid == 0)
{
   *video_pid =  0x384;
   demux[demux_id].STREAMpidsType[demux[demux_id].STREAMpidcount] = STREAM_VIDEO;
   demux[demux_id].STREAMpids[demux[demux_id].STREAMpidcount] = *video_pid;
   demux[demux_id].STREAMpidcount++;
}`

scooserver, 2021-11-22:
Hi,you been able to fix this channels?can i apply this patch manually ?

EnoSat, 2021-11-22:
i am download oscam with this patch, but in my box not work

oscam-emu, 2021-11-22:
This "solution" cannot be accepted. It just hardcodes the exact video pid into the code.

EnoSat, 2021-11-23:
yes, better read ECM pid etc. from A line
Discovery Hungary HD
p:,C:0e00
0384:00c00000:0021:0085:1:0:0
[image]
0384:0e080000:0002:005e:1:0:0
Discovery Hungary HD
p:,c:000384,c:010385,c:030384,c:050001,c:0903a70202,c:1603de,C:0e00,f:4
[image]

EnoSat, 2021-11-24:
today all transponder 5E/12360V without VPID in PMT
https://mega.nz/file/sQsXSaZR#Cz9Q8zhh7HMzS3OMdqV5SUZJQrw58LhskZOMhyNLemI

zinonino, 2021-11-25:
Here in TVH when manuel add vpid, when not in PMT descriptor - oscam not see Video demuxer 

>  Demuxer 0 descrambling PID 0 CAID 0E00 PROVID 000000 ECMPID 177B CHID 00 VPID 0384

> 2021/11/25 06:59:12 2E4071E1 c   (dvbapi)   00 00 6F 2A 01 00 00 03 DE 
2021/11/25 06:59:12 2E4071E1 c   (dvbapi) Demuxer 0 found MPEG-2 audio stream (type: 04 pid: 0385)
2021/11/25 06:59:12 2E4071E1 c   (dvbapi) Demuxer 0 found ISO 639 language descriptor (tag: 0A length: 04)
2021/11/25 06:59:12 2E4071E1 c   (dvbapi) Demuxer 0 found MPEG-2 audio stream (type: 04 pid: 0386)
2021/11/25 06:59:12 2E4071E1 c   (dvbapi) Demuxer 0 found ISO 639 language descriptor (tag: 0A length: 04)
2021/11/25 06:59:12 2E4071E1 c   (dvbapi) Demuxer 0 found MPEG-2 audio stream (type: 04 pid: 0387)
2021/11/25 06:59:12 2E4071E1 c   (dvbapi) Demuxer 0 found ISO 639 language descriptor (tag: 0A length: 04)
2021/11/25 06:59:12 2E4071E1 c   (dvbapi) Demuxer 0 found MPEG-2 audio stream (type: 04 pid: 0388)
2021/11/25 06:59:12 2E4071E1 c   (dvbapi) Demuxer 0 found ISO 639 language descriptor (tag: 0A length: 04)
2021/11/25 06:59:12 2E4071E1 c   (dvbapi) Demuxer 0 found MPEG-2 PES private data stream (type: 06 pid: 03A7)
2021/11/25 06:59:12 2E4071E1 c   (dvbapi) Demuxer 0 found subtitling descriptor (tag: 59 length: 08)
2021/11/25 06:59:12 2E4071E1 c   (dvbapi) Demuxer 0 found 1 ECM pids and 5 STREAM pids in PMT

> 

with this patch or wthout is same not have viodeo

scooserver, 2021-11-25:
I edited the tvheadend channel (Discovery HD) now i have the h264 VPID,but oscam not decrypting 0000 .Tried with and without the  patch provided on sat-universe  dot com

EnoSat, 2021-11-25:
and levellog 128 ?

scooserver, 2021-11-25:
I belive it was 512.

EnoSat, 2021-11-27:
`--- a/module-dvbapi.c	2021-11-26 08:45:06.494672401 +0100
+++ b/module-dvbapi.c	2021-11-26 08:40:22.729713470 +0100
@@ -346,6 +346,14 @@
 
 int32_t pausecam = 0, disable_pmt_files = 0;
 
+struct vpid_mapping
+{
+	uint16_t apid;
+	uint16_t vpid;
+	
+	struct vpid_mapping *next;
+} *vpid_mapping = NULL;
+
 DEMUXTYPE demux[MAX_DEMUX];
 struct s_dvbapi_priority *dvbapi_priority;
 struct s_client *dvbapi_client;
@@ -2994,7 +3002,7 @@
 		type = tolower((uint8_t)type);
 
 		if(ret < 1 || (type != 'p' && type != 'i' && type != 'm' && type != 'd' &&
-			type != 's' && type != 'l' && type != 'j' && type != 'a' && type != 'x'))
+			type != 's' && type != 'l' && type != 'j' && type != 'a' && type != 'x' && type != 'v'))
 		{
 			//fprintf(stderr, "Warning: line containing %s in %s not recognized, ignoring line\n", token, cs_prio);
 			// fprintf would issue the warning to the command line, which is more consistent with other config warnings
@@ -3041,6 +3049,44 @@
 			continue;
 		}
 #endif
+		
+		if( type == 'v' )
+		{
+			struct vpid_mapping *map;
+			
+			if(!cs_malloc(&map, sizeof(struct vpid_mapping)))
+			{
+				return;
+			}
+
+			map->apid = 0;
+			map->apid = 0;
+			map->next = NULL;
+
+			if( sscanf(str1, "%04x:%04x", &map->apid, &map->vpid ) != 2 )
+			{
+				NULLFREE( map );
+			}
+			else
+			{
+				if( vpid_mapping == NULL )
+				{
+					vpid_mapping = map;
+				}
+				else
+				{
+					struct vpid_mapping *p;
+					
+					for(p = vpid_mapping; p->next != NULL; p = p->next) { ; }
+					
+					p->next = map;
+				}
+			}
+
+			cs_log("Added new mapping for channels without vpid in PMT: apid: %04X -> vpid: %04X", map->apid, map->vpid);
+			continue;
+		}
+		
 		char c_srvid[34];
 		c_srvid[0] = '\0';
 		uint32_t caid = 0, provid = 0, srvid = 0, ecmpid = 0, cadata = 0;;
@@ -4118,6 +4164,8 @@
 {
 	uint16_t i, elementary_pid, es_info_length, offset = 0;
 	uint8_t stream_type, type;
+	uint16_t new_video_pid = 0;
+	struct vpid_mapping *map;
 
 	for(i = 0; i + 4 < length; i += 5 + es_info_length)
 	{
@@ -4135,6 +4183,18 @@
 		cs_log_dbg(D_DVBAPI,"Demuxer %d found %s stream (type: %02X pid: %04X)",
 			demux_id, get_stream_type_txt(stream_type), stream_type, elementary_pid);
 
+		if( new_video_pid == 0 )
+		{
+			for( map = vpid_mapping; map != NULL; map = map->next )
+			{
+				if( elementary_pid == map->apid )
+				{
+					new_video_pid = map->vpid;
+					break;
+				}
+			}
+		}
+
 		if(es_info_length != 0 && es_info_length < length)
 		{
 			if(ca_pmt_cmd_id != NULL) // We are on CA PMT parsing
@@ -4207,6 +4267,15 @@
 		demux[demux_id].STREAMpids[demux[demux_id].STREAMpidcount] = elementary_pid;
 		demux[demux_id].STREAMpidcount++;
 	}
+
+	if(*video_pid == 0 && new_video_pid != 0)
+	{
+		cs_log("Channel without VPID in PMT found in mapping list -> forcing VPID to %04X", new_video_pid);
+		*video_pid = new_video_pid;
+		demux[demux_id].STREAMpidsType[demux[demux_id].STREAMpidcount] = STREAM_VIDEO;
+		demux[demux_id].STREAMpids[demux[demux_id].STREAMpidcount] = new_video_pid;
+		demux[demux_id].STREAMpidcount++;
+	}
 }
 
 static void dvbapi_parse_pmt_info(int32_t demux_id, const uint8_t *buffer, uint16_t length, uint8_t *ca_pmt_cmd_id)
`

seagate-msi, 2021-11-29:
I join the question. also tried solution from EnoSat. does not want to compile oscam after that ... apparently my knowledge is not enough

oscam-emu, 2021-11-29:
The problem has been fixed in the oscam side, with rev 11704. This patch or any other patch is not needed at the moment. Neither any modification by the users in their config files is needed, like it was necessary with the patch above.

closed 2021-11-29 by oscam-emu

seagate-msi, 2021-11-29:
unfortunately this does not work on the Thor 0.8W satellite. no video pid

Nov 29 23:05:45[Discovery Science HD i/1] PMT pnr:3200 version:18
Nov 29 23:05:45[Discovery Science HD i/1] PMT    cas:0x0E00 pid:6035
Nov 29 23:05:45[Discovery Science HD i/1] PMT    descriptor:0x0F0453415053
Nov 29 23:05:45[Discovery Science HD i/1] PMT    descriptor:0x050443554549
Nov 29 23:05:45[Discovery Science HD i/1] PCR pid:3200
Nov 29 23:05:45[Discovery Science HD i/1] AUDIO pid:3201 type:0x06
Nov 29 23:05:45[Discovery Science HD i/1] AUDIO    lang:eng
Nov 29 23:05:45[Discovery Science HD i/1] AUDIO    descriptor:0x6A020000
Nov 29 23:05:45[Discovery Science HD i/1] AUDIO pid:3202 type:0x06
Nov 29 23:05:45[Discovery Science HD i/1] AUDIO    lang:hun
Nov 29 23:05:45[Discovery Science HD i/1] AUDIO    descriptor:0x6A020000
Nov 29 23:05:45[Discovery Science HD i/1] AUDIO pid:3203 type:0x06
Nov 29 23:05:45[Discovery Science HD i/1] AUDIO    lang:tur
Nov 29 23:05:45[Discovery Science HD i/1] AUDIO    descriptor:0x6A020000
Nov 29 23:05:45[Discovery Science HD i/1] AUDIO pid:3204 type:0x06
Nov 29 23:05:45[Discovery Science HD i/1] AUDIO    lang:rus
Nov 29 23:05:45[Discovery Science HD i/1] AUDIO    descriptor:0x6A020000
Nov 29 23:05:45[Discovery Science HD i/1] DATA pid:3212 type:0x86
Nov 29 23:05:45[Discovery Science HD i/1] SUB pid:3221 type:0x06
Nov 29 23:05:45[Discovery Science HD i/1] SUB    subtitling lang:swe
Nov 29 23:05:45[Discovery Science HD i/1] SUB pid:3222 type:0x06
Nov 29 23:05:45[Discovery Science HD i/1] SUB    subtitling lang:dan
Nov 29 23:05:45[Discovery Science HD i/1] SUB pid:3223 type:0x06
Nov 29 23:05:45[Discovery Science HD i/1] SUB    subtitling lang:nor
Nov 29 23:05:45[Discovery Science HD i/1] SUB pid:3224 type:0x06
Nov 29 23:05:45[Discovery Science HD i/1] SUB    subtitling lang:dut
Nov 29 23:05:45[Discovery Science HD i/1] SUB pid:3225 type:0x06
Nov 29 23:05:45[Discovery Science HD i/1] SUB    subtitling lang:fin
Nov 29 23:05:45[Discovery Science HD i/1] SUB pid:3226 type:0x06
Nov 29 23:05:45[Discovery Science HD i/1] SUB    subtitling lang:ara
Nov 29 23:05:45[Discovery Science HD i/1] SUB pid:3228 type:0x06
Nov 29 23:05:45[Discovery Science HD i/1] SUB    subtitling lang:rum
Nov 29 23:05:45[Discovery Science HD i/1] SUB pid:3230 type:0x06
Nov 29 23:05:45[Discovery Science HD i/1] SUB    subtitling lang:gre
Nov 29 23:05:45[Discovery Science HD i/1] SUB pid:3232 type:0x06
Nov 29 23:05:45[Discovery Science HD i/1] SUB    subtitling lang:bul
Nov 29 23:05:45[Discovery Science HD i/1] SUB pid:3233 type:0x06
Nov 29 23:05:45[Discovery Science HD i/1] SUB    subtitling lang:heb
Nov 29 23:05:45[Discovery Science HD i/1] SUB pid:3234 type:0x06
Nov 29 23:05:45[Discovery Science HD i/1] SUB    subtitling lang:eng
Nov 29 23:05:45[Discovery Science HD i/1] SUB pid:3235 type:0x06
Nov 29 23:05:45[Discovery Science HD i/1] SUB    subtitling lang:scc
Nov 29 23:05:45[Discovery Science HD i/1] PMT crc32:0x363ECD56
Nov 29 23:05:45[DTX HD i/1] PAT tsid:2 version:18
Nov 29 23:05:45[DTX HD i/1] PAT    PMT pid:16 pnr:0
Nov 29 23:05:45[DTX HD i/1] PAT    PMT pid:990 pnr:900
Nov 29 23:05:45[DTX HD i/1] PAT    PMT pid:1390 pnr:1300
Nov 29 23:05:45[DTX HD i/1] PAT    PMT pid:2390 pnr:2300
Nov 29 23:05:45[DTX HD i/1] PAT    PMT pid:2690 pnr:2600
Nov 29 23:05:45[DTX HD i/1] PAT    PMT pid:2692 pnr:2602
Nov 29 23:05:45[DTX HD i/1] PAT    PMT pid:2693 pnr:2603
Nov 29 23:05:45[DTX HD i/1] PAT    PMT pid:2694 pnr:2604
Nov 29 23:05:45[DTX HD i/1] PAT    PMT pid:2695 pnr:2605
Nov 29 23:05:45[DTX HD i/1] PAT    PMT pid:2696 pnr:2606
Nov 29 23:05:45[DTX HD i/1] PAT    PMT pid:2790 pnr:2700
Nov 29 23:05:45[DTX HD i/1] PAT    PMT pid:2791 pnr:2701
Nov 29 23:05:45[DTX HD i/1] PAT    PMT pid:2792 pnr:2702
Nov 29 23:05:45[DTX HD i/1] PAT    PMT pid:2890 pnr:2800
Nov 29 23:05:45[DTX HD i/1] PAT    PMT pid:3090 pnr:3000
Nov 29 23:05:45[DTX HD i/1] PAT    PMT pid:3091 pnr:3001
Nov 29 23:05:45[DTX HD i/1] PAT    PMT pid:3092 pnr:3002
Nov 29 23:05:45[DTX HD i/1] PAT    PMT pid:3290 pnr:3200
Nov 29 23:05:45[DTX HD i/1] PAT    PMT pid:3291 pnr:3201
Nov 29 23:05:45[DTX HD i/1] PAT    PMT pid:3292 pnr:3202
Nov 29 23:05:45[DTX HD i/1] PAT    PMT pid:3293 pnr:3203
Nov 29 23:05:45[DTX HD i/1] PAT    PMT pid:3294 pnr:3204
Nov 29 23:05:45[DTX HD i/1] PAT    PMT pid:3295 pnr:3205
Nov 29 23:05:45[DTX HD i/1] PAT crc32:0xE9B6E277

OSCam: 1.20_svn Build: r11704-798 Compiler: mipsel-tuxbox-linux-gnu-ssl

oscam-emu, 2021-11-29:
This is not a log from oscam(-emu)...

seagate-msi, 2021-11-29:
2021/11/29 22:25:03 604DF95E c   (dvbapi) Demuxer 1 ecmpid 0 CAID: 0E00 ECM_PID: 1786 PROVID: 000000 
2021/11/29 22:25:03 604DF95E c   (dvbapi) Demuxer 1 found 1 ECM pids and 3 STREAM pids in CA PMT
2021/11/29 22:25:03 604DF95E c   (dvbapi) Demuxer 1 trying to descramble PID 0 CAID 0E00 PROVID 000000 ECMPID 1786 ANY CHID PMTPID 0A84 VPID 0000
2021/11/29 22:25:04 604DF95E c      (ecm) dvbapi (0E00@000000/0000/0A2A/40:333C6A2A49EB763391BECAA2015EC08A): found (1 ms) by emulator - Animal Planet HD Russian
2021/11/29 22:25:04 604DF95E c   (dvbapi) Demuxer 1 found 1 ECM pids and 3 STREAM pids in PMT
2021/11/29 22:25:04 604DF95E c   (dvbapi) Demuxer 1 found no video pid. Using the PCR pid 0A28 instead

seagate-msi, 2021-11-29:
as a result, a black screen was read, without video. only channels work on this satellite:
Discovery Poland HD and Discovery Historia Polska

oscam-emu, 2021-11-29:
Do you see the last line of your log? It says it will use the pcr pid (which is 0x0A28) as video pid. So, oscam does its job perfectly fine!

It's your part, that is modify the channel list, e.g. lamedb if you use enigma2, that is not done correctly...

EnoSat, 2021-11-29:
> as a result, a black screen was read, without video. only channels work on this satellite: Discovery Poland HD and Discovery Historia Polska

without manual update lamedb in e2, screen not work

seagate-msi, 2021-11-30:
unfortunately, even after the thread VPID, PCR is added, it does not work. after that even the audio doesn't work.
I tried it in stream relay mode, nothing .. unfortunately, everyone on the forums also complains ...

dennmtr, 2021-11-30:
> unfortunately, even after the thread VPID, PCR is added, it does not work. after that even the audio doesn't work. I tried it in stream relay mode, nothing .. unfortunately, everyone on the forums also complains ...

yes same here even with 11704

oscam-emu, 2021-11-30:
Listen, you keep posting that it doesn't work, though you provide no evidence about it. I told you, oscam is working just fine, but you need to fix your channel list. You haven't done that properly I assume. You haven't even tell which player you're on, enigma2, tveheadend, windows dvb player, etc.

Now you say that there's people complaining around. This is totally not the case. Stop writing inaccurate things and don't waist my time.

dennmtr, 2021-11-30:
> Listen, you keep posting that it doesn't work, though you provide no evidence about it. I told you, oscam is working just fine, but you need to fix your channel list. You haven't done that properly [...]

i assume that the problem is the pid numbers, echannelizer doesnt accept hex numbers and i dont even known them, i can't figure out the oscam logs. i am losing my favourite programs in discovery channel btw

seagate-msi, 2021-11-30:
dear developer, forgive us for persistence ..
for example I use ENIGMA2, VU + DUO 2. firmware from VTI latest version.
here is an example from LAMEDB

Animal Planet HD
p:,c:000a28,c:030a28,c:050001,C:0e00,f:4
0a2a:0e08b00f:0002:005e:1:0

with this setting, there are no logs in Oscam at all.

If you delete the VpID, then everything is ok in the logs, but there is no picture.
the satellite on which I tested this THOR 0.8W, I kept the picture only for 1 minute, after the reboot it did not appear anymore

maybe we are doing something wrong, please clarify us

seagate-msi, 2021-12-06:
is it possible to fix this for stream relay too?

oscam-emu, 2021-12-06:
Why does stream relay need a fix? It uses whatever pids it gets from your channel list, including the pcr pid.

Have you actually tested it and it didn't work?

seagate-msi, 2021-12-07:
Unfortunately no. the channels on the tuner itself are working, but there are no relay through the streams

oscam-emu, 2021-12-07:
So, you only hear sound? Do you have a log?