oscam-emu/oscam-patched-old #91: PowerVU - Discovery 58W

bracarino, 2022-07-17:
Hi,

All Discovery transponders (3840V, 3880V, 3960V) in 58W are not opening anymore. 
Keys are auto-rolling, but image is still scrambled.

`2022/07/17 20:15:29 0B030161 r      (emu) csaUsed: 0, xorMode: 0, ecmSrvid: 00E7, hashModeCw: 1, modeCW: 25`

cbandcatcher, 2024-03-30:
Hi. Are there any boxes opening PowerVu+ for Discovery channels in 58 West c band?

EnoSat, 2024-03-30:
some Chinese boxes should open it

bracarino, 2024-03-30:
Guys, let's not confuse stuff.

There is no PowerVu+ (0e01) in 58W. All of them are PowerVu (0e00). However, some PowerVu transponders in 58W are using an updated PowerVu (0e00) that adds a nano 01 in ECM and are not supported by oscam as of today.

cbandcatcher, 2024-03-30:
Thks for informing. However, are there any boxes supporting this updated
PowerVu?

bracarino, 2024-03-30:
Yes

cbandcatcher, 2024-03-31:
Thank you for your response. We hope it will be available soon on oscams or
simple softwares.

nautilus7, 2024-03-31:
Feel free to send us the code to add...

closed 2024-03-31 by nautilus7

lorba7068, 2024-04-02:
if a developer does not have it

cbandpowervu, 2024-04-11:
Hi all. There some news about opening Ten sports chs PowerVu in 75 east c band. Any confirmation about this?

EnoSat, 2024-04-12:
try the chinese freesat forum