oscam-emu/oscam-patched-old #98: No video 3720v 58w whit keys valid

lorba7068, 2023-01-12:
A couple of days ago this problem has been occurring, the hashmode and cwmode have not changed, however there is no video... the keys are calculated by oscam emu correctly, because these same keys work and teams like GTmedia open the channels of this tp...3720v Sat: 058w but enigma equipment does not...

arys123, 2023-01-28:
hello, the same problem on Telestar 12° west 11377v, auto roll emm keys are calculating the cw correct but no picture and audio. 
Can anyone fix this or help?

lorba7068, 2023-01-28:
I am right now testing oscam 11424 r770 and I get this log cwmode: 182 ?
[oscam.log](https://github.com/oscam-emu/oscam-patched/files/10528492/oscam.log)

televisivo, 2023-04-09:
> A couple of days ago this problem has been occurring, the hashmode and cwmode have not changed, however there is no video... the keys are calculated by oscam emu correctly, because these same keys [...]

GTmedia still opens this tp?

lorba7068, 2023-04-12:
The channels and only some can be seen video, but not heard. The tp 3720 does not exist, I change to 3710

televisivo, 2023-04-13:
> The channels and only some can be seen video, but not heard. The tp 3720 does not exist, I change to 3710

What version of Oscam are you using?

nautilus7, 2024-03-31:
Feel free to send us any code to add...

closed 2024-03-31 by nautilus7