oscam-emu/oscam-patched-old #102: Add Oscam emu?

lorba7068, 2023-03-22:
Is it possible to add this arrangement? I notice that some have a patch to fix this problem, PV emu has become private...
[image]
[image]
[image]
[image]

WXbet, 2024-03-17:
Should work with oscam-emu 800.

closed 2024-03-17 by WXbet

lorba7068, 2024-03-18:
emu 800 does not seem to solve the problem, despite having emmkey and keys 00 and 01 valid.
[log.txt](https://github.com/oscam-emu/oscam-patched/files/14643123/log.txt)
[image]
[image]

EnoSat, 2024-03-18:
> emu 800 does not seem to solve the problem, despite having emmkey and keys 00 and 01 valid.
emu 800 does not support nano01