oscam-emu/oscam-patched-old #120: Permissions

tromoto, 2024-03-26:
config.sh has attr 664 instead of 755

nautilus7, 2024-03-26:
Yes, that was done with upstream oscam 11754...

WXbet, 2024-03-26:
Maybe this was my fault during merge 😑

nautilus7, 2024-03-26:
https://stackoverflow.com/questions/10516201/updating-and-committing-only-a-files-permissions-using-git-version-control

I have done the same mistake before and took me a few hours to figure out how to fix it.

nautilus7, 2024-03-26:
Later, not now. But no problem, everyone can wait.

AbuBaniaz, 2024-03-26:
https://github.com/oscam-emu/oscam-patched/commit/75676529bd40c70c592ae0649c23f8f8890f9a8d 

Another guide
http://blog.lesc.se/2011/11/how-to-change-file-premissions-in-git.html

nautilus7, 2024-03-26:
It also needs to be fixed in the trunk I think.

nautilus7, 2024-03-26:
No, I mean the streamboard svn repository. I think the changes there is the source of this problem.

AbuBaniaz, 2024-03-26:
> You mean the oscam-svn branch of this same repo?

Checked, that is fine
```100755 blob 07026a96c1f28de5a5e01a4d8f359a4ee4332f49    config.sh
```
So is this
https://github.com/oscam-emu/oscam-svn/tree/trunk

I think it was an accidental/windows error

WXbet, 2024-03-26:
No, trunk dhould be okay. The problem occured during merge the the changes to oscam-patched. This was done on Windows by me. And git on Windows seems respect the file permissions by default. I fogot to explicit switch the file to 755. I can change it, but not before tonight.

AbuBaniaz, 2024-03-26:
It's fixed.  I'll close the issue now.

closed 2024-03-26 by AbuBaniaz

nautilus7, 2024-03-26:
Yes, all good now in every repo.

WXbet, 2024-03-26:
I will re-apply the merge from trunk tonight incl. correct file permissions. So we don’t have this seperate commit.

nautilus7, 2024-03-26:
Don't bother, not worth the trouble. Let alone more commits are already done.

I am also thinking protecting the master branch from force commits, since now there are more people working on the repo.
So, it can only create difficulties.