oscam-emu/oscam-patched-old #128: aarch64

EnoSat, 2024-03-31:
How does **DES decoding** work for you in PowerVu (1W/TV5, 5E/SIS, 5E/MTN, 16E/SNAI, 16E/IPPICA) in aarch64 (DM One-Two)?

CW shows OK (sometimes ECM Checksum error), but no picture.

2024/03/28 17:11:54 1B890B64 c (ecm) get cw for ecm:
2024/03/28 17:11:54 1B890B64 c (ecm) 80 30 3D 30 37 20 0E 00 00 00 00 64 A0 00 EC CB
2024/03/28 17:11:54 1B890B64 c (ecm) A8 E1 BE DF D5 CA 90 D1 42 B9 57 1A 00 40 E0 09
2024/03/28 17:11:54 1B890B64 c (ecm) 00 9A EC BF 1C 12 5F 08 D9 46 31 BD 01 E5 0E D5
2024/03/28 17:11:54 1B890B64 c (ecm) AD 71 6A FA F3 53 54 6A F3 BA C7 88 65 41 53 8F
2024/03/28 17:11:54 18E23D79 r (emu) csaUsed: 0, xorMode: 0, ecmSrvid: 040E (1038), hashModeCw: 0, modeCW: 0
2024/03/28 17:11:54 18E23D79 r (emu) channel hash: 32D2A9BD, group id: 3592
2024/03/28 17:11:54 18E23D79 r (emu) calculated cw 0: 43F7B597D0CE02D3
2024/03/28 17:11:54 18E23D79 r (emu) calculated cw 1: 54927F3BC1792F89
2024/03/28 17:11:54 18E23D79 r (emu) calculated cw 2: 8F0E52F27AE6D6B9
2024/03/28 17:11:54 18E23D79 r (emu) calculated cw 3: 7A259D6D3E5E2551
2024/03/28 17:11:54 18E23D79 r (emu) calculated cw 4: 7A259D6D3E5E2551
2024/03/28 17:11:54 18E23D79 r (emu) calculated cw 5: 7A259D6D3E5E2551
2024/03/28 17:11:54 18E23D79 r (emu) calculated cw 6: B5EF3746EF7FCBCE
2024/03/28 17:11:54 18E23D79 r (emu) calculated cw 7: 94197F7A3D707F8C
2024/03/28 17:11:54 1B890B64 c (ecm) LOCAL (0E00&/040E/1775/0000/40_43F7B597D0CE02D30000000000000000 HOP:00:0000_0000): found (9 ms) by EMU - FEED OCCAS HD
2024/03/28 17:11:54 1B890B64 c (ecm) cw:
2024/03/28 17:11:54 1B890B64 c (ecm) 43 F7 B5 97 D0 CE 02 D3 00 00 00 00 00 00 00 00
2024/03/28 17:11:55 1B890B64 c (ecm) get cw for ecm:
2024/03/28 17:11:55 1B890B64 c (ecm) 81 30 3D 30 36 20 0C 00 00 00 00 65 A0 08 10 99
2024/03/28 17:11:55 1B890B64 c (ecm) BA 11 97 7A 7E C2 C3 2D 15 97 1B B3 02 40 E0 09
2024/03/28 17:11:55 1B890B64 c (ecm) 40 D6 6F 9C 60 35 BF 26 E4 A0 2F 8A F3 49 4B BD
2024/03/28 17:11:55 1B890B64 c (ecm) D0 C3 7B 44 FB D5 24 BC 1C 23 7D DD 23 19 D8 B1
2024/03/28 17:11:55 18E23D79 r (emu) ECM failed: Checksum error
2024/03/28 17:11:55 1B890B64 c (ecm) LOCAL (0E00&/040E/1775/0000/40_00000000000000000000000000000000 HOP:00:0000_0000): not found (4 ms) by EMU - FEED OCCAS HD
2024/03/28 17:11:55 1B890B64 c (dvbapi) Demuxer 0 restarting decoding requests after 0 ms with 1 enabled and 0 disabled ecmpids!
2024/03/28 17:11:55 1B890B64 c (dvbapi) Demuxer 0 trying to descramble PID 0 CAID 0E00 PROVID 000000 ECMPID 1775 ANY CHID PMTPID 0472 VPID 04D6
2024/03/28 17:11:55 1B890B64 c (ecm) cw:
2024/03/28 17:11:55 1B890B64 c (ecm) 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
2024/03/28 17:11:57 1B890B64 c (ecm) get cw for ecm:
2024/03/28 17:11:57 1B890B64 c (ecm) 80 30 3D 30 37 20 0E 00 00 00 00 66 A0 00 BE 49
2024/03/28 17:11:57 1B890B64 c (ecm) D1 71 B6 00 A9 70 B3 92 CD 5E AF 1F 00 40 E0 09
2024/03/28 17:11:57 1B890B64 c (ecm) 00 C7 82 2F BF 0E 7D B6 9F 98 3C 6A D6 E6 C9 50
2024/03/28 17:11:57 1B890B64 c (ecm) A1 AE 5D 0E 67 00 63 CE AA 4C B6 0C 81 41 89 BD
2024/03/28 17:11:57 18E23D79 r (emu) csaUsed: 0, xorMode: 0, ecmSrvid: 040E (1038), hashModeCw: 0, modeCW: 0
2024/03/28 17:11:57 18E23D79 r (emu) channel hash: 32D2A9BD, group id: 3592
2024/03/28 17:11:57 18E23D79 r (emu) calculated cw 0: 4A15152F374CC432
2024/03/28 17:11:57 18E23D79 r (emu) calculated cw 1: 3B1F681020CD92D0
2024/03/28 17:11:57 18E23D79 r (emu) calculated cw 2: A470F2986D0D4552
2024/03/28 17:11:57 18E23D79 r (emu) calculated cw 3: 1C7AB3CEECA44F23
2024/03/28 17:11:57 18E23D79 r (emu) calculated cw 4: 1C7AB3CEECA44F23
2024/03/28 17:11:57 18E23D79 r (emu) calculated cw 5: 1C7AB3CEECA44F23
2024/03/28 17:11:57 18E23D79 r (emu) calculated cw 6: C449C13E0E382992
2024/03/28 17:11:57 18E23D79 r (emu) calculated cw 7: 51FD2592B69286B9
2024/03/28 17:11:57 1B890B64 c (ecm) LOCAL (0E00&/040E/1775/0000/40_4A15152F374CC4320000000000000000 HOP:00:0000_0000): found (6 ms) by EMU - FEED OCCAS HD
2024/03/28 17:11:57 1B890B64 c (ecm) cw:
2024/03/28 17:11:57 1B890B64 c (ecm) 4A 15 15 2F 37 4C C4 32 00 00 00 00 00 00 00 00

With **CSA** in PowerVu (9E/AFN) , the picture is fine.

nautilus7, 2024-03-31:
Is your reception ok? The crc error might indicate low signal.

Also, if I remember correctly, all CWs except from the 1st one should be identical...? This isn't the case here. Can you post a debug log with level 1+128 enabled, starting from when you tune to that channel? I want to see how oscam-emu parses the streams.

nautilus7, 2024-03-31:
Another tought...

Do you have snother receiver to try? Ehat makes you think is related to the specific stb?

EnoSat, 2024-04-01:
Only Dreambox One and Dreambox Two are available with aarch64. Once upon a time (SDG6) 64b was also applied to SatDreamGr for EDISION. And maybe aarch64 can be used for VU Duo 4K, but I have no way to verify that.

EnoSat, 2024-04-01:
log level 129 - start oscam with FTA TV5MONDE EUROPE HD, switch to FEED OCCAS HD and decode with CW ok - but no picture + decode TVMONDE AFRIQUE HD with new nano01 in ECM
[oscam_1W_12643V.log](https://github.com/oscam-emu/oscam-patched/files/14818370/oscam_1W_12643V.log)

EnoSat, 2024-04-01:
same log on EDISION aarch64 with CW ok and picture OK
[oscam_1W_MIO.log](https://github.com/oscam-emu/oscam-patched/files/14822620/oscam_1W_MIO.log)

nautilus7, 2024-04-01:
Since you say it works in edision, then the powervu algo is ok.

If more people confirm this, then it might be that dreambox use a different api to set the hardware descramblers, which oscam is not aware of. Can you point the dreambox guys to this issue and ask them to inform us of their api and what their stb expect from oscam?

EnoSat, 2024-04-01:
Do you know which ones?

nautilus7, 2024-04-01:
Which what?

If there is a new api for newer dreambox, we should be informed of it by the devs.

EnoSat, 2024-04-08:
:(

nautilus7, 2024-04-08:
> :(

What does this suppose to mean?

EnoSat, 2024-04-10:
a sad smiley face